zlgopen/awtk #379: 用的最新的程序,slide_view的页是listview带scroll_view的,能左右滑动,不能上下滑动

21pages, 2020-04-03:
每个页里都是label,没有占据focus的
之前您改过,我写了,您看看(代码外边的<window>被吞了)
https://github.com/zlgopen/awtk/commit/aabb1e8825b0c7b94ef7e20cfe113e80cdc21a61#commitcomment-38185793

xianjimli, 2020-04-03:
inc_netconfig_cat.xml 等文件没有，我没法试试。你贴代码的时候用 ``` 把代码包起来(参考markdown的语法)，否则很难看。

21pages, 2020-04-04:
```
<window anim_hint="htranslate">
  <slide_view x="0" y="0" w="100%" h="100%" focused="true" loop="true" auto_play="0" anim_hint="translate" vertical="false">
    <view x="0" y="0" w="100%" h="100%" children_layout="default(r=0,c=1,s=0,m=0)">
        <list_view w="100%" h="100%" item_height="20" auto_hide_scroll_bar="false"
                    focused="true">
            <scroll_view name="view" w="100%" h="100%" children_layout="list_view(xm=1,ym=1)">
                <list_item children_layout="default(r=1,c=0)">
                    <label w="100%" style="center" text="1/2:"/>
                </list_item>

                <list_item children_layout="default(r=1,c=0)">
                    <label w="100%" style="center" text="test"/>
                </list_item>
                <list_item children_layout="default(r=1,c=0)">
                    <label w="100%" style="center" text="test"/>
                </list_item>
                <list_item children_layout="default(r=1,c=0)">
                    <label w="100%" style="center" text="test"/>
                </list_item>
                <list_item children_layout="default(r=1,c=0)">
                    <label w="100%" style="center" text="test"/>
                </list_item>
                <list_item children_layout="default(r=1,c=0)">
                    <label w="100%" style="center" text="test"/>
                </list_item>
                <list_item children_layout="default(r=1,c=0)">
                    <label w="100%" style="center" text="test"/>
                </list_item>
                <list_item children_layout="default(r=1,c=0)">
                    <label w="100%" style="center" text="test"/>
                </list_item>

            </scroll_view>
            <scroll_bar_m name="bar" x="right" y="0" w="3" h="100%" value="0"/>
        </list_view>
    </view>

    <view x="0" y="0" w="100%" h="100%" children_layout="default(r=0,c=1,s=0,m=0)">
        <list_view w="100%" h="100%" item_height="20" auto_hide_scroll_bar="false"
                    focused="true">
            <scroll_view name="view" w="100%" h="100%" children_layout="list_view(xm=1,ym=1)">
                <list_item children_layout="default(r=1,c=0)">
                    <label w="100%" style="center" text="2/2:"/>
                </list_item>

                <list_item children_layout="default(r=1,c=0)">
                    <label w="100%" style="center" text="test"/>
                </list_item>
                <list_item children_layout="default(r=1,c=0)">
                    <label w="100%" style="center" text="test"/>
                </list_item>
                <list_item children_layout="default(r=1,c=0)">
                    <label w="100%" style="center" text="test"/>
                </list_item>
                <list_item children_layout="default(r=1,c=0)">
                    <label w="100%" style="center" text="test"/>
                </list_item>
                <list_item children_layout="default(r=1,c=0)">
                    <label w="100%" style="center" text="test"/>
                </list_item>
                <list_item children_layout="default(r=1,c=0)">
                    <label w="100%" style="center" text="test"/>
                </list_item>
                <list_item children_layout="default(r=1,c=0)">
                    <label w="100%" style="center" text="test"/>
                </list_item>

            </scroll_view>
            <scroll_bar_m name="bar" x="right" y="0" w="3" h="100%" value="0"/>
        </list_view>
    </view>

  </slide_view>
</window>
```

xianjimli, 2020-04-04:
用最新代码，我这里上下左右都可滑动啊。

21pages, 2020-04-04:
用键盘

xianjimli, 2020-04-05:
加了按键处理，你更新代码试试。

21pages, 2020-04-05:
用键盘还是不好用,现象是这样的,我进界面后按下键不滑动,按右键翻页后发现第二页已经滑到下面了,应该是刚才在第一页的下键下滑了第二页,然后第二页上下翻是好用的,第一页就是不好用,但是如果先用鼠标滑动第一页,再用键盘就好用了.....还一个现象是滑动的特别慢

xianjimli, 2020-04-06:
滑动的慢的问题，你参考这个文档，把up/down映射成pageup/pagedown: https://github.com/zlgopen/awtk/blob/master/docs/map_key.md

xianjimli, 2020-04-06:
为了支持每个页面都有初始焦点控件，为view增加了属性default_focused_widget。请参考：https://github.com/zlgopen/awtk/blob/master/docs/pages_slide_view_default_focused_child_for_each_page.md

需要更新AWTK代码。

21pages, 2020-04-06:
完美!

closed 2020-04-06 by 21pages