zlgopen/awtk #381: lcd使用double fb模式时如何通知驱动交换fb?

innovak, 2020-04-10:
项目使用的是cpu总线屏,原本的gui是写脏后通知驱动读取fb数据发送到屏.移植awtk后参考stm32f767的例程,尚未理解.又查看f429的例程,似乎是通过创建lcd时返回的LCD结构体中设置lcd->swap回调函数通知驱动交换?
lcd_t结构体除了swap,似乎还有flush之类的回调,请问有没有这些回调函数设置的说明?另外比如我只使用单fb的话,是否可以设置flush回调函数来通知驱动刷屏?

xianjimli, 2020-04-11:
>另外比如我只使用单fb的话,是否可以设置flush回调函数来通知驱动刷屏?

可以。请参考：https://github.com/zlgopen/awtk/blob/master/docs/lcd.md

一般通过下面几种方式将 offline framebuffer 上的图像显示到 online framebuffer 上：

* swap 的方式。一般重载 lcd 的 swap 函数，通过系统调用切换 online framebuffer 和 offline framebuffer 的角色。一般适用于 linux 2/3 framebuffer 系统。
参考： https://github.com/zlgopen/awtk-linux-fb/blob/master/awtk-port/lcd\_linux\_fb.c

* 缺省 flush 的方式。一般由 lcd 的 flush 函数把 offline framebuffer 中的图像（脏矩形内的部分） 拷贝到 offline framebuffer。这是缺省实现，一般不需修改。在这种情况下，一般用 lcd_mem_xxx_create_double_fb 创建 lcd 对象，注意 online framebuffer 参数一定要是系统显存。
如果要做显示同步，也就是等显卡把数据真正显示到屏幕上，可以重载 lcd 的 sync 函数。一般适用于 linux 单 framebuffer 的情况。参考： https://github.com/zlgopen/awtk-linux-fb/blob/master/awtk-port/lcd\_linux\_fb.c

* 自定义 flush 的方式。有的系统没有 online framebuffer，只有 offline framebuffer。比如显示屏与 MCU 之间用 SPI 连接，那就需要重载 flush 函数，把 offline framebuffer 中的图像（脏矩形内的部分） 数据传输到显示屏。在这种情况下一般用 lcd_mem_xxx_create_single_fb 创建 lcd 对象，并重载 lcd 的 flush 函数。

innovak, 2020-04-11:
感谢回复,我这情况是只有offline的fb,现在是重载了swap函数发送,拷贝加发送的时间用了35ms,没法和te同步,闪屏比较严重.我再研究研究自定义flush的方式.

closed 2020-04-11 by innovak