zlgopen/awtk #382: 脏矩形的尺寸问题,能否在一帧中实现多个脏矩形区域?

innovak, 2020-04-12:
移植的项目采用单offline的fb,重载了lcd的flush函数使用脏矩形刷屏.遇到的问题是因为硬件接口限制,刷全屏会比较慢.在一些特殊情况:分布很分散的图标状态变化时,脏矩形尺寸会很大.
能否实现每帧内刷多个尺寸粒度更精细的脏矩形来减少数据量? 接触awtk不久,一时没有头绪.

xianjimli, 2020-04-12:
目前只支持单脏矩形。

poweel, 2020-04-14:
@innovak

poweel, 2020-04-14:
最后刷屏的是这函数，你是怎么传递脏矩形给驱动的？

static ret_t lcd_mem_linux_sync(lcd_t* lcd) {
  fb_info_t* fb = (fb_info_t*)(lcd->impl_data);

  fb_sync(fb);

  return RET_OK;
}

poweel, 2020-04-14:
我之前曾经把屏幕分成10*10的小块，通过内存比较来确定刷哪块

innovak, 2020-04-14:
> 最后刷屏的是这函数，你是怎么传递脏矩形给驱动的？
> 
> static ret_t lcd_mem_linux_sync(lcd_t* lcd) {
> fb_info_t* fb = (fb_info_t*)(lcd->impl_data);
> 
> fb_sync(fb);
> 
> return RET_OK;
> }

我这不是在linux平台,不过原理类似,lcd->flush重载为自己的刷屏函数.
被回调时会传递lcd指针过来,参考默认flush中的做法从lcd->dirty_rect中取出脏矩形的xy坐标和宽高
    lcd_mem_t *mem = (lcd_mem_t *)lcd;
    rect_t *r = &(lcd->dirty_rect);
    abcd(r->x,r->y, r->w,r->h,mem->offline_fb);//真正刷屏的函数