zlgopen/awtk #384: mledit多行文编辑器无法设置插入模式/覆盖模式切换，可否添加

BFDX, 2020-04-13:
目前需要增加一个覆盖模式，9宫格键盘输入时需要切换文字

xianjimli, 2020-04-13:
"9宫格键盘输入时需要切换文字" 是什么意思

BFDX, 2020-04-14:
类似早期诺基亚手机的九宫格输入法，比如此时我按数字“1”，那么实际上我想输入的是“1”这个按键上面的标注的“a/b/c”三个字母的其中一个，此时需要切换为覆盖模式，连续按“1”切换“a/b/c”，而不是按“1”直接插入字符“a”

xianjimli, 2020-04-14:
明白了，我想想

xianjimli, 2020-04-15:
你取t9-exper分支试试，基本功能可以用了。把kb_t9.xml拷贝到kb_default.xml，或者给编辑器指定keyboard:

```xml
<window anim_hint="htranslate" >
  <edit x="10" y="10" w="-20" h="30" keyboard="kb_t9" focused="true" />
  <mledit x="10" y="50" w="-20" h="80" keyboard="kb_t9" focused="true" />
  <button name="close" x="center" y="bottom:10" w="25%" h="30" text="Close"/>
</window>
```

BFDX, 2020-04-15:
你好，我目前的项目是外部键盘，比较理想的实现是mledit本神可以切换 插入/覆盖 模式，可以参考这个代码吗？还是只能通过弹出这个虚拟T9键盘去做

xianjimli, 2020-04-15:
应该可以，用这几个函数(具体用法参考keyboard.c)：
```c
 /**
 * @method input_method_dispatch_preedit
 * 分发进入预编辑状态的事件。
 * @annotation ["scriptable"]
 * @param {input_method_t*} im 输入法对象。
 *
 * @return {ret_t} 返回RET_OK表示成功，否则表示失败。
 */
ret_t input_method_dispatch_preedit(input_method_t* im);

/**
 * @method input_method_dispatch_preedit_confirm
 * 分发确认预编辑状态的事件(提交预编辑内容，退出预编辑状态)。
 * @annotation ["scriptable"]
 * @param {input_method_t*} im 输入法对象。
 *
 * @return {ret_t} 返回RET_OK表示成功，否则表示失败。
 */
ret_t input_method_dispatch_preedit_confirm(input_method_t* im);

/**
 * @method input_method_commit_text
 * 提交输入文本。
 * @annotation ["scriptable"]
 * @param {input_method_t*} im 输入法对象。
 * @param {char*} text 文本。
 *
 * @return {ret_t} 返回RET_OK表示成功，否则表示失败。
 */
ret_t input_method_commit_text(input_method_t* im, const char* text);
```

closed 2020-05-21 by xianjimli