zlgopen/awtk #385: 上下拉动页面出现图片重叠

huamiao1996, 2020-04-14:
在一个button控件中写了一个edit或者combobox子控件，在拖动这个子控件后，整个界面花屏了，这个问题有什么好的解决方法吗

xianjimli, 2020-04-15:
启用脏矩形时，控件只能在父控件范围内拖动。

huamiao1996, 2020-04-15:
不好意思，我描述的不是很准确。我的xml是一个列表，每个列表项是一个button，button靠右边有子控件（edit或者combo_box），想实现的效果是向上滑动手指，实现列表的滚动，如果点中了button滑动是正常的，如果点中了button的子控件滑动不了。

xianjimli, 2020-04-15:
把XML贴上来我试试

huamiao1996, 2020-04-15:
<window move_focus_prev_key="up" move_focus_next_key="down"  style="menu_bg">
  <list_view x="0"  y="0" w="100%" h="36"  style="title">
    <label  name="detail_label" x="0"  y="0" w="100%" h="36" tr_text="Detail" style="white_mid"/>
  </list_view>
  <pages h="190" w="100%" x="0" y="40" style="nostyle">
    <list_view style="bg_ground" x="0"  y="0" w="100%" h="100%" item_height="38">
      <scroll_view style="bg_ground" auto_hide_scroll_bar="true" name="contact_option_detail_view" x="0"  y="0" w="100%" h="100%">
        <button name="item1" style="lv2_menu_left" with_focus_state="true">
          <label x="20" y="0" w="210" h="100%" style="black_left_small_fontsize" tr_text="1.Name"/>
          <edit x="250" y="0" w="210" h="100%" style="black_text_nobg" name="contact_option_detail_name" max="20" auto_fix="true" focusable="true" focus="true" im_method="abc" input_method_options="2aB;123;abc;ABC;Ab2;pinyin;"/>
          <label  x="0" y="38" w="100%" h="1"   style="only_horizontal_line"/>
        </button>
        <button name="item2" style="lv2_menu_left" with_focus_state="true">
          <label x="20" y="0" w="210" h="100%" tr_text="2.Office Number" style="black_left_small_fontsize"/>
          <edit x="250" y="0" w="210" h="100%" style="black_text_nobg" name="contact_option_detail_officenumber" max="20" auto_fix="true" focusable="true" im_method="123" />
          <label  x="0" y="38" w="100%" h="1"   style="only_horizontal_line"/>
        </button>
		<button name="item3" style="lv2_menu_left" with_focus_state="true">
          <label x="20" y="0" w="210" h="100%" tr_text="3.Mobile Number" style="black_left_small_fontsize"/>
          <edit x="250" y="0" w="210" h="100%" style="black_text_nobg" name="contact_option_detail_mobilenumber" max="20" auto_fix="true" focusable="true" im_method="123" />
          <label  x="0" y="38" w="100%" h="1"   style="only_horizontal_line"/>
        </button>
		<button name="item4" style="lv2_menu_left" with_focus_state="true">
          <label x="20" y="0" w="210" h="100%" tr_text="4.Other Number" style="black_left_small_fontsize"/>
          <edit x="250" y="0" w="210" h="100%" style="black_text_nobg" name="contact_option_detail_othernumber" max="20" auto_fix="true" focusable="true" im_method="123" />
          <label  x="0" y="38" w="100%" h="1"   style="only_horizontal_line"/>
        </button>
		<button name="item5" style="lv2_menu_left" with_focus_state="true">
		  <label  x="20" y="0" w="210" h="100%"  tr_text="5.Account" style="black_left_small_fontsize"/>
          <combo_box right_margin="0" style="white" x="250" y="0" w="210" h="100%" name="account" tr_text="Auto" options="Auto;" selected_index="0" focusable="true"/>
          <label  x="0" y="38" w="100%" h="1"   style="only_horizontal_line"/>
        </button>
        <button name="item6" style="lv2_menu_left" with_focus_state="true">
		  <label  x="20" y="0" w="210" h="100%"  tr_text="6.Ring Type" style="black_left_small_fontsize"/>
          <combo_box right_margin="0" style="white" x="250" y="0" w="210" h="100%" name="ring_type" tr_text="Type 1" options="Type 1;Type 2;Type 3;Type 4;Type 5;Type 6;Type 7;Type 8;Type 9;Type 10;" selected_index="0" focusable="true"/>
          <label  x="0" y="38" w="100%" h="1"   style="only_horizontal_line"/>
        </button>
        <button name="item7" style="lv2_menu_left" with_focus_state="true">
		  <label  x="20" y="0" w="210" h="100%" tr_text="7.Group" style="black_left_small_fontsize"/>
          <combo_box right_margin="0" style="white" x="250" y="0" w="210" h="100%" name="contact_option_detail_group" tr_text="All Contacts" options="All Contacts;" selected_index="0" focusable="true"/> 
          <label  x="0" y="38" w="100%" h="1"   style="only_horizontal_line"/>
        </button>
		<button name="item8" style="lv2_menu_left" focusable="true">
          <label  x="20" y="0" w="210" h="100%"  tr_text="8.Photo" style="black_left_small_fontsize"/>
		  <label  x="250" y="0" w="210" h="100%" name="photo" text="Contact.png" style="black_left_small_fontsize"/>
          <label  x="0" y="38" w="100%" h="1"   style="only_horizontal_line"/>
        </button>
		<button name="item9" style="lv2_menu_left" with_focus_state="true">
          <label x="20" y="0" w="210" h="100%" tr_text="9.Auto Divert" style="black_left_small_fontsize"/>
          <edit x="250" y="0" w="210" h="100%" style="black_text_nobg" name="auto_divert" max="20" auto_fix="true" focusable="true" im_method="123" />
          <label  x="0" y="38" w="100%" h="1"   style="only_horizontal_line"/>
        </button>
      </scroll_view>
	  <scroll_bar_m name="bar" x="right" y="0" w="5" h="100%" value="0" visible="false"/>
    </list_view>
  </pages>
  <view x="0" y="bottom" w="100%" h="38" children_layout="default(r=1,c=4)" style="softkey_bg" is_keyboard="true">
    <button name="contact_option_detail_back" tr_text="Back" style="nostyle"  is_keyboard="true"/>
    <button name="im_switch" tr_text="" style="nostyle" is_keyboard="true"/>
    <button name="input_delete" tr_text="" style="nostyle" is_keyboard="true"/>
    <button name="contact_option_detail_save" tr_text="Save" style="nostyle" is_keyboard="true"/>
  </view>
</window>

huamiao1996, 2020-04-15:
您能看得到吗

huamiao1996, 2020-04-15:
> 把XML贴上来我试试

<window move_focus_prev_key="up" move_focus_next_key="down"  style="menu_bg">
  <list_view x="0"  y="0" w="100%" h="36"  style="title">
    <label  name="detail_label" x="0"  y="0" w="100%" h="36" tr_text="Detail" style="white_mid"/>
  </list_view>
  <pages h="190" w="100%" x="0" y="40" style="nostyle">
    <list_view style="bg_ground" x="0"  y="0" w="100%" h="100%" item_height="38">
      <scroll_view style="bg_ground" auto_hide_scroll_bar="true" name="contact_option_detail_view" x="0"  y="0" w="100%" h="100%">
        <button name="item1" style="lv2_menu_left" with_focus_state="true">
          <label x="20" y="0" w="210" h="100%" style="black_left_small_fontsize" tr_text="1.Name"/>
          <edit x="250" y="0" w="210" h="100%" style="black_text_nobg" name="contact_option_detail_name" max="20" auto_fix="true" focusable="true" focus="true" im_method="abc" input_method_options="2aB;123;abc;ABC;Ab2;pinyin;"/>
          <label  x="0" y="38" w="100%" h="1"   style="only_horizontal_line"/>
        </button>
        <button name="item2" style="lv2_menu_left" with_focus_state="true">
          <label x="20" y="0" w="210" h="100%" tr_text="2.Office Number" style="black_left_small_fontsize"/>
          <edit x="250" y="0" w="210" h="100%" style="black_text_nobg" name="contact_option_detail_officenumber" max="20" auto_fix="true" focusable="true" im_method="123" />
          <label  x="0" y="38" w="100%" h="1"   style="only_horizontal_line"/>
        </button>
		<button name="item3" style="lv2_menu_left" with_focus_state="true">
          <label x="20" y="0" w="210" h="100%" tr_text="3.Mobile Number" style="black_left_small_fontsize"/>
          <edit x="250" y="0" w="210" h="100%" style="black_text_nobg" name="contact_option_detail_mobilenumber" max="20" auto_fix="true" focusable="true" im_method="123" />
          <label  x="0" y="38" w="100%" h="1"   style="only_horizontal_line"/>
        </button>
		<button name="item4" style="lv2_menu_left" with_focus_state="true">
          <label x="20" y="0" w="210" h="100%" tr_text="4.Other Number" style="black_left_small_fontsize"/>
          <edit x="250" y="0" w="210" h="100%" style="black_text_nobg" name="contact_option_detail_othernumber" max="20" auto_fix="true" focusable="true" im_method="123" />
          <label  x="0" y="38" w="100%" h="1"   style="only_horizontal_line"/>
        </button>
		<button name="item5" style="lv2_menu_left" with_focus_state="true">
		  <label  x="20" y="0" w="210" h="100%"  tr_text="5.Account" style="black_left_small_fontsize"/>
          <combo_box right_margin="0" style="white" x="250" y="0" w="210" h="100%" name="account" tr_text="Auto" options="Auto;" selected_index="0" focusable="true"/>
          <label  x="0" y="38" w="100%" h="1"   style="only_horizontal_line"/>
        </button>
        <button name="item6" style="lv2_menu_left" with_focus_state="true">
		  <label  x="20" y="0" w="210" h="100%"  tr_text="6.Ring Type" style="black_left_small_fontsize"/>
          <combo_box right_margin="0" style="white" x="250" y="0" w="210" h="100%" name="ring_type" tr_text="Type 1" options="Type 1;Type 2;Type 3;Type 4;Type 5;Type 6;Type 7;Type 8;Type 9;Type 10;" selected_index="0" focusable="true"/>
          <label  x="0" y="38" w="100%" h="1"   style="only_horizontal_line"/>
        </button>
        <button name="item7" style="lv2_menu_left" with_focus_state="true">
		  <label  x="20" y="0" w="210" h="100%" tr_text="7.Group" style="black_left_small_fontsize"/>
          <combo_box right_margin="0" style="white" x="250" y="0" w="210" h="100%" name="contact_option_detail_group" tr_text="All Contacts" options="All Contacts;" selected_index="0" focusable="true"/> 
          <label  x="0" y="38" w="100%" h="1"   style="only_horizontal_line"/>
        </button>
		<button name="item8" style="lv2_menu_left" focusable="true">
          <label  x="20" y="0" w="210" h="100%"  tr_text="8.Photo" style="black_left_small_fontsize"/>
		  <label  x="250" y="0" w="210" h="100%" name="photo" text="Contact.png" style="black_left_small_fontsize"/>
          <label  x="0" y="38" w="100%" h="1"   style="only_horizontal_line"/>
        </button>
		<button name="item9" style="lv2_menu_left" with_focus_state="true">
          <label x="20" y="0" w="210" h="100%" tr_text="9.Auto Divert" style="black_left_small_fontsize"/>
          <edit x="250" y="0" w="210" h="100%" style="black_text_nobg" name="auto_divert" max="20" auto_fix="true" focusable="true" im_method="123" />
          <label  x="0" y="38" w="100%" h="1"   style="only_horizontal_line"/>
        </button>
      </scroll_view>
	  <scroll_bar_m name="bar" x="right" y="0" w="5" h="100%" value="0" visible="false"/>
    </list_view>
  </pages>
  <view x="0" y="bottom" w="100%" h="38" children_layout="default(r=1,c=4)" style="softkey_bg" is_keyboard="true">
    <button name="contact_option_detail_back" tr_text="Back" style="nostyle"  is_keyboard="true"/>
    <button name="im_switch" tr_text="" style="nostyle" is_keyboard="true"/>
    <button name="input_delete" tr_text="" style="nostyle" is_keyboard="true"/>
    <button name="contact_option_detail_save" tr_text="Save" style="nostyle" is_keyboard="true"/>
  </view>
</window>

huamiao1996, 2020-04-15:
[随笔.docx](https://github.com/zlgopen/awtk/files/4479325/default.docx)

huamiao1996, 2020-04-15:
[contact_option_detail.zip](https://github.com/zlgopen/awtk/files/4479535/contact_option_detail.zip)

xianjimli, 2020-04-15:
>拖动这个子控件后，整个界面花屏

 我没看到。

> 如果点中了button滑动是正常的，如果点中了button的子控件滑动不了。

是希望拖动编辑器时也可以滑动吧

huamiao1996, 2020-04-16:
[contact_option_detail.zip](https://github.com/zlgopen/awtk/files/4484447/contact_option_detail.zip)

huamiao1996, 2020-04-16:
[error.zip](https://github.com/zlgopen/awtk/files/4484448/error.zip)

huamiao1996, 2020-04-16:
这个是效果图，滑动button的子控件是不正常的，滑动button是正常的

closed 2020-04-16 by huamiao1996