zlgopen/awtk #389: 对tab_button的疑问

YinWenshuo, 2020-04-17:
我的理解是一个tab_button对应<pages>...</pages>里面的一个view，但是在实际中，label，button等控件也会占用一个page。
附件中有问题的window_page_ng.xml会导致点击btn_b时开启时page_c，点击btn_c时开的是空的。。。附件中window_page_ok.xml就可以达到我的预期：
btn_a开启page_a,
btn_b开启page_b,
btn_c开启page_c.

我的功能可以实现了，只是对tab_button的理解有疑问，请答疑解惑下，谢谢。
[xml.tar.gz](https://github.com/zlgopen/awtk/files/4491330/xml.tar.gz)

xianjimli, 2020-04-17:
demoui里 有很多例子，你参考一下例子：
```xml
<window anim_hint="htranslate" theme="tab_bottom">
  <tab_control x="0" y="0" w="100%" h="100%" >
    <pages x="0" y="0" w="100%" h="-60" style="at_top">
      <?include filename="tab_views.inc" ?>
    </pages>
    <tab_button_group x="0" y="bottom" w="100%" h="60" >
      <tab_button icon="msg" active_icon="msg_active" text="Message"/>
      <tab_button icon="contact" active_icon="contact_active" text="Contact" value="true" />
      <tab_button icon="discovery" active_icon="discovery_active" text="Discovery" />
      <tab_button icon="me" active_icon="me_active" text="Me" />
    </tab_button_group>
  </tab_control>
</window>
```

YinWenshuo, 2020-04-17:
对tab_button的使用我没有问题了，例子是比较常规的用法。直奔主题，我如果按下面这种写法会有啥问题？

<window anim_hint="htranslate" theme="tab_bottom">
  <tab_control x="0" y="0" w="100%" h="100%" >
    <pages x="0" y="0" w="100%" h="-60" style="at_top" value="0">
      <view name="a" >
            //省略view的具体内容
       </view>
       <view name="b" >
            //省略view的具体内容
       </view>
       <view name="c" >
            //省略view的具体内容
       </view>
    </pages>
    <tab_button_group x="0" y="bottom" w="100%" h="60" >
      <tab_button name="aa"/>
      <label name="1111" />
      <tab_button name="bbbb"  value="true" />
       <label name="2222" />
      <tab_button name="ccccc" />
      <label name="3333" />
    </tab_button_group>
  </tab_control>
</window>

YinWenshuo, 2020-04-17:
贴的xml能看到吗？或者下载我上面上传的gz包，里面有xml

xianjimli, 2020-04-17:
你的写法和我的设计目标不一样，你想要实现什么效果？贴个图看看。

YinWenshuo, 2020-04-18:
> 你的写法和我的设计目标不一样，你想要实现什么效果？贴个图看看。

我的效果其实就是三个tab_button对应三个page，然后ta_button之间有竖线(就是我的xml里面的label)分割，示意图如下：（我们产品的图我这里没有，在公司，示意图应该也能表达我的意思）
[image]

xianjimli, 2020-04-21:
你更新代码试试。

commit 1bb27bf9 (2020-04-21): improve tab button(fix #389)
closed 2020-04-21 by xianjimli