zlgopen/awtk #390: scons执行的时候提示：Windows找不到文件'powershell.exe'

jampha, 2020-04-20:
有谁遇到过这样的问题，在win7 x64执行scons编译的时候，会弹出提示：Windows找不到文件'powershell.exe'
尝试过更新powershel到7.0.0版本也不行。如果有大佬解决了，望告知，谢谢！
CMD执行如下：
G:\~StPrj\awtk\awtk>scons
scons: Reading SConscript files ...
MACH=AMD64 ARCH=('64bit', 'WindowsPE') TARGET_ARCH=
TK_ROOT: G:\~StPrj\awtk\awtk
TypeError: write() argument must be str, not bytes:
  File "G:\~StPrj\awtk\awtk\SConstruct", line 16:
    OS_SUBSYSTEM_WINDOWS=awtk.OS_SUBSYSTEM_WINDOWS
  File "d:\developtool\python\python37\lib\site-packages\scons\SCons\Defaults.py", line 88:
    _default_env = SCons.Environment.Environment(*args, **kw)
  File "d:\developtool\python\python37\lib\site-packages\scons\SCons\Environment.py", line 990:
    apply_tools(self, tools, toolpath)
  File "d:\developtool\python\python37\lib\site-packages\scons\SCons\Environment.py", line 102:
    env.Tool(tool)
  File "d:\developtool\python\python37\lib\site-packages\scons\SCons\Environment.py", line 1810:
    tool(self)
  File "d:\developtool\python\python37\lib\site-packages\scons\SCons\Tool\__init__.py", line 303:
    self.generate(env, *args, **kw)
  File "d:\developtool\python\python37\lib\site-packages\scons\SCons\Tool\default.py", line 41:
    SCons.Tool.Tool(t)(env)
  File "d:\developtool\python\python37\lib\site-packages\scons\SCons\Tool\__init__.py", line 303:
    self.generate(env, *args, **kw)
  File "d:\developtool\python\python37\lib\site-packages\scons\SCons\Tool\mslink.py", line 314:
    msvc_setup_env_once(env)
  File "d:\developtool\python\python37\lib\site-packages\scons\SCons\Tool\MSCommon\vc.py", line 695:    
       msvc_setup_env(env)
  File "d:\developtool\python\python37\lib\site-packages\scons\SCons\Tool\MSCommon\vc.py", line 814:
    d = msvc_find_valid_batch_script(env,version)
  File "d:\developtool\python\python37\lib\site-packages\scons\SCons\Tool\MSCommon\vc.py", line 762:
    d = script_env(vc_script, args=arg)
  File "d:\developtool\python\python37\lib\site-packages\scons\SCons\Tool\MSCommon\vc.py", line 621:
    stdout = common.get_output(script, args)
  File "d:\developtool\python\python37\lib\site-packages\scons\SCons\Tool\MSCommon\common.py", line 238:
    sys.stderr.write(stderr)

xianjimli, 2020-04-20:
百度一下。

jampha, 2020-04-26:
在win10上就没问题，不知道为什么

soyzhc, 2020-07-28:
可以尝试把  C:\Windows\System32\WindowsPowerShell\v1.0   下的 powershell.exe 拷贝到 windows 目录下

closed 2020-10-15 by soyzhc