zlgopen/awtk #391: 关于rich_text控件的内容一页显示不下，翻页的问题

lizhenningi, 2020-04-22:
您好！
在使用rich_text控件时，rich_text所要显示的内容一页显示不下，我想要实现按“上下方向键”实现翻页或者显示内容上下移动一行的效果，这个需要如何设置？

xianjimli, 2020-04-23:
已经修改，你更新代码试试。

lizhenningi, 2020-04-23:
您好！
试过了，是我想要的效果，谢谢你

lizhenningi, 2020-04-24:
您好！
我在scroll_view中使用rich_text控件，想要实现rich_text控件内容滚动时scroll_bar也要有相应变化（也就是rich_text控件与scroll_bar控件结合使用），要达到这个效果需要如何设置？

xianjimli, 2020-04-24:
增加了rich_text_view控件，你参考这个：https://github.com/zlgopen/awtk/blob/master/demos/assets/default/raw/ui/rich_text_view.xml

lizhenningi, 2020-04-26:
您好！
增加的rich_text_view控件我已经应用了，但在使用的过程中发现了一些小问题，现描述如下：
在rich_text控件中显示的内容超出了当前页面能显示的范围（超出了大概0.9行），也就是说还有差不多一行没有显示出来（此时页面滚动条中不显示滚动块，我用的是scroll_bar_m），这时点击下键，页面可以滚动显示（此时页面滚动条未发生任何变化）。
[image]
[image]

xianjimli, 2020-04-27:
你把xml发给我试试。

lizhenningi, 2020-04-27:
我是在awtk-hello-master里面做的实验，这个页面的xml文件是window1.xml。window1.xml中的rich_text控件内容的显示是用函数控制的，在window1.c中实现。default.xml是window1.xml的style。
这三个文件都在“rich_text相关.zip”中
[rich_text相关.zip](https://github.com/zlgopen/awtk/files/4536965/rich_text.zip)

xianjimli, 2020-04-28:
已修改，你更新代码试试。谢谢

lizhenningi, 2020-04-30:
不客气，我试过了，是这个效果。谢谢你