zlgopen/awtk #393: 关于button控件的text超长显示问题

lizhenningi, 2020-04-27:
您好！
在使用button控件时text显示的文本超出了button控件范围，可以怎样设置一下，让这种显示text超出控件显示范围的超出的部分显示"..."
比如text=“1111122222”显示超长，显示在屏幕上的为“11111...”

xianjimli, 2020-04-27:
在button里放一个hscroll_label。

YinWenshuo, 2020-04-28:
我也试着用了一下hscroll_label，但发现有点问题，下面第一种方法是ok的，第二种方法不滚动显示
方法1、在xml里面写好text，就可以滚动，xml如下
  <hscroll_label
    lull="1000"
    loop="true"
    yoyo="true"
    ellipses="true"
    focusable="true"
    x="742" y="0" w="538" h="60"
    style="doors_open" text=“我是中国人我爱中国，我是江西人我爱江西，我是深圳人我爱深圳。”/> 
方法2、xml里面不给xml初始化，而是在代码里面动态设置
...
wchar_t tips[128];
...
（调用widget_set_visible(tipsName,TRUE,FALSE);widget_set_text(tipsName,tips);），就不会滚动了，看到的显示只是显示了一部分"我是中国人我爱中国，我是江西人我爱江"
 <hscroll_label name="tipsName"
    lull="1000"
    loop="true"
    yoyo="true"
    ellipses="true"
    focusable="true"
    x="742" y="0" w="538" h="60"
    style="doors_open" />

YinWenshuo, 2020-04-28:
@xianjimli

xianjimli, 2020-04-29:
我是了一下，可以啊：https://github.com/zlgopen/awtk-c-demos/blob/master/demos/hscroll_label.c