zlgopen/awtk #395: overlay和popup不响应 EVT_BLUR 事件

sqqdfny, 2020-04-28:
在做弹窗的时候，有个需求，需要捕获窗口失去焦点的事件，发现 overlay和popup不响应 EVT_BLUR 事件

xianjimli, 2020-04-28:
设置为focusable试试。

sqqdfny, 2020-04-29:
<overlay anim_hint="popup" self_layout="default(x=40,y=300,w=400,h=300)" style="default" focusable="true">
	<button name="btn_ok" x="center" y="240" w="200" h="50" text="" style="btn_ok"/>

    <label name="label_text" x="0" y="110" w="400" h="130" text="" style="showMessageBox"/>  

    <image name="image"　draw_type="center" image="icon_succ" x="150" y="10" w="100" h="100"/>
</overlay>

我按照上面的参数，把focusable设置为true, 没有效果， ovrelay 和 popup 都不响应 EVT_BLUR 事件。
其中 popup 类似于 dialog，把下层窗口的所有控件都不响应触摸屏了。
overlay窗口打开时，点下层窗口时，控件有响应，但overlay窗口不会隐藏，也不响应 EVT_BLUR 事件。

xianjimli, 2020-04-29:
那注册点击事件，如果点击在外面，就认为失去焦点。