zlgopen/awtk #396: 多窗口 应用软件 窗口管理 怎么做?

YinWenshuo, 2020-04-28:
我现在的做法跟awtk提供的例子一样，大概流程如下：
开机-->main函数-->application_init()-->默认打开窗口a：window_open("window_a");
然后，在window_a上面点击btn_a_b,可以打开窗口b（window_b）,
另外，在window_a上面还可以点击另外一个按钮btn_a_c,可以打开窗口c（window_c）,
目前这种做法，在打开了窗口window_c的时候，如果想在window_c切换到window_b，先得从window_c退到window_a,然后在window_a上点击btn_a_b来打开窗口window_b，
我的想法：开机起来时有个空窗口window_manager，然后可以默认打开window_a，也可以打开window_b，或者window_c，也就是说window_a，window_b，window_c是同一个级别的，而不是父窗口子窗口关系，这需要怎么做？能给个建议吗？

xianjimli, 2020-04-28:
window_b如果已经打开了，可以用widget_restack把它提到前面来。

YinWenshuo, 2020-04-29:
> window_b如果已经打开了，可以用widget_restack把它提到前面来。

这是要预先打开了window_b，也就是说在window_c的时候，window_b其实也是在后台运行的，只是不刷.我的想法是任何时候就只有一个window，window切换由空窗口window_manger来控制，但我不知道怎么实现。

xianjimli, 2020-04-29:
那每次打开之前，关闭当前窗口就好了。

closed 2020-04-30 by YinWenshuo