zlgopen/awtk #397: arm9上来回点击不同的button，button状态切换慢

bill611, 2020-04-29:
你好，我在arm板上跑发现这种现象，没有鼠标。有2个button，每个button设置了press和normal使用2种不同的图片显示不同的状态。
当只按同一个button时，图片状态刷新的很及时，
当来回按这2个button时，这2个button的状态刷新有卡顿，但是声音反馈是及时的，说明了事件已经及时响应了。
在ubuntu上跑是没问题，响应都很快。
我想问这来回按不同的button时，刷新图标有什么区别？没有找到最终调用button刷新的地方。
我有尝试打印这绘图的地方，但是发现触发了point事件，整个窗口所有空间都会被调用。
[image]

bill611, 2020-04-30:
我找到这个慢的点了，在 src/base/widget.c widget_set_focused_internal里，调用widget_set_as_key_target这个函数，我把这句屏蔽了，就很快了。
我看到widget_set_as_key_target这里面做了一个递归的实现，我想问，具体作用是怎样的？屏蔽了有什么影响？谢谢。

xianjimli, 2020-04-30:
widget\_set\_as\_key\_target会影响键盘操作。你可以把这个函数花的时间打印处理看看。

bill611, 2020-04-30:
我刚打印看了，这个函数本身执行的很快，us级别，屏蔽了widget_set_as_key_target里的widget_set_need_update_style，也能达到同样的及时响应的效果。
如果只屏蔽这句，是否是 获取焦点时，只是UI不刷新而已？