zlgopen/awtk #412: 有几个透明bug

FandelXiong, 2020-05-19:
# 环境 linux framebuffer brga888 后端agge
1. window窗体支持全透明，但不支持半透明，发现是pixel.h的pixel_bgra8888_blend_rgba_dark和pixel_bgra8888_blend_rgba_premulti 对alpha没有做混合处理
2. 如果现在显示一个背景全透明的window(window里有控件，能看到显示内容)，打开一个dialog,当dialog关闭后，现象是dialog窗体只是往底层移动了，并没有关闭，全透明的window显示在最上层后，能透过背景看到dilaog

WNsACE, 2020-05-21:
1，通过 pixel.h 看到，这两个函数应该是有和 alpha 通道做相乘的，而其中 pixel_bgra8888_blend_rgba_premulti 函数中的 rgba 参数中的 r，g，b 值是在传入函数前做了预乘 alpha 的操作。
`
static inline void pixel_bgra8888_blend_rgba_dark(void* pixel, uint8_t a) {
  uint8_t* p = (uint8_t*)pixel;

  if (p[3] > 0xf4) {
    p[0] = (p[0] * a) >> 8;
    p[1] = (p[1] * a) >> 8;
    p[2] = (p[2] * a) >> 8;
    p[3] = 0xff;
  } else {
    uint8_t s_a = 0xff - a;
    uint8_t out_a = tk_pixel_limit_uint8(p[3] + s_a - ((p[3] * s_a) >> 8));
    if (out_a > 0) {
      uint8_t d_a = (p[3] * (0xff - a)) >> 8;
      p[0] = (p[0] * d_a) / out_a;
      p[1] = (p[1] * d_a) / out_a;
      p[2] = (p[2] * d_a) / out_a;
    }
    p[3] = out_a;
  }
}

static inline void pixel_bgra8888_blend_rgba_premulti(void* pixel, rgba_t rgba) {
  uint8_t a = rgba.a;
  uint8_t* p = (uint8_t*)pixel;

  if (p[3] > 0xf4) {
    p[0] = ((p[0] * a) >> 8) + rgba.b;
    p[1] = ((p[1] * a) >> 8) + rgba.g;
    p[2] = ((p[2] * a) >> 8) + rgba.r;
    p[3] = 0xff;
  } else {
    uint8_t s_a = 0xff - a;
    uint8_t out_a = tk_pixel_limit_uint8(p[3] + s_a - ((p[3] * s_a) >> 8));
    if (out_a > 0) {
      uint8_t d_a = (p[3] * (0xff - rgba.a)) >> 8;
      p[0] = (p[0] * d_a + (rgba.b << 8)) / out_a;
      p[1] = (p[1] * d_a + (rgba.g << 8)) / out_a;
      p[2] = (p[2] * d_a + (rgba.r << 8)) / out_a;
    }
    p[3] = out_a;
  }
}
`
2，第二个问题应该是没有定义 WITH_LCD_CLEAR_ALPHA 宏，而你看到的底层的 dialog 窗体只是残影来的，而不是真实存在的，如果需要让窗体半透明或者全透明的话，需要开启 WITH_LCD_CLEAR_ALPHA 宏

FandelXiong, 2020-05-21:
嗯，第一个问题确实修复了，我当时看的时候，没有做混合；
第二个我尝试开启后，残影问题解决了，但背景半透明情况下，性能会大幅度降低，1080p分辨率，需要绘制很长时间；

WNsACE, 2020-05-21:
主要是两个问题导致速度变慢：
1，半透明的效果主要是每个像素点都进行混合运行（没有提供硬件加速的前提， WITH_G2D 宏就是开启硬件加速），而且当背景色为半透明的时候，混合运算的公式更加复杂了，所以增加每个像素点融合的运算量，我们可以看到 pixel_bgra8888_blend_rgba_premulti 函数中的 p[3] <= 0xf4 的时候就是背景色为半透明，可以看到算法变复杂了。
2，而且开了窗口半透效果后，每一帧都会在脏矩形区域上面刷新一层透明色来解决残影的问题，这样也会消耗性能，如果不开启脏矩形机制，就会整屏幕刷透明色，这样性能会比较低一点。

FandelXiong, 2020-05-21:
感觉脏矩形机制有些问题，我现在背景是半透明，上面有些一些按钮，现在点击按钮，会触发整个界面重绘制，导致性能很低，按照我的理解，应该只重绘变化的这个区域把

WNsACE, 2020-05-21:
因为不会吧，你把你的 xml 或者代码发出来看看？又或者你检查一下啊，你的使用三缓冲机制还是用双缓冲机制的，一般三缓冲机制都是不启用脏矩形机制的。

FandelXiong, 2020-05-21:
`<window name="home_page" style:normal:bg_color ="rgba(0,0,0,0.8)">
  <pages name="pages" x="1060" y="116" w="678" h="848" style:normal:round_radius="16" style:normal:bg_color="#2B3C5B" style:normal:border="none"/>
  <tab_control name="tab_control" x="1106" y="116" w="598" h="800" style:normal:border="none">
    <tab_button_group name="tab_button_group" x="0" y="0" w="100%" h="122" style:normal:border="bottom" style:normal:border_color="#1C233C" style:focused:font_size="32" style:focused:text_color="#00B1E5" style:normal:bg_color="#00b1e500" style:focused:border_width="1">
      <property name="style:focused:border">
        <![CDATA[bottom]]>
      </property>
      <tab_button name="tab_button" style="tab_button_style" tr_text="扫码" />
      <tab_button name="tab_button1" x="406" y="0" w="192" h="20" style="tab_button_style" tr_text="密码解锁" />
    </tab_button_group>
    <pages name="pages" x="0" y="122" w="100%" h="700" style:normal:border="none">
      <view name="view" x="0" y="0" w="100%" h="100%">
        <label name="password_label" x="0" y="18" w="84" h="100" style="password_label_style"/>
        <label name="password_label" x="102" y="18" w="84" h="100" selectable="false" clickable="false" style="password_label_style"/>
        <label name="password_label" x="204" y="18" w="84" h="100" selectable="false" clickable="false" style="password_label_style"/>
        <label name="password_label" x="307" y="18" w="84" h="100" selectable="false" clickable="false" style="password_label_style"/>
        <label name="password_label" x="409" y="18" w="84" h="100" selectable="false" clickable="false" style="password_label_style"/>
        <label name="password_label" x="512" y="18" w="84" h="100" selectable="false" clickable="false" style="password_label_style"/>
        <view name="view4" children_layout="default(c=3,r=4,x=0,y=0,s=16)" y="225" w="598" h="466" is_keyboard="true">
          <button name="key_number" repeat="0" enable_long_press="false" style="keyboard_button_style" tr_text="1" style:focused:text_color="#0095E3"/>
          <button name="key_number" repeat="0" enable_long_press="false" style="keyboard_button_style" tr_text="2"/>
          <button name="key_number" repeat="0" enable_long_press="false" style="keyboard_button_style" tr_text="3"/>
          <button name="key_number" repeat="0" enable_long_press="false" style="keyboard_button_style" tr_text="4"/>
          <button name="key_number" repeat="0" enable_long_press="false" style="keyboard_button_style" tr_text="5"/>
          <button name="key_number" repeat="0" enable_long_press="false" style="keyboard_button_style" tr_text="6"/>
          <button name="key_number" repeat="0" enable_long_press="false" style="keyboard_button_style" tr_text="7"/>
          <button name="key_number" repeat="0" enable_long_press="false" style="keyboard_button_style" tr_text="8"/>
          <button name="key_number" repeat="0" enable_long_press="false" style="keyboard_button_style" tr_text="9"/>
          <button name="key_del" repeat="0" enable_long_press="false" style="keyboard_button_style" tr_text="10"/>
          <button name="key_number" repeat="0" enable_long_press="false" style="keyboard_button_style" tr_text="0"/>
          <button name="key_confim" repeat="0" enable_long_press="false" style="keyboard_button_style" tr_text="11"/>
        </view>
      </view>
      <view name="view10" x="0" y="0" w="100%" h="100%" children_layout="default(r=0,c=1,s=20)">
          <qr name="qr_code" self_layout="default(x=20,y=20,w=482 h=482)" style:normal:bg_color="#FFFFFF" style:normal:margin="20" text="teststt" pix_size="1" border="none"/>
        <button name="show_popup" tr_text="点击" h="40" style="main_button"/>
        <button name="show_interact" tr_text="高亮" h="40" style="main_button"/>
      </view>
    </pages>
  </tab_control>
  <view name="view2" x="1193" y="234" w="107" h="6" style:normal:border="bottom" style:normal:round_radius="3" style:normal:border_width="0" style:normal:bg_color="#00B1E5"/>
  <view name="view3" x="1501" y="234" w="107" h="6" style:normal:border="bottom" style:normal:round_radius="3" style:normal:border_width="0" style:normal:bg_color="#00B1E5"/>
</window>`

这是整个xml，我使用的双buffer，我是在canvas_fill_rect函数这里打的断点，发现绘制区域是1080p

WNsACE, 2020-05-21:
canvas_fill_rect 函数只是给指定的区域填充颜色，但是在实际画到 buffer 上面的时候受到脏矩形的裁剪，所以你最好在 lcd_mem.inc 文件中的 lcd_mem_end_frame 函数中打印 lcd->dirty_rect 出来看看，你说的点击按钮会导致全屏刷新具体是指点击那个按钮？

FandelXiong, 2020-05-21:
我刚才在lcd_mem_end_frame上打了断点，胀矩形还是1080p,按钮就是数字键盘上的任意一个

WNsACE, 2020-05-21:
不要断点，一直用printf打印出来看吧，因为脏矩形会一直变的，你的意思是按钮指键盘上面的按钮？

FandelXiong, 2020-05-21:
打印出来也是一样的，我的xml有自定义的数字键盘，就是点击上面的按钮

WNsACE, 2020-05-21:
但是我看这个 xml 中也没有什么特别的地方，你用的是 awtk-linux-fb 的是适配层么？

WNsACE, 2020-05-21:
代码可以发出来么？

FandelXiong, 2020-05-21:
整个工程的代码吗

WNsACE, 2020-05-21:
又或者你可以试一下只通过 window_open 打开这个 xml，然后就不执行其他的代码，看一下是不是代码引起的

FandelXiong, 2020-05-21:
是的，这样胀区域，就是正确的，忘记了，原来的代码按钮按下时，有几个label也会变，但不至于整个屏幕的画面都变化;我把代码给你分析下
[AwtkApplication1.tar.gz](https://github.com/zlgopen/awtk/files/4661586/AwtkApplication1.tar.gz)

WNsACE, 2020-05-21:
我这边编译不了项目，你可以删除一些代码试试看，脏矩形机制是是把所有脏的区域合并在一起的，比如说左上角脏了一个像素点和右下角脏了一个像素点，合并在一起，脏矩形区域就是整个屏幕了

YinWenshuo, 2020-05-22:
我使用的是双buffer，没有开启脏矩形算法，没有发现这个问题。@FandelXiong
另外，请教个问题@WNsACE，
你说的硬件加速打开宏WITH_G2D，我不清楚我的硬件可不可以硬件加速，那么，我自行定义WITH_G2D宏，会有问题吗？或者说怎么知道我的硬件是不是支持硬件加速？