zlgopen/awtk #413: 调用widget_focus_next导致程序挂掉

huangxunyan, 2020-05-23:
我在一个窗口中添加了三个控件，想通过widget_focus_prev和widget_focus_next在它们之间来回切换，调用widget_focus_prev时功能正常，可以聚焦到前一个控件，但是调用widget_focus_next时会导致程序挂掉，报错信息：
Exception thrown at 0x00007FF690795D69 in demo.exe: 0xC0000005: Access violation reading location 0xFFFFFFFFFFFFFFFF.
The program '[7100] demo.exe' has exited with code 0 (0x0).
[image]

xianjimli, 2020-05-23:
正常情况应该不会有问题，把可以运行的代码贴上来看看。

huangxunyan, 2020-05-23:
`#include "awtk.h"

static ret_t on_key_press(void* ctx, event_t* e) {
    // widget_t* win = WIDGET(e->target);
    widget_t* win = WIDGET(ctx);
    key_event_t* evt = key_event_cast(e);

    if (win != NULL)
    {
      if (evt->key == TK_KEY_a)
      {
        widget_focus_prev(win);
      }
      else if (evt->key == TK_KEY_d)
      {
        widget_focus_next(win);
      }
    }
    
    return RET_OK;
}

static ret_t sensitive_widget(void* ctx, const void* iter) {
    
    widget_t* widget = WIDGET(iter);
    widget_t* parent_widget = WIDGET(ctx);
    (void)ctx;

    if (tk_str_eq(widget->name, parent_widget->name))
    {

    }
    else
    {
      widget_set_sensitive(widget, TRUE);
    }

    return RET_OK;
}
static ret_t insensitive_widget(void* ctx, const void* iter) {
    
    widget_t* widget = WIDGET(iter);
    widget_t* parent_widget = WIDGET(ctx);
    (void)ctx;

    if (tk_str_eq(widget->name, parent_widget->name))
    {

    }
    else
    {
      widget_set_sensitive(widget, FALSE);
    }

    return RET_OK;
}
static ret_t on_eq_focus(void* ctx, event_t* e)
{
    widget_t* win = WIDGET(ctx);
    widget_t* target = WIDGET(e->target);
    value_t v;
    (void)e;

    widget_foreach(target, sensitive_widget, target);

    return RET_OK;
}
static ret_t on_eq_blur(void* ctx, event_t* e)
{
    widget_t* win = WIDGET(ctx);
    widget_t* target = WIDGET(e->target);
    value_t v;
    (void)e;

    widget_foreach(target, insensitive_widget, target);

    return RET_OK;
}

/**
 * 子控件初始化(主要是设置click回调、初始显示信息)
 */
static ret_t init_widget(void* ctx, const void* iter) {
    widget_t* widget = WIDGET(iter);
    widget_t* win = widget_get_window(widget);
    (void)ctx;

    if (widget->name != NULL)
    {
        const char* name = widget->name;

        if (strstr(name, "eq_1") != NULL)
        {
            widget_on(widget, EVT_FOCUS, on_eq_focus, win);
            widget_on(widget, EVT_BLUR, on_eq_blur, win);
            widget_on(widget, EVT_KEY_DOWN, on_key_press, widget);
            widget_set_prop_bool(widget, WIDGET_PROP_FOCUSED, TRUE);
        }
        else if (strstr(name, "eq_2") != NULL)
        {
            widget_on(widget, EVT_FOCUS, on_eq_focus, win);
            widget_on(widget, EVT_BLUR, on_eq_blur, win);
            widget_on(widget, EVT_KEY_DOWN, on_key_press, widget);
        }
        else if (strstr(name, "eq_3") != NULL)
        {
            widget_on(widget, EVT_FOCUS, on_eq_focus, win);
            widget_on(widget, EVT_BLUR, on_eq_blur, win);
            widget_on(widget, EVT_KEY_DOWN, on_key_press, widget);
        }
    }

    return RET_OK;
}

/**
 * 初始化窗口上的子控件
 */
static void init_children_widget(widget_t* widget) 
{
  widget_foreach(widget, init_widget, widget);
}

/**
 * 打开equalizer
 */
ret_t open_eq_window(void) 
{
  widget_t* win = window_open("win_eq/win_eq");
  if (win) 
  {
    init_children_widget(win);

    return RET_OK;
  }

  return RET_FAIL;
}
`
按下a键聚焦到前一个控件，按下d键聚焦到后一个控件

xianjimli, 2020-05-23:
把xml也贴上来吧

xianjimli, 2020-05-23:
用指定键切换焦点，在xml中指定就可以了：

```xml
<window anim_hint="htranslate" move_focus_prev_key="a" move_focus_next_key="d" text="Basic Controls">
```

huangxunyan, 2020-05-23:
@xianjimli 
`<window name="win_eq">
  <label name="label" x="38" y="24" w="41" h="21" tr_text="+12dB"/>
  <label name="label1" x="38" y="170" w="41" h="21" tr_text="0dB"/>
  <label name="label2" x="38" y="316" w="41" h="21" tr_text="-12dB"/>
  <label name="label_gain" x="49" y="337" w="30" h="19" tr_text="Gain"/>
  <label name="label_q" x="49" y="366" w="30" h="19" tr_text="Q"/>
  <label name="label_freq" x="49" y="395" w="30" h="19" tr_text="Freq"/>
  <list_item name="eq_1" x="101" y="10" w="38" h="418" style="eq_list" focusable="true">
    <slider name="gain_slider" x="9" y="10" w="20" h="318" bar_size="0" dragger_size="10" vertical="true" style:normal:fg_color="#D8D8D8" focusable="true" sensitive="false">
      <property name="value">
        <![CDATA[40]]>
      </property>
    </slider>
    <edit name="gain_value" x="5" y="386" w="28" h="18" password_visible="false" auto_fix="true" tips="" step="0.1" input_type="float" focusable="true">
      <property name="sensitive">
        <![CDATA[false]]>
      </property>
      <property name="min">
        <![CDATA[-12]]>
      </property>
      <property name="max">
        <![CDATA[12]]>
      </property>
    </edit>
    <edit name="q_value" x="5" y="357" w="28" h="18" password_visible="false" auto_fix="true" tips="" step="0.1" input_type="float" focusable="true">
      <property name="sensitive">
        <![CDATA[false]]>
      </property>
      <property name="min">
        <![CDATA[0]]>
      </property>
      <property name="max">
        <![CDATA[20]]>
      </property>
    </edit>
    <edit name="freq_value" x="5" y="328" w="28" h="18" password_visible="false" auto_fix="true" tips="" step="1" input_type="uint" focusable="true">
      <property name="sensitive">
        <![CDATA[false]]>
      </property>
      <property name="min">
        <![CDATA[24]]>
      </property>
      <property name="max">
        <![CDATA[24000]]>
      </property>
    </edit>
  </list_item>
  <list_item name="eq_" x="229" y="10" w="38" h="418" style="eq_list" focusable="true">
    <slider name="gain_slider1" x="9" y="10" w="20" h="318" bar_size="0" dragger_size="10" vertical="true" style:normal:fg_color="#D8D8D8" focusable="true" sensitive="false">
      <property name="value">
        <![CDATA[40]]>
      </property>
    </slider>
    <edit name="gain_value1" x="5" y="386" w="28" h="18" password_visible="false" auto_fix="true" tips="" step="0.1" input_type="float" focusable="true">
      <property name="sensitive">
        <![CDATA[false]]>
      </property>
      <property name="min">
        <![CDATA[-12]]>
      </property>
      <property name="max">
        <![CDATA[12]]>
      </property>
    </edit>
    <edit name="q_value1" x="5" y="357" w="28" h="18" password_visible="false" auto_fix="true" tips="" step="0.1" input_type="float" focusable="true">
      <property name="sensitive">
        <![CDATA[false]]>
      </property>
      <property name="min">
        <![CDATA[0]]>
      </property>
      <property name="max">
        <![CDATA[20]]>
      </property>
    </edit>
    <edit name="freq_value1" x="5" y="328" w="28" h="18" password_visible="false" auto_fix="true" tips="" step="1" input_type="uint" focusable="true">
      <property name="sensitive">
        <![CDATA[false]]>
      </property>
      <property name="min">
        <![CDATA[24]]>
      </property>
      <property name="max">
        <![CDATA[24000]]>
      </property>
    </edit>
  </list_item>
  <list_item name="eq_3" x="357" y="10" w="38" h="418" style="eq_list" focusable="true">
    <slider name="gain_slider3" x="9" y="10" w="20" h="318" bar_size="0" dragger_size="10" vertical="true" style:normal:fg_color="#D8D8D8" focusable="true" sensitive="false">
      <property name="value">
        <![CDATA[40]]>
      </property>
    </slider>
    <edit name="gain_value3" x="5" y="386" w="28" h="18" password_visible="false" auto_fix="true" tips="" step="0.1" input_type="float" focusable="true">
      <property name="sensitive">
        <![CDATA[false]]>
      </property>
      <property name="min">
        <![CDATA[-12]]>
      </property>
      <property name="max">
        <![CDATA[12]]>
      </property>
    </edit>
    <edit name="q_value3" x="5" y="357" w="28" h="18" password_visible="false" auto_fix="true" tips="" step="0.1" input_type="float" focusable="true">
      <property name="sensitive">
        <![CDATA[false]]>
      </property>
      <property name="min">
        <![CDATA[0]]>
      </property>
      <property name="max">
        <![CDATA[20]]>
      </property>
    </edit>
    <edit name="freq_value3" x="5" y="328" w="28" h="18" password_visible="false" auto_fix="true" tips="" step="1" input_type="uint" focusable="true">
      <property name="sensitive">
        <![CDATA[false]]>
      </property>
      <property name="min">
        <![CDATA[24]]>
      </property>
      <property name="max">
        <![CDATA[24000]]>
      </property>
    </edit>
  </list_item>
</window>`
目前我是通过AWTK Designer设计的界面，上面似乎没有那么灵活的设置。我这边的设计思路是界面通过AWTK Designer设计，逻辑功能通过调用API实现，请问按照这个思路目前来说可行吗？

xianjimli, 2020-05-23:
思路没问题。

你在代码里设置窗口的属性也是一样：

```c
widget_set_prop_str(win, WIDGET_PROP_MOVE_FOCUS_PREV_KEY, "a");
widget_set_prop_str(win, WIDGET_PROP_MOVE_FOCUS_NEXT_KEY, "d");
```

xianjimli, 2020-05-26:
widget_dispatch里没有target，可能导致崩溃，我改了，你更新代码试试。

huangxunyan, 2020-05-26:
确实是因为没有target导致的，现在不会挂掉了，感谢！！

closed 2020-05-26 by huangxunyan