zlgopen/awtk #414: 关于克隆后界面刷新的问题

gaoyuan1111, 2020-05-25:
我在界面打开后，每次按键会克隆20个button。这时我遇到的问题是，我移动到最低端的时候的时候，再按下键，用widget_invalidate刷新页面，焦点下移了，但是页面并没有反到第二页。请问有什么方法可以刷新正在不停克隆button的scroll_view呢？并且保证当前选中不会被刷新。 之前我尝试过重新设置焦点，但是问题是我当前选项在页面中间的button上，重新设置焦点，会把当前选中的button放到scroll_view的最底端。

xianjimli, 2020-05-26:
把可以运行的代码贴上来看看。

gaoyuan1111, 2020-05-26:
widget_t* widget= widget_lookup(win,"all_contacts_btn_list_test",TRUE);  
    uint64_t start = time_now_ms();
    uint64_t cost;

    int i=0;
    static int isfocued = 0;
    value_t v;
    char widget_name[128]="";
    value_set_bool(&v, 1);
    widget_t* newwd;
    for(i=a*UP_PHONEBOOK_USR_NUM;i<(a+1)*UP_PHONEBOOK_USR_NUM;i++){
        
        if(PhonebookData[i].present != 0 ){
            newwd = widget_clone(widget, widget->parent);

            widget_set_prop(newwd, WIDGET_PROP_VISIBLE, &v);
            widget_set_prop(newwd, WIDGET_PROP_ENABLE, &v);
            button_set_enable_long_press(newwd, FALSE);
            widget_off_by_func(newwd, EVT_LONG_PRESS, open_phonebook_option_window, NULL);
            widget_off_by_func(newwd, EVT_CLICK, on_dial, newwd);
            widget_off_by_func(newwd->children->elms[0], EVT_FOCUS, on_show_button, newwd);
            widget_off_by_func(newwd->children->elms[0], EVT_BLUR, on_hide_button, newwd);
            widget_off_by_func(newwd->children->elms[1], EVT_FOCUS, on_show_button, newwd);
            widget_off_by_func(newwd->children->elms[1], EVT_BLUR, on_hide_button, newwd);

            button_set_enable_long_press(newwd, TRUE);
            widget_on(newwd, EVT_LONG_PRESS, open_phonebook_option_window, NULL);
            widget_on(newwd, EVT_CLICK, on_dial, newwd);
            widget_on(newwd->children->elms[0], EVT_FOCUS, on_show_button, newwd);
            widget_on(newwd->children->elms[0], EVT_BLUR, on_hide_button, newwd);
            widget_on(newwd->children->elms[1], EVT_FOCUS, on_show_button, newwd);
            widget_on(newwd->children->elms[1], EVT_BLUR, on_hide_button, newwd);
            widget_set_text_utf8(WIDGET(newwd->children->elms[0]), PhonebookData[i].name);
            if(strlen(PhonebookData[i].phonenumber) > 0)
            widget_set_tr_text(WIDGET(newwd->children->elms[1]), PhonebookData[i].phonenumber);
            else if(strlen(PhonebookData[i].mobilenumber) > 0)
            widget_set_tr_text(WIDGET(newwd->children->elms[1]), PhonebookData[i].mobilenumber);
            else if(strlen(PhonebookData[i].othernumber) > 0)
            widget_set_tr_text(WIDGET(newwd->children->elms[1]), PhonebookData[i].othernumber);
            else
            widget_set_tr_text(WIDGET(newwd->children->elms[1]), "None");

            set_hscroll_label_style(newwd->children->elms[0]);
            widget_update_style(newwd);
            
        } 
    }
widget_layout_children(widget->parent);
     widget_invalidate(widget_get_window(widget), NULL);
逻辑是打开界面clone20个button。 之后每次点击下键，触发EVT_KEY_DOWN时间，执行上面代码进行clone。 但是切换第二个页面后，就在没有刷新成功。 猜测是，先执行了下键的widget_focus_next操作，焦点设置成功页面刷新，然后才执行上面的widget_invalidate，把界面刷回了scroll_view的第一行。

tangMans, 2020-05-27:
你想要的效果是：
在list_view上点击list_item，然后克隆20个button，然后scroll_view就滚动到界面的最低端，并且保持被点击的list_item的状态不变吗？

如果是，那么假如20个button刚满一页，即使“被点击的list_item的状态不变”,那么当你返回上一页的时候，你要点击界面，也会失去焦点