zlgopen/awtk #415: 怎么切换键盘UI？

ytchw101, 2020-05-26:
键盘默认使用ui资源里的kb_defalut 我现在修改后再生成一个键盘UI资源  如kb_xxx     这两个资源程序中要实现可切换   有什么函数能指定编辑控件使用某个键盘吗？

xianjimli, 2020-05-26:
参考这个文档：https://github.com/zlgopen/awtk/blob/master/docs/manual/edit_t.md#edit_t_keyboard

ytchw101, 2020-05-27:
你好 更新源码会出现一堆错误 甚至有的文件都找不到  比如更新后提示找不到input_engine_create 这个函数  在input_engine.h中有定义但input_engine.c中没有实现 等等问题   所以 我想不更新源码的话 在之前基础上   怎么实现 两个键盘 资源动态切换？   目前我用的源码是2020-01-03时更新的 目前较稳定

xianjimli, 2020-05-27:
1.4.1比较稳定，改了一些内存问题。最好升级一下，有问题可以发给我看看。

ytchw101, 2020-05-27:
更新到最新源码后，为什么中文输入法键盘 点击输入法键盘没字出来？ (我工程所使用的资源目前还没更新 因为我scons编译awtk源码老是出错， 所以用旧的资源文件  目前界面显示图片 文字都正常)

xianjimli, 2020-05-27:
资源要更新。拼音数据放到文件系统了。

ytchw101, 2020-05-27:
好吧  我得把编译AWTK搞好

xianjimli, 2020-05-27:
很抱歉，有问题尽管找我。

ytchw101, 2020-05-28:
assets_manager_add(rm, data_gpinyin_dat);
assets_manager_add(rm, data_suggest_words_zh_cn_dat);  

另外 开启 #define WITH_IME_PINYIN  1 //使用拼音输入法

资源都加载了  不知为何中文输入  还是不出来字呢？

xianjimli, 2020-05-28:
是什么平台，有文件系统吗？其它中文文字能显示出来吗？

ytchw101, 2020-05-28:
嵌入式+LCD 480*272  界面其他文字 图片等都正常 现在就是 输入法键盘出来后 中文输入没字出来 英文大小写都可以   不知道哪地方还需要添加啥吗？

xianjimli, 2020-05-28:
按键的时候，看看input_engine_pinyin_search有没有被调用，是否查找到数据

ytchw101, 2020-05-28:
按键时  
没有被调用

ytchw101, 2020-05-28:
rt_kprintf("im=%d\r\n", im);
  im = input_method_default_create();
 rt_kprintf("input_method_default_create...im=%d\r\n", im);
返回打印：
im=0
input_engine_create...
input_method_default_create...im=-2134900268

为什么按键input_engine_pinyin_search没被调用  不知道啥问题？

xianjimli, 2020-05-28:
看看input_engine_pinyin.cpp:input_engine_create有调用吗？

ytchw101, 2020-05-28:
找到问题  input_engine_create中 data = assets_manager_ref(assets_manager(), ASSET_TYPE_DATA, "gpinyin.dat");
data返回0

如下包含资源有错吗？ 我这个gpinyin_dat.data资源直接从最新源码的demeos中拷贝来的  编译资源文件其他资源都重新编译的 但 这个data文件夹下的资源怎么编译
#include "assets/inc/data/gpinyin_dat.data"
#include "assets/inc/data/suggest_words_zh_cn_dat.data"
。。。。。
 assets_manager_add(rm, data_gpinyin_dat);
assets_manager_add(rm, data_suggest_words_zh_cn_dat);

xianjimli, 2020-05-28:
update_res.py会处理data目录下的文件。assets/default/inc/data/gpinyin_dat.data文件有生成吗吗？

ytchw101, 2020-05-28:
是指编译awtk源码的时候 会生成gpinyin_dat.data 文件吗？

我现在用的是源码里面demeos/default/assets/inc/data下的gpinyin_dat.dat文件

重新编译我以前是用如下方式生成资源
[image]

我现在改成 -data 未重新生成gpinyin_dat.dat

xianjimli, 2020-05-28:
拷贝过去应该也是可以的。那调一下，看看assets_manager_ref为什么没加载到资源。

ytchw101, 2020-05-28:
好吧  谢谢 幸苦

ytchw101, 2020-05-29:
最终是下面的函数返回 0
asset_info_t* asset_loader_load(asset_loader_t* loader, uint16_t type, uint16_t subtype,
                                const char* path, const char* name) {
																	
	if(ASSET_TYPE_DATA == type)
	{
		 rt_kprintf("path=%s\r\n", path);
	}
  return_value_if_fail(path != NULL && name != NULL, NULL);
  return_value_if_fail(loader != NULL && loader->vt != NULL && loader->vt->load != NULL, NULL);

  return loader->vt->load(loader, type, subtype, path, name);
}

path打印为：path=/SD/assets/default/raw/data/gpinyin.dat
我SD卡这个目录下 是没有放gpinyin.dat文件的

这个gpinyin.dat资源不是编译成 gpinyin_dat.data资源文件  然后程序中包含就行了吗 如下？
#include "assets/inc/data/gpinyin_dat.data"
。。。。
  assets_manager_add(rm, data_gpinyin_dat);

xianjimli, 2020-05-29:
我知道原因了，初始化输入法时资源还没有加载。你更新代码试试。谢谢

ytchw101, 2020-06-02:
可以

closed 2020-07-03 by ytchw101