zlgopen/awtk #419: 对 “如何用 valgrind 查找内存相关问题”  文档的一点疑问

YinWenshuo, 2020-05-28:
我的理解是：为了要使用valgrind工具来排查自己的工程是不是存在内存泄漏、野指针等方面的问题，需要在字符模式下启动ubuntu，然后运行自己的awtk程序。例如，我的工程目录如下：
awtk awtk-demo awtk-linux-fb（这三个在同一级目录）
我按照这个文档https://github.com/zlgopen/awtk-linux-fb/blob/master/README.UbuntuVM.md，修改了下面步骤：

##### 3. 编译 demo 并抽取资源文件
确保 awtk 和 awtk-linux-fb 在同一级目录

```
cd awtk-linux-fb
scons 

报错：
tk/3rd/gtest/googletest/include awtk-port/fb_test.c
gcc: warning: '-mcpu=' is deprecated; use '-mtune=' or '-march=' instead
awtk-port/fb_test.c:1:0: error: bad value (cortex-a15) for -mtune= switch
 
 ^
scons: *** [awtk-port/fb_test.o] Error 1
scons: building terminated because of errors.

xianjimli, 2020-05-29:
好像是工具链没选对。

xianjimli, 2020-05-29:
```
看下 awtk_config.py
#for pc build
TOOLS_PREFIX=''
TSLIB_LIB_DIR=''
```

YinWenshuo, 2020-06-01:
1、这步编译指令没错吧
     cd awtk-linux-fb
     scons 

2、我的工具链是按照你的说明配的，即：
#for pc build
TOOLS_PREFIX=''
TSLIB_LIB_DIR=''

我awtk-linux-fb目录下的awtk_config.py如附件，麻烦帮我看看。

YinWenshuo, 2020-06-01:
附件上传不了，直接贴上来

import os
import platform

OS_NAME = platform.system()

def joinPath(root, subdir):
  return os.path.normpath(os.path.join(root, subdir))

CWD=os.path.normpath(os.path.abspath(os.path.dirname(__file__)));

TK_LINUX_FB_ROOT = CWD
TK_ROOT          = joinPath(TK_LINUX_FB_ROOT, '../awtk')
TK_SRC           = joinPath(TK_ROOT, 'src')
TK_3RD_ROOT      = joinPath(TK_ROOT, '3rd')
GTEST_ROOT       = joinPath(TK_ROOT, '3rd/gtest/googletest')

BUILD_DIR        = joinPath(TK_LINUX_FB_ROOT, 'build')
BIN_DIR          = joinPath(BUILD_DIR, 'bin')
LIB_DIR          = joinPath(BUILD_DIR, 'lib')
VAR_DIR          = joinPath(BUILD_DIR, 'var')
TK_DEMO_ROOT     = joinPath(TK_ROOT, 'demos')

LCD='LINUX_FB'
INPUT_ENGINE='pinyin'
NANOVG_BACKEND='AGGE'

COMMON_CCFLAGS=' -DHAS_STD_MALLOC -DHAS_STDIO -DWITH_VGCANVAS -DWITH_UNICODE_BREAK -DLINUX'
COMMON_CCFLAGS=COMMON_CCFLAGS+' -DWITH_ASSET_LOADER -DWITH_FS_RES ' 
COMMON_CCFLAGS=COMMON_CCFLAGS+' -DSTBTT_STATIC -DSTB_IMAGE_STATIC -DWITH_STB_IMAGE -DWITH_STB_FONT '
COMMON_CCFLAGS=COMMON_CCFLAGS+' -DWITH_NANOVG_AGGE -DWITH_WIDGET_TYPE_CHECK -DWITH_LCD_CLEAR_ALPHA'

if INPUT_ENGINE == 't9':
    COMMON_CCFLAGS = COMMON_CCFLAGS + ' -DWITH_IME_T9 '
elif INPUT_ENGINE == 't9ext' :
    COMMON_CCFLAGS = COMMON_CCFLAGS + ' -DWITH_IME_T9EXT'
elif INPUT_ENGINE == 'pinyin' :
    COMMON_CCFLAGS = COMMON_CCFLAGS + ' -DWITH_IME_PINYIN '
elif INPUT_ENGINE == 'spinyin' :
    COMMON_CCFLAGS = COMMON_CCFLAGS + ' -DWITH_IME_SPINYIN '
elif INPUT_ENGINE == 'null' :
    COMMON_CCFLAGS = COMMON_CCFLAGS + ' -DWITH_IME_NULL '

GRAPHIC_BUFFER='default'
#GRAPHIC_BUFFER='jzgpu'
#if GRAPHIC_BUFFER == 'jzgpu':
#  COMMON_CCFLAGS=COMMON_CCFLAGS+' -DWITH_JZGPU'

#only for c compiler flags
COMMON_CFLAGS=''
COMMON_CFLAGS=COMMON_CFLAGS+' -std=gnu99 -rdynamic -funwind-tables -ffunction-sections'

OS_LIBS=[]
OS_LIBPATH=[]
OS_CPPPATH=[]
OS_LINKFLAGS=''
OS_SUBSYSTEM_CONSOLE=''
OS_SUBSYSTEM_WINDOWS=''
OS_FLAGS='-g -Wall -Os -mcpu=cortex-a15 -mfpu=neon -mfloat-abi=hard '
#OS_FLAGS='-g -Wall -Os -mfloat-abi=hard '

#for build tslib
#TSLIB_INC_DIR=joinPath(TK_LINUX_FB_ROOT, '3rd/tslib/src')
#TSLIB_LIB_DIR=joinPath(TK_LINUX_FB_ROOT, '3rd/tslib/src/.libs')

#for prebuild tslib
#TSLIB_LIB_DIR='/opt/28x/tslib/lib'
#TSLIB_INC_DIR='/opt/28x/tslib/include'
#TOOLS_PREFIX='/opt/28x/gcc-4.4.4-glibc-2.11.1-multilib-1.0/arm-fsl-linux-gnueabi/bin/arm-linux-'
[... 100 lines cut ...]
  LIBS = ['eventhubproxy'] + LIBS
  LIBPATH = [EVENTHUB_PROXY_LIB_DIR] + LIBPATH;
  CPPPATH = [EVENTHUB_PROXY_INC_DIR] + CPPPATH;

os.environ['LCD'] = LCD
os.environ['TARGET_ARCH'] = 'arm'
os.environ['BIN_DIR'] = BIN_DIR;
os.environ['LIB_DIR'] = LIB_DIR;
os.environ['TK_ROOT'] = TK_ROOT;
os.environ['CCFLAGS'] = CCFLAGS;
os.environ['VGCANVAS'] = 'NANOVG'
os.environ['INPUT_ENGINE'] = INPUT_ENGINE;
os.environ['TSLIB_LIB_DIR'] = TSLIB_LIB_DIR;
os.environ['NANOVG_BACKEND'] = NANOVG_BACKEND;
os.environ['TK_3RD_ROOT'] = TK_3RD_ROOT;
os.environ['GTEST_ROOT'] = GTEST_ROOT;
os.environ['NATIVE_WINDOW'] = 'raw';
os.environ['GRAPHIC_BUFFER'] = GRAPHIC_BUFFER;

os.environ['OS_WHOLE_ARCHIVE'] = OS_WHOLE_ARCHIVE;
os.environ['AWTK_DLL_DEPS_LIBS'] = ';'.join(AWTK_DLL_DEPS_LIBS)
os.environ['STATIC_LIBS'] = ';'.join(STATIC_LIBS)

def has_custom_cc():
    return True

YinWenshuo, 2020-06-03:
@xianjimli 李老师，这个有空帮我看下，我偶尔遇到死机问题，还是要靠valgrind排查一遍。多谢！

xianjimli, 2020-06-03:
缺省的awtk_config.py放开下面这两行就行了，在虚拟机模拟运行，不要加硬件相关的那些参数。

```
#for pc build
TOOLS_PREFIX=''
TSLIB_LIB_DIR=''
```

YinWenshuo, 2020-06-04:
可以编译过了，awtk_config.py里面把所有跟arm板子相关的东西关掉，且
#for pc build
TOOLS_PREFIX=''
TSLIB_LIB_DIR=''
然后
scons APP=../awtk-demo/

再然后valgrind --leak-check=full ./build/bin/demo

我仍然有个疑问：
大费周章关掉这么多配置，在awtk-linux-fb下面编译，我感觉这样编译后的应用软件其实跟我直接在awtk-demo下面scons编译一样啊。

xianjimli, 2020-06-04:
valgrind不支持SDL。

closed 2020-06-04 by YinWenshuo