zlgopen/awtk #423: scroll_view中的滑动

gaoyuan1111, 2020-06-11:
请问一下。我在scroll_view中滑动button后。有没有什么事件能让我知道现在滑动结束了？有没有什么方法能控制滑动的距离或者速度？

xianjimli, 2020-06-11:
可以重载这两个函数：
```c
142   scroll_view_on_scroll_t on_scroll;
143   scroll_view_on_scroll_to_t on_scroll_to;
```

你要实现什么效果？

gaoyuan1111, 2020-06-12:
我想通过鼠标点击scroll_view中的button，然后向上划一下。就像手机向翻页这种。 但是呢我想在每次滑动结束后，列表不滚动后，通过注册事件的回调，执行一个函数，处理一些信息。 同时呢我也想控制一下这个滑动的数量，比如我想一次最多滑20个button。

gaoyuan1111, 2020-06-12:
我们要做一个列表，里面预计有500项，为了提高加载速度，列表初始化的时候加载了100项，想每次滑动上翻20项，滑动结束后新加载20项（滑动结束在加载防止影响滑动流畅度）。
我想找到滑动结束的事件来执行加载动作。同时想调整每次滑动的总条目，以及滑动动画每帧更新的距离。

xianjimli, 2020-06-12:
这个有点麻烦，周末我想想。

xianjimli, 2020-06-14:
加了3个事件：

```c
147 /**
148  * @event {event_t} EVT_SCROLL_START
149  * 开始滚动事件。
150  */
151 
152 /**
153  * @event {event_t} EVT_SCROLL_END
154  * 结束滚动事件。
155  */
156 
157 /**
158  * @event {event_t} EVT_SCROLL
159  * 滚动事件。
160  */
```
用法请参考：https://github.com/zlgopen/awtk-c-demos/blob/master/demos/list_item_clone.c