zlgopen/awtk #429: 问题咨询，demoui和AWTK designer显示异常

qwegary, 2020-06-19:
[image]
[image]
这个是怎么回事，请求帮助

xianjimli, 2020-06-19:
可能是opengl驱动兼容性问题。

xianjimli, 2020-06-19:
你可以修改awtk_config.py，开启AGGE试试。

qwegary, 2020-06-19:
开启AGGE，编译出来的demoui已经可以正常显示。但是AWTK designer显示还是异常，这个还有什么可能。

soyzhc, 2020-06-19:
你的电脑配置是什么，显卡型号发上来看看，可能是显卡驱动兼容性问题，升级显卡驱动看看，目前designer只支持opengles模式

qwegary, 2020-06-22:
我这边使用的是vmware虚拟机运行。
[image]

soyzhc, 2020-06-22:
vmtools有安装吗，我们这边用的是vmware player 12，还没有遇到过类似的情况

qwegary, 2020-06-23:
我这边使用的是
[image]

qwegary, 2020-06-23:
vmtools有安装的

qwegary, 2020-06-23:
把vmtools删除，然后重启打开，直接运行designer后，designer直接闪退

qwegary, 2020-06-23:
demoui运行正常

soyzhc, 2020-06-24:
我们这边的虚拟机没有出现类似的问题，只能猜测还是显卡驱动的问题
你尝试一下OpenGLES模式下的demoui是否能正常显示（把awtk_config.py中下面这句话注释了，再重新scons编译awtk）
```
#NANOVG_BACKEND='AGGE'
```
是否尝试过host主机里运行designer？
可以把你的host主机的设备管理器里面的显示适配器发上来看看，或者我们推荐的vmware版本

qwegary, 2020-06-24:
demoui是可以正常使用的
designer在主机上运行也是直接闪退的。
设备管理器如下图：
[image]

qwegary, 2020-06-24:
我重新安装主机的显卡驱动，出现下面状况
[image]

soyzhc, 2020-06-24:
到designer安装目录，运行一个gl-checker测试命令行工具，把输出结果发上来看看
```
C:\AWTK\Tools\gl-checker.exe
```
你的cpu是哪个型号？也发上来看看

qwegary, 2020-06-24:
[image]
测试工具gl-checker，打开闪一下就自动没了

soyzhc, 2020-06-24:
需要先打开cmd命令行，再手动敲命令运行
```
c:
cd C:\AWTK\Tools
gl-checker.exe
或者输出到日志文件
gl-checker.exe > log.txt
```

qwegary, 2020-06-24:
信息如下：
true
GL Vendor     : VMware, Inc.
GL Renderer   : Gallium 0.4 on SVGA3D; build: RELEASE;  LLVM;
GL Version    : 3.0 Mesa 11.2.0 (git-b9d3786)
GL Extensions : GL_ARB_multisample GL_EXT_abgr GL_EXT_bgra GL_EXT_blend_color GL_EXT_blend_minmax GL_EXT_blend_subtract GL_EXT_copy_texture GL_EXT_polygon_offset GL_EXT_subtexture GL_EXT_texture_object GL_EXT_vertex_array GL_EXT_compiled_vertex_array GL_EXT_texture GL_EXT_texture3D GL_IBM_rasterpos_clip GL_ARB_point_parameters GL_EXT_draw_range_elements GL_EXT_packed_pixels GL_EXT_point_parameters GL_EXT_rescale_normal GL_EXT_separate_specular_color GL_EXT_texture_edge_clamp GL_SGIS_generate_mipmap GL_SGIS_texture_border_clamp GL_SGIS_texture_edge_clamp GL_SGIS_texture_lod GL_ARB_framebuffer_sRGB GL_ARB_multitexture GL_EXT_framebuffer_sRGB GL_IBM_multimode_draw_arrays GL_IBM_texture_mirrored_repeat GL_ARB_texture_cube_map GL_ARB_texture_env_add GL_ARB_transpose_matrix GL_EXT_blend_func_separate GL_EXT_fog_coord GL_EXT_multi_draw_arrays GL_EXT_secondary_color GL_EXT_texture_env_add GL_EXT_texture_filter_anisotropic GL_EXT_texture_lod_bias GL_INGR_blend_func_separate GL_NV_blend_square GL_NV_light_max_exponent GL_NV_texgen_reflection GL_NV_texture_env_combine4 GL_SUN_multi_draw_arrays GL_ARB_texture_border_clamp GL_ARB_texture_compression GL_EXT_framebuffer_object GL_EXT_texture_env_combine GL_EXT_texture_env_dot3 GL_MESA_window_pos GL_NV_packed_depth_stencil GL_NV_texture_rectangle GL_ARB_depth_texture GL_ARB_occlusion_query GL_ARB_shadow GL_ARB_texture_env_combine GL_ARB_texture_env_crossbar GL_ARB_texture_env_dot3 GL_ARB_texture_mirrored_repeat GL_ARB_window_pos GL_EXT_stencil_two_side GL_EXT_texture_cube_map GL_NV_depth_clamp GL_NV_fog_distance GL_APPLE_packed_pixels GL_APPLE_vertex_array_object GL_ARB_draw_buffers GL_ARB_fragment_program GL_ARB_fragment_shader GL_ARB_shader_objects GL_ARB_vertex_program GL_ARB_vertex_shader GL_ATI_draw_buffers GL_ATI_texture_env_combine3 GL_ATI_texture_float GL_EXT_shadow_funcs GL_EXT_stencil_wrap GL_MESA_pack_invert GL_NV_primitive_restart GL_ARB_depth_clamp GL_ARB_fragment_program_shadow GL_ARB_half_float_pixel GL_ARB_occlusion_query2 GL_ARB_point_sprite GL_ARB_shading_language_100 GL_ARB_sync GL_ARB_texture_non_power_of_two GL_ARB_vertex_buffer_object GL_ATI_blend_equation_separate GL_EXT_blend_equation_separate GL_OES_read_format GL_ARB_color_buffer_float GL_ARB_pixel_buffer_object GL_ARB_texture_compression_rgtc GL_ARB_texture_float GL_ARB_texture_rectangle GL_EXT_packed_float GL_EXT_pixel_buffer_object GL_EXT_texture_compression_rgtc GL_EXT_texture_rectangle GL_EXT_texture_sRGB GL_EXT_texture_shared_exponent GL_ARB_framebuffer_object GL_EXT_framebuffer_blit GL_EXT_framebuffer_multisample GL_EXT_packed_depth_stencil GL_ARB_vertex_array_object GL_ATI_separate_stencil GL_EXT_draw_buffers2 GL_EXT_draw_instanced GL_EXT_gpu_program_parameters GL_EXT_texture_array GL_EXT_texture_integer GL_EXT_texture_sRGB_decode GL_EXT_timer_query GL_OES_EGL_image GL_ARB_copy_buffer GL_ARB_depth_buffer_float GL_ARB_draw_instanced GL_ARB_half_float_vertex GL_ARB_instanced_arrays GL_ARB_map_buffer_range GL_ARB_texture_rg GL_ARB_texture_swizzle GL_ARB_vertex_array_bgra GL_EXT_texture_swizzle GL_EXT_vertex_array_bgra GL_NV_conditional_render GL_AMD_conservative_depth GL_ARB_ES2_compatibility GL_ARB_blend_func_extended GL_ARB_debug_output GL_ARB_draw_elements_base_vertex GL_ARB_explicit_attrib_location GL_ARB_fragment_coord_conventions GL_ARB_provoking_vertex GL_ARB_sampler_objects GL_ARB_seamless_cube_map GL_ARB_shader_texture_lod GL_ARB_texture_multisample GL_ARB_texture_rgb10_a2ui GL_ARB_uniform_buffer_object GL_ARB_vertex_type_2_10_10_10_rev GL_EXT_provoking_vertex GL_EXT_texture_snorm GL_MESA_texture_signed_rgba GL_ARB_get_program_binary GL_ARB_robustness GL_ARB_separate_shader_objects GL_ARB_shader_bit_encoding GL_ARB_timer_query GL_ARB_compressed_texture_pixel_storage GL_ARB_conservative_depth GL_ARB_internalformat_query GL_ARB_map_buffer_alignment GL_ARB_shading_language_420pack GL_ARB_shading_language_packing GL_ARB_texture_storage GL_EXT_framebuffer_multisample_blit_scaled GL_EXT_transform_feedback GL_AMD_shader_trinary_minmax GL_ARB_ES3_compatibility GL_ARB_arrays_of_arrays GL_ARB_clear_buffer_object GL_ARB_explicit_uniform_location GL_ARB_invalidate_subdata GL_ARB_program_interface_query GL_ARB_texture_query_levels GL_ARB_texture_storage_multisample GL_ARB_vertex_attrib_binding GL_KHR_debug GL_ARB_multi_bind GL_EXT_shader_integer_mix GL_ARB_get_texture_sub_image GL_KHR_context_flush_control 
Screen BPP:          24
SDL_GL_RED_SIZE:     requested 8, got 8
SDL_GL_GREEN_SIZE:   requested 8, got 8
SDL_GL_BLUE_SIZE:    requested 8, got 8
SDL_GL_ALPHA_SIZE:   requested 8, got 8
SDL_GL_STENCIL_SIZE: requested 8, got 8
SDL_GL_DOUBLEBUFFER: requested 1, got 1

qwegary, 2020-06-24:
主机上的信息是：
false
GL Vendor     : Intel
GL Renderer   : Intel(R) HD Graphics 2000
GL Version    : 3.1.0 - Build 9.17.10.4459
GL Extensions : GL_EXT_blend_minmax GL_EXT_blend_subtract GL_EXT_blend_color GL_EXT_abgr GL_EXT_texture3D GL_EXT_clip_volume_hint GL_EXT_compiled_vertex_array GL_SGIS_texture_edge_clamp GL_SGIS_generate_mipmap GL_EXT_draw_range_elements GL_SGIS_texture_lod GL_EXT_rescale_normal GL_EXT_packed_pixels GL_EXT_texture_edge_clamp GL_EXT_separate_specular_color GL_ARB_multitexture GL_EXT_texture_env_combine GL_EXT_bgra GL_EXT_blend_func_separate GL_EXT_secondary_color GL_EXT_fog_coord GL_EXT_texture_env_add GL_ARB_texture_cube_map GL_ARB_transpose_matrix GL_ARB_texture_env_add GL_IBM_texture_mirrored_repeat GL_EXT_multi_draw_arrays GL_NV_blend_square GL_ARB_texture_compression GL_3DFX_texture_compression_FXT1 GL_EXT_texture_filter_anisotropic GL_ARB_texture_border_clamp GL_ARB_point_parameters GL_ARB_texture_env_combine GL_ARB_texture_env_dot3 GL_ARB_texture_env_crossbar GL_EXT_texture_compression_s3tc GL_ARB_shadow GL_ARB_window_pos GL_EXT_shadow_funcs GL_EXT_stencil_wrap GL_ARB_vertex_program GL_EXT_texture_rectangle GL_ARB_fragment_program GL_EXT_stencil_two_side GL_ATI_separate_stencil GL_ARB_vertex_buffer_object GL_EXT_texture_lod_bias GL_ARB_occlusion_query GL_ARB_fragment_shader GL_ARB_shader_objects GL_ARB_shading_language_100 GL_ARB_texture_non_power_of_two GL_ARB_vertex_shader GL_NV_texgen_reflection GL_ARB_point_sprite GL_ARB_fragment_program_shadow GL_EXT_blend_equation_separate GL_ARB_depth_texture GL_ARB_texture_rectangle GL_ARB_draw_buffers GL_ARB_color_buffer_float GL_ARB_half_float_pixel GL_ARB_texture_float GL_ARB_pixel_buffer_object GL_EXT_framebuffer_object GL_ARB_draw_instanced GL_ARB_half_float_vertex GL_ARB_occlusion_query2 GL_EXT_draw_buffers2 GL_WIN_swap_hint GL_EXT_texture_sRGB GL_ARB_multisample GL_EXT_packed_float GL_EXT_texture_shared_exponent GL_ARB_texture_rg GL_ARB_texture_compression_rgtc GL_NV_conditional_render GL_EXT_texture_swizzle GL_ARB_sync GL_ARB_framebuffer_sRGB GL_EXT_packed_depth_stencil GL_ARB_depth_buffer_float GL_EXT_transform_feedback GL_EXT_framebuffer_blit GL_EXT_framebuffer_multisample GL_ARB_framebuffer_object GL_EXT_texture_array GL_EXT_texture_integer GL_ARB_map_buffer_range GL_EXT_texture_snorm GL_INTEL_performance_queries GL_ARB_copy_buffer GL_ARB_sampler_objects GL_NV_primitive_restart GL_ARB_seamless_cube_map GL_ARB_uniform_buffer_object GL_ARB_depth_clamp GL_ARB_vertex_array_bgra GL_ARB_shader_bit_encoding GL_ARB_draw_buffers_blend GL_ARB_texture_query_lod GL_ARB_explicit_attrib_location GL_ARB_draw_elements_base_vertex GL_ARB_instanced_arrays GL_ARB_fragment_coord_conventions GL_EXT_gpu_program_parameters GL_ARB_texture_buffer_object_rgb32 GL_ARB_compatibility GL_ARB_texture_rgb10_a2ui GL_ARB_vertex_type_2_10_10_10_rev GL_ARB_timer_query GL_INTEL_map_texture GL_ARB_vertex_array_object GL_ARB_provoking_vertex 
Screen BPP:          24
SDL_GL_RED_SIZE:     failed, OpenGL error: 00000502
SDL_GL_GREEN_SIZE:   failed, OpenGL error: 00000502
SDL_GL_BLUE_SIZE:    failed, OpenGL error: 00000502
SDL_GL_ALPHA_SIZE:   failed, OpenGL error: 00000502
SDL_GL_STENCIL_SIZE: failed, OpenGL error: 00000502
SDL_GL_DOUBLEBUFFER: requested 1, got 1

soyzhc, 2020-06-27:
```
SDL_GL_RED_SIZE: failed, OpenGL error: 00000502
SDL_GL_GREEN_SIZE: failed, OpenGL error: 00000502
SDL_GL_BLUE_SIZE: failed, OpenGL error: 00000502
SDL_GL_ALPHA_SIZE: failed, OpenGL error: 00000502
SDL_GL_STENCIL_SIZE: failed, OpenGL error: 00000502
SDL_GL_DOUBLEBUFFER: requested 1, got 1
```
从host机的信息来看，还是驱动兼容新的问题，抱歉我们这边都没有碰到过，只能等待后续推出兼容性很好的版本