zlgopen/awtk #430: 更新AWTK源码后，程序起不来？

ytchw101, 2020-06-22:
之前是使用AWTK 1.4.1版本的源码 ，应用程序运行显示都正常 （只是中文键盘输入时，一直按‘s’会卡死）。  上次更新了AWTK源码（2020-06-15）主要想解决中文键盘输入一直按's'会卡死问题。但是更新后程序直接挂了。  调试定位 死在如下：
注：我的应用程序启动会进行语言切换。
	 switch(lang)
	 {
	 case 0: //中文
		  rt_kprintf("11..........\r\n");
		 	 ReleaseFont();  rt_kprintf("22..........\r\n");
		   LoadFont(DEF_FONT);  rt_kprintf("33..........\r\n");
		   locale_info_change(locale_info(), "zh", "CN");  rt_kprintf("44..........\r\n");
	 	   system_info_set_default_font(system_info(), DEF_FONT);  rt_kprintf("55..........\r\n");
		   break;
	 case 1: //英文
		 	 ReleaseFont();
		   locale_info_change(locale_info(), "en", "US");
	 	   system_info_set_default_font(system_info(), DEF_FONT);
		   break;
               .......................................
    最后只打印到 33....................  就挂了  应该是locale_info_change 这句触发哪里挂了
  另外把语言切换都注释掉， 程序会运行 tk_run();  后也会挂掉 （具体不知道哪里）
 注1：
void LoadFont(const char* font_name)
{
		 if(strcmp(font_name, DEF_FONT) != 0) //预加载SD卡中字体
		 {
			  asset_info_t* iter = assets_manager_load(assets_manager(), ASSET_TYPE_FONT, 
                          font_name);
		 }
		 else  //系统默认字体
		 {
		    assets_manager_preload(assets_manager(), ASSET_TYPE_FONT, DEF_FONT);
		 }
}

注2：平台  嵌入式ARM+RTT+文件系统+LCD （480*272）

xianjimli, 2020-06-22:
资源有重新生成吗？

xianjimli, 2020-06-22:
6/15的版本有点问题，你更新到最新的吧。

ytchw101, 2020-06-22:
都重新生成的

soyzhc, 2020-06-22:
有尝试在windows模拟吗，看看是否也会崩溃

ytchw101, 2020-06-23:
更新到最新版本(AWTK 2020-06-22) 可以了

closed 2020-06-23 by ytchw101