zlgopen/awtk #440: 请教一下，tslib添加进去，提示没定义的，有人遇到过吗，是什么原因

qwegary, 2020-07-03:
[image]
编译不通过，一直提示函数没定义

xianjimli, 2020-07-03:
通常是没找到对应的库，或者库的格式不匹配。

qwegary, 2020-07-06:
我把库文件tslib.a拿掉直接提示cannot find -lts，把库加上去，就是提示里面的一些函数没定义，但是查看tslib.h文件中，存在对应的函数，tslib.a  tslib.h是板卡上其他demo上使用的。

qwegary, 2020-07-06:
[image]

xianjimli, 2020-07-06:
看看文件格式：

```
file tslib.a
```

qwegary, 2020-07-06:
[image]

qwegary, 2020-07-06:
[image]

xianjimli, 2020-07-06:
那不知道了。百度一下。

YinWenshuo, 2020-07-06:
你用的是.a静态库?我用的是.so动态库，没问题啊。估计你的库打包有问题，找不到对应的函数。@qwegary

qwegary, 2020-07-06:
我试一下用动态库的

qwegary, 2020-07-07:
使用动态链接库可以正常跑起来了，感谢

closed 2020-07-07 by qwegary