zlgopen/awtk #442: 希伯来文的显示顺序问题

lizhenningi, 2020-07-07:
您好！
我在strings.xml中添加希伯来文的翻译时，发现页面控件上显示的希伯来文字符串是反着的。但我使用“windows记事本、Notepad++”打开strings.xml显示的希伯来文顺序是正常的。

xianjimli, 2020-07-07:
目前还不支持R2L排版。

xianjimli, 2020-07-09:
基本控件支持了unicode bidirectional algorithm（编辑器和richtext等控件还不支持），你更新代码试试。

lizhenningi, 2020-07-10:
您好！
我更新了关于双向排布的代码，这些修改是生效的，但是使用的时候我发现在阿拉伯语中嵌入“标点符号、数字、英文字符”会与notepad++的显示不同。具体的说明在文档“阿拉伯语双向排版对比.docx”中。
[阿拉伯语双向排版对比.docx](https://github.com/zlgopen/awtk/files/4900423/default.docx)

lizhenningi, 2020-07-10:
您好！
关于非基本控件支持unicode bidirectional algorithm您这边有时间安排吗？由于我现在的项目使用的“hscroll_label”比较多，因此我要根据您的时间安排来对我现在的项目做出一些调整

xianjimli, 2020-07-10:
hscroll_label应该可以支持。rich_text和edit暂时支持不了，我不知道怎么搞。

xianjimli, 2020-07-10:
你把那个文本文件贴上来看看。

xianjimli, 2020-07-11:
把下面的代码拷贝到 https://www.w3schools.com/cssref/tryit.asp?filename=trycss_text_unicode-bidi 里测试

```html
<!DOCTYPE html>
<html>
<head>
<style>
div.ex1 {
  direction: rtl;
  unicode-bidi: bidi-override;
}
</style>
</head>
<body>

<div>1.ﺝﺎﻤﻋﺓ ﻙﻮﻟﻮﻤﺒﻳﺍ ﻉﺎﻣ.</div>
<div>10 يوليو/ تموز 2020</div>

<div class="ex1">1.ﺝﺎﻤﻋﺓ ﻙﻮﻟﻮﻤﺒﻳﺍ ﻉﺎﻣ.</div>
<div class="ex1">10 يوليو/ تموز 2020</div>
</body>
</html>
```

可以看到效果如下:
[image]

不开启RTL时，Notepad++和浏览器日期显示都是错误的。

开启RTL时，浏览器的显示效果与AWTK相同。Notepad++也一样，当前开启RTL时，显示效果与AWTK相同。

AWTK的处理没有问题啊，目前使用fribidi来实现bidi算法，参数用的自动检查。

lizhenningi, 2020-07-14:
您好!
您上面使用html做测试的style中使用了“direction: rtl;”这个属性，在AWTK的代码中有类似这个属性的的定义吗？如果有的话哪个接口能够设置它？

xianjimli, 2020-07-15:
bidi="ltr"

参考：demos/assets/default/raw/ui/bidi.xml

```c
  86 /**
  87  * @const WIDGET_PROP_BIDI
  88  * bidi type(rtl,ltr,auto,wrtl,wltr,lro,rlo)。
  89  */
  90 #define WIDGET_PROP_BIDI "bidi"
```

lizhenningi, 2020-07-15:
好的，谢谢