zlgopen/awtk #453: AWTK空闲的时候莫名挂掉

hanlong-coclean, 2020-07-23:
awtk的任务堆栈分配了3K大小，s_heam_mem使用的外扩psram的1M空间，板子跑起来了，一直没有动，突然发现挂掉了，下面抓到的日志，看着像是awtk自己空闲唤醒的时候死掉了，有人遇到过相似的问题么
`slist_size
D:\Dev\Em_Earth\EarthGDGUI\GD324xx_EVAL_Demo_Suites\Projects\Earth_Screen\MDK-ARM/..\awtk\src\/tkc\slist.c:159
0x0804015c
idle_manager_count
D:\Dev\Em_Earth\EarthGDGUI\GD324xx_EVAL_Demo_Suites\Projects\Earth_Screen\MDK-ARM/..\awtk\src\/tkc\idle_manager.c:152
0x0803bfda
event_source_idle_get_wakeup_time
D:\Dev\Em_Earth\EarthGDGUI\GD324xx_EVAL_Demo_Suites\Projects\Earth_Screen\MDK-ARM/..\awtk\src\/tkc\event_source_idle.c:43
0x0803bf5a
event_source_get_wakeup_time
D:\Dev\Em_Earth\EarthGDGUI\GD324xx_EVAL_Demo_Suites\Projects\Earth_Screen\MDK-ARM/..\awtk\src\/tkc\event_source.c:45
0x0803c16a
event_source_manager_get_wakeup_time
D:\Dev\Em_Earth\EarthGDGUI\GD324xx_EVAL_Demo_Suites\Projects\Earth_Screen\MDK-ARM/..\awtk\src\/tkc\event_source_manager.c:123
0x0803c068
event_source_manager_default_dispatch
D:\Dev\Em_Earth\EarthGDGUI\GD324xx_EVAL_Demo_Suites\Projects\Earth_Screen\MDK-ARM/..\awtk\src\/tkc\event_source_manager_default.c:135
0x0803c12c
event_source_manager_dispatch
D:\Dev\Em_Earth\EarthGDGUI\GD324xx_EVAL_Demo_Suites\Projects\Earth_Screen\MDK-ARM/..\awtk\src\/tkc\event_source_manager.c:64
0x08044d50
main_loop_simple_step
D:\Dev\Em_Earth\EarthGDGUI\GD324xx_EVAL_Demo_Suites\Projects\Earth_Screen\MDK-ARM/..\/awtk\src\main_loop\main_loop_simple.c:179
0x08044dce
main_loop_step
D:\Dev\Em_Earth\EarthGDGUI\GD324xx_EVAL_Demo_Suites\Projects\Earth_Screen\MDK-ARM/..\awtk\src\/base\main_loop.c:121
0x08044d18
main_loop_simple_run
D:\Dev\Em_Earth\EarthGDGUI\GD324xx_EVAL_Demo_Suites\Projects\Earth_Screen\MDK-ARM/..\/awtk\src\main_loop\main_loop_simple.c:195
0x08044b68
main_loop_run
D:\Dev\Em_Earth\EarthGDGUI\GD324xx_EVAL_Demo_Suites\Projects\Earth_Screen\MDK-ARM/..\awtk\src\/base\main_loop.c:31
0x08059fac
tk_run
D:\Dev\Em_Earth\EarthGDGUI\GD324xx_EVAL_Demo_Suites\Projects\Earth_Screen\MDK-ARM/..\awtk\src\/awtk_global.c:292
0x0803f90c
gui_app_start
D:\Dev\Em_Earth\EarthGDGUI\GD324xx_EVAL_Demo_Suites\Projects\Earth_Screen\MDK-ARM/..\/awtk\src\awtk_main.inc:197
0x0803f92e
gui_task
D:\Dev\Em_Earth\EarthGDGUI\GD324xx_EVAL_Demo_Suites\Projects\Earth_Screen\MDK-ARM/..\/main.c:282`
[image]

xianjimli, 2020-07-23:
看样子是内存问题，idle manager的内存被破坏了。

hanlong-coclean, 2020-07-24:
我现在给AWTK开辟了一个任务，在别的任务里面操作控件会不会有问题呢

xianjimli, 2020-07-24:
会有问题。请参考这个文档：https://github.com/zlgopen/awtk/blob/master/docs/how_to_non_gui_thread_operate_widget.md

hanlong-coclean, 2020-07-24:
好的，我改一下。
还有个问题：在idle_queue里面使用timer_add、timer_reset不会有问题吧！？