zlgopen/awtk #456: mledit的字体设置大点或者小点，鼠标贯标（cursor_text）不能准确地定位到目标位置

YinWenshuo, 2020-07-29:
问题描述如题，我在窗口里面的mledit如下：
`<mledit name="xml_edit" x="c" y="10" h="300" w="560" focus="true" input_type="float" text="MooYin@awtk"
      left_margin="33" top_margin="2" bottom_margin="2" right_margin="16" wrap_word="true" max_lines="500" style="half_trans"> 
     <line_number x="0" y="0" w="32" h="100%" value="0" style="half_trans"/>
     <scroll_bar_d x="right" y="0" w="28" h="100%" value="0"/>
   </mledit>`

对应的mledit style如下：
`<mledit selected_fg_color="#020202" selected_text_color="#010101" selected_bg_color="#b5d7fd">
  <style name="default" border_color="#a0a0a0"  text_color="#010101" text_align_h="left">
    <normal     bg_color="#f0f0f0" />
    <focused    bg_color="#f0f0f0" border_color="#010101"/>
    <disable    bg_color="gray" text_color="#d0d0d0" />
    <error      bg_color="#f0f0f0" text_color="red" />
    <empty      bg_color="#f0f0f0" text_color="#a0a0a0" />
    <empty_focus bg_color="#f0f0f0" text_color="#a0a0a0" border_color="#010101"/>
  </style>
  <style name="half_trans" border_color="#a0a0a0" font_size="26" text_color="#010101" text_align_h="left">
    <normal     bg_color="rgba(240,240,240,0.2)" />
    <focused    bg_color="rgba(240,240,240,0.2)" border_color="#010101"/>
    <disable    bg_color="gray" text_color="#d0d0d0" />
    <error      bg_color="#f0f0f0" text_color="red" />
    <empty      bg_color="rgba(240,240,240,0.2)" text_color="#a0a0a0" />
    <empty_focus bg_color="rgba(240,240,240,0.2)" text_color="#a0a0a0" border_color="#010101"/>
  </style>
</mledit>`

在awtk微信群里按照awtk团队的兄弟建议“设置一下dragger控件中scroll_bar样式的font_size与mledit的一样大小试试”
我将styles/default.xml中的dragger如下：
<dragger>
  <style name="scroll_bar" font_size="26" >
    <normal     bg_color="#d0d0d0" />
    <pressed    bg_color="#c0c0c0" />
    <over       bg_color="#c0c0c0" />
  </style>
</dragger>

按照如上修改了dragger的style后，问题依然存在。

tangMans, 2020-07-29:
需要在half_trans样式下面添加：<nomal />状态，如下：

<style name="half_trans" border_color="#a0a0a0" font_size="26" text_color="#010101" text_align_h="left">
<nomal />
 </style>

tangMans, 2020-07-29:
需要在half_trans样式下面添加：状态，如下：

<style name="half_trans" border_color="#a0a0a0" font_size="26" text_color="#010101" text_align_h="left">
<normal />
 </style>

tangMans, 2020-07-29:
需要在half_trans样式下面添加：<normal />

tangMans, 2020-07-29:
需要在half_trans样式中添加: normal 状态

YinWenshuo, 2020-07-29:
我有normal状态，上面贴的xml代码怎么好像看不全，我再在这里贴一下：

'''<style name="half_trans" border_color="#a0a0a0" font_size="18" text_color="#010101" text_align_h="left">
    <normal     bg_color="rgba(240,240,240,0.2)" />
    <focused    bg_color="rgba(240,240,240,0.2)" border_color="#010101"/>
    <disable    bg_color="gray" text_color="#d0d0d0" />
    <error      bg_color="#f0f0f0" text_color="red" />
    <empty      bg_color="rgba(240,240,240,0.2)" text_color="#a0a0a0" />
    <empty_focus bg_color="rgba(240,240,240,0.2)" text_color="#a0a0a0" border_color="#010101"/>
  </style>'''

YinWenshuo, 2020-07-29:
拉了最新的awtk，重新编译问题依然存在。

closed 2020-07-31 by YinWenshuo

ytchw101, 2022-11-24:
解决了吗？ 我的这个换了个大LCD（800*480）屏  字体只能在默认 很小时  点击才精准  设置其他字体 光标基本基本没法点  
以前是480*272的电阻触摸屏  光标  基本点哪都行