zlgopen/awtk #458: 关于中文输入 是否有内存泄漏问题？

ytchw101, 2020-07-31:
程序里设置两个界面。 在主界面时点击一个按钮测试可动态申请多少空间 得到：20254720字节  点击另外按钮进入二级界面后 每次直接返回到主界面或者英文输入后再返回到主界面 后  ， 点击测试按钮 一样可以申请到20254720字节 空间。
但是一旦以中文输入后（随便输入一个或多字符），再返回到主界面 就只能申请到 19435520字节空间， 然后多次进入二级界面以中文输入再返回主界面 也都是只能申请到19435520字节空间。   
1. 不知道这20254720- 19435520 = 819200字节空间是在哪被申请并且不释放？
2. 我关闭输入法有什么特别操作吗？我每次都是直接关闭界面。

ytchw101, 2020-07-31:
我在tk_alloc中加打印 在中文输入法下 随便输入一个字符 打印：
tk_alloc size=12500
tk_alloc size=284
tk_alloc size=64
tk_alloc size=4
tk_alloc size=13372
tk_alloc size=128
tk_alloc size=24
tk_alloc size=3304
tk_alloc size=8
tk_alloc size=16
tk_alloc size=8
tk_alloc size=8
tk_alloc size=8
tk_alloc size=184
tk_alloc size=24
tk_alloc size=8
tk_alloc size=40
tk_alloc size=24
tk_alloc size=8
tk_alloc size=16
tk_alloc size=8
tk_alloc size=24
tk_alloc size=16
tk_alloc size=8
tk_alloc size=48
tk_alloc size=24
tk_alloc size=8
tk_alloc size=8
tk_alloc size=8
tk_alloc size=16
tk_alloc size=8
tk_alloc size=32
tk_alloc size=8
tk_alloc size=40
tk_alloc size=24
tk_alloc size=8
tk_alloc size=8
tk_alloc size=8
tk_alloc size=16
tk_alloc size=8
tk_alloc size=32
tk_alloc size=16
tk_alloc size=8
tk_alloc size=40
tk_alloc size=24
tk_alloc size=8
tk_alloc size=16
tk_alloc size=8
tk_alloc size=32
tk_alloc size=16
tk_alloc size=8
tk_alloc size=16
tk_alloc size=8
tk_alloc size=32
tk_alloc size=8
tk_alloc size=24
tk_alloc size=8
tk_alloc size=40
tk_alloc size=8
tk_alloc size=8
tk_alloc size=16
tk_alloc size=8
tk_alloc size=8
tk_alloc size=8
tk_alloc size=8
tk_alloc size=32
tk_alloc size=24
tk_alloc size=8
tk_alloc size=16
tk_alloc size=8
tk_alloc size=16
tk_alloc size=8
tk_alloc size=32
tk_alloc size=16
tk_alloc size=8
tk_alloc size=32
tk_alloc size=24
tk_alloc size=8
tk_alloc size=16
tk_alloc size=8
tk_alloc size=16
tk_alloc size=8
tk_alloc size=32
tk_alloc size=16
tk_alloc size=8
tk_alloc size=16
tk_alloc size=40
tk_alloc size=16
tk_alloc size=8
tk_alloc size=8
tk_alloc size=8
tk_alloc size=8
tk_alloc size=24
tk_alloc size=8
tk_alloc size=32
tk_alloc size=24
tk_alloc size=8
tk_alloc size=16
tk_alloc size=8
tk_alloc size=16
tk_alloc size=8
tk_alloc size=32
tk_alloc size=16
tk_alloc size=8
tk_alloc size=48
tk_alloc size=24
tk_alloc size=8
tk_alloc size=16
tk_alloc size=8
tk_alloc size=32
tk_alloc size=16
tk_alloc size=8
tk_alloc size=8
tk_alloc size=16
tk_alloc size=8
tk_alloc size=32
tk_alloc size=8
tk_alloc size=40
tk_alloc size=24
tk_alloc size=8
tk_alloc size=16
tk_alloc size=8
tk_alloc size=32
tk_alloc size=16
tk_alloc size=8
tk_alloc size=8
tk_alloc size=48
tk_alloc size=24
tk_alloc size=8
tk_alloc size=16
tk_alloc size=8
tk_alloc size=32
tk_alloc size=16
tk_alloc size=8
tk_alloc size=8
tk_alloc size=16
tk_alloc size=8
tk_alloc size=24
tk_alloc size=8
tk_alloc size=8
tk_alloc size=40
tk_alloc size=24
tk_alloc size=8
tk_alloc size=16
tk_alloc size=8
tk_alloc size=24
tk_alloc size=16
tk_alloc size=8
tk_alloc size=8
tk_alloc size=16
tk_alloc size=40
tk_alloc size=16
tk_alloc size=8
tk_alloc size=8
tk_alloc size=8
tk_alloc size=8
tk_alloc size=24
tk_alloc size=8
tk_alloc size=40
tk_alloc size=16
tk_alloc size=8
tk_alloc size=8
tk_alloc size=8
tk_alloc size=16
tk_alloc size=8
tk_alloc size=32
tk_alloc size=8
tk_alloc size=48
tk_alloc size=24
tk_alloc size=8
tk_alloc size=8
tk_alloc size=8
tk_alloc size=16
tk_alloc size=8
tk_alloc size=32
tk_alloc size=8
tk_alloc size=40
tk_alloc size=24
tk_alloc size=8
tk_alloc size=16
tk_alloc size=8
tk_alloc size=8
tk_alloc size=32
tk_alloc size=16
tk_alloc size=8
tk_alloc size=40
tk_alloc size=24
tk_alloc size=8
tk_alloc size=16
tk_alloc size=8
tk_alloc size=24
tk_alloc size=16
tk_alloc size=8
tk_alloc size=16
tk_alloc size=8
tk_alloc size=32
tk_alloc size=8
tk_alloc size=32
tk_alloc size=16
tk_alloc size=8
tk_alloc size=16
tk_alloc size=8
tk_alloc size=16
tk_alloc size=40
tk_alloc size=16
tk_alloc size=8
tk_alloc size=8
tk_alloc size=8
tk_alloc size=8
tk_alloc size=24
tk_alloc size=8
tk_alloc size=40
tk_alloc size=16
tk_alloc size=8
tk_alloc size=8
tk_alloc size=8
tk_alloc size=16
tk_alloc size=8
tk_alloc size=24
tk_alloc size=8
tk_alloc size=48
tk_alloc size=24
tk_alloc size=8
tk_alloc size=16
tk_alloc size=8
tk_alloc size=16
tk_alloc size=8
tk_alloc size=32
tk_alloc size=8
tk_alloc size=40
tk_alloc size=24
tk_alloc size=8
tk_alloc size=16
tk_alloc size=8
tk_alloc size=16
tk_alloc size=8
tk_alloc size=32
tk_alloc size=16
tk_alloc size=8
tk_alloc size=826
tk_alloc size=267586
tk_alloc size=34076
tk_alloc size=34076
tk_alloc size=6624
tk_alloc size=461720
tk_alloc size=195333
tk_alloc size=828
tk_alloc size=2400
tk_alloc size=400
tk_alloc size=65102
tk_alloc size=512
tk_alloc size=356
tk_alloc size=7
tk_alloc size=11
tk_alloc size=356
tk_alloc size=7
tk_alloc size=11
tk_alloc size=588
tk_alloc size=80
tk_alloc size=4
tk_alloc size=264
tk_alloc size=680
tk_alloc size=56004

觉得应该是如下：这几个申请 但关闭没释放吧 

tk_alloc size=267586
tk_alloc size=34076
tk_alloc size=34076
tk_alloc size=6624
tk_alloc size=461720
tk_alloc size=195333
目前还在跟踪 查找。。。。。

xianjimli, 2020-07-31:
google pinyin应该没有泄漏，我用valgrind查过。

xianjimli, 2020-07-31:
它内部是缓存的，但是具体实现我不太清楚。

ytchw101, 2020-08-07:
目前定位应该是键盘引起内存泄漏。  比如在主机面时能申请到20766720字节  ，切换到次级界面（次级界面一个返回按钮，一个mledit控件）  我如果不打开键盘返回到主界面 依然能申请到20766720字节（重复多次都是）， 一旦我进入次级界面并且打开键盘， 再返回到主界面 就只能申请到20152320字节    重复多次进入次级界面在返回  始终都是申请到20152320字节   

所以：1、键盘打开时是不是申请某个空间并且一直不用释放？ 或者是不是某个指针未置空？
           2、我退出有键盘打开的界面 我是直接window_close 掉界面，对于键盘需要主动调用什么去关闭吗？

注：我进入次级界面，键盘打开是通过	mledit_set_focus(edit, TRUE); //设置为焦点则键盘自动打开

xianjimli, 2020-08-07:
你用的哪个版本，记得keyboard以前是发现过内存泄漏，你用最新代码试试。

ytchw101, 2020-08-07:
用的是2020-06-22版本

ytchw101, 2020-08-07:
查看 你们动态更新记录 子2020-06-22后 没看到有更新解决内存泄漏问题  所以就没更新了

xianjimli, 2020-08-08:
我查了一下，确实有泄漏，由于该泄漏会在窗口关闭时会释放，所以用valgrind没查出来。你可以更新代码试试。谢谢

xianjimli, 2020-08-08:
在PC上可以开启宏ENABLE_MEM_LEAK_CHECK，辅助查内存泄漏。

```
COMMON_CCFLAGS=COMMON_CCFLAGS+' -DENABLE_MEM_LEAK_CHECK=1 '
```

ytchw101, 2020-08-19:
更新到最新， 对我i这边还是同样问题。  怀疑是不是 中文输入时，某地方申请1M多字节空间A ，紧接着又申请很少自己空间B，  界面关闭后A空间释放的， 但是某个很小的B空间未释放 导致了我内存不连续。    反正目前还是只要中文输入后 再返回 怎么都申请不到原来的空间了  比原来的固定少了1M多