zlgopen/awtk #463: 怎么固定显示输入法键盘呢？

ytchw101, 2020-08-20:
1.edit控件获得焦点后输入法键盘弹出， edit失去焦点就会自动关闭输入法。 怎么样固定这个键盘始终打开？

2.使用 method edit_set_open_im_when_focused 设置编辑器是否在获得焦点时打开输入法。 设置为false为什么 点击edit后输入法键盘还能弹出？

xianjimli, 2020-08-23:
edit_set_open_im_when_focused 属性有两个用途：
* 设置默认焦点时，打开窗口时不弹出软键盘。
* 用键盘切换焦点时，编辑器获得焦点时不弹出软键盘。

xianjimli, 2020-08-23:
> 焦点

> 1.edit控件获得焦点后输入法键盘弹出， edit失去焦点就会自动关闭输入法。 怎么样固定这个键盘始终打开？
目前不行，什么样的场景需要？具有普适性的话，我加个属性来控制。

ytchw101, 2020-08-26:
也不是什么特殊场景  主要比如我LCD某个区域 就是用来显示键盘  但点击非edit控件后失去焦点后键盘自动关闭了  这样感觉 那一块就空了 （界面看起来不对称 ）

ytchw101, 2020-09-15:
个人觉得 输入法键盘 应该 加个属性来控制 键盘是否固定 并最好在键盘获得焦点时自动适配界面的某个编辑框控件，目前我们基于AWTK的嵌入式产品 还没出去多少 就已经有几个客户提出键盘能否固定  因为在LCD触摸屏上 很容易碰到除编辑框外的其他地方 导致编辑框失去焦点从而导致键盘关闭问题 - 这样以来不进界面空了一块  客户还得再次点下编辑框 键盘才能出来。

xianjimli, 2020-09-15:
好的。谢谢

xianjimli, 2020-09-30:
```c
  /**
   * @property {bool_t} close_im_when_blured
   * @annotation ["set_prop","get_prop","readable","persitent","design","scriptable"]
   * 
   * 是否在失去焦点时关闭输入法(默认是)。
   *
   */
  bool_t close_im_when_blured;
```

ytchw101, 2020-09-30:
mledit  为啥不加 close_im_when_blured ？

closed 2020-10-09 by ytchw101