zlgopen/awtk #465: 请教 子控件布局 问题

YinWenshuo, 2020-08-25:
我的xml代码如下：
`<view name="note_indicate" x="0" y="0" w="742" h="60" children_layout="default(r=1,c=4,ki=true)" >
      <image name="od_tips" x="30" y="0" w="80" h="60"  image="avm_od_warning" visible="TRUE"/>
      <image name="MEB_tips" x="110" y="0" w="80" h="60" image="avm_MEB_on" visible="TRUE"/>
      <label name="safe_tips" x="198" y="0" w="328" h="60" style="normal_tips" visible="TRUE"/>
      <image name="view_chnl_indicate" y="0" x="534" w="80" h="60" image="avm_front_icon" visible="TRUE"/>
    </view>`
运行后里面的子控件没有按照我的设想显示，我将children_layout="default(r=1,c=4,ki=true)"改成children_layout="default(r=1,c=0,ki=true)"也不行。
图1是我设想的显示，图2是c=4的显示，图3是c=0的显示。请问，问题在哪儿，子控件指定的x/y/w/h怎么不起作用？

我要怎么才能达到我设想的显示。

xianjimli, 2020-08-25:
画个示意图看看

YinWenshuo, 2020-08-25:
[image]
[image]
[image]

@xianjimli

xianjimli, 2020-08-26:
你都用了绝对坐标，就不需要children_layout了。

YinWenshuo, 2020-08-26:
> 你都用了绝对坐标，就不需要children_layout了。

对，可以了。谢谢！

closed 2020-08-26 by YinWenshuo