zlgopen/awtk #466: 超出overlay窗口范围的控件没有点击事件

hang1014, 2020-08-26:
在做界面的时候有个场景需要弹出一个窗口覆盖当前窗口的一部分，但不能影响当前窗口其余部分的正常显示和点击操作。我们使用了overlay作为弹出窗口，这个窗口分为三部分：顶部的状态栏、中间的主窗口、底部的软按键，overlay的范围就是中间的主窗口，其余部分是超出窗口范围显示，xml如下
`
 
    <overlay style="menu_bg" h="160" x="0" y="40" w="217">

		<view name="statu_bar" x="0" y="-40" w="320"  h="32" style="call_title_bg" >
		</view>

		<list_view  style="nobg" auto_hide_scroll_bar="true" h="100%" x="0" y="0" w="100%" item_height="160">
		</list_view>
			
		<view name="call_soft_key" x="0" y="166" w="320" h="34" children_layout="default(c=4,r=1)" style="softkey_bg">  
			<button name="soft_key_1" tr_text="" style="nostyle"/>        
			<button name="soft_key_2" tr_text="" style="nostyle"/>     
			<button name="soft_key_3" tr_text="" style="nostyle"/>  
			<button name="soft_key_4" tr_text="" style="nostyle"/>   
		</view> 
		
	</overlay>
`

经过试验发现底部的软按键栏可以正常显示，但点击却作用在了下层的窗口上，我们试了用dialog代替overlay，但是下层的窗口就不能点击了，针对这个问题您有什么建议吗？

xianjimli, 2020-08-27:
没看明白，贴个图上来看看。

hang1014, 2020-08-27:
[image]
红框部分是上层的overlay窗口，其余部分是下层的窗口，底下的按键和右侧的按钮需要能点击，现在底下的按键点击会作用在下层窗口上

WNsACE, 2020-08-28:
你好，请问你为什么这样子做？从 awtk 的机制中是通过鼠标坐标查找 overlay 窗口，找到 overlay 窗口后再查找对应的 button 控件，但是由于你上面提供的 xml 中的 button 按钮的绝对坐标已经超过 overlay 窗口的区域，当鼠标移动到 button 上面的坐标，该坐标已经不在  overlay 窗口的区域中了，所以导致 widget_find_target 函数已经找不到 overlay 窗口了，所以更加无法找到 overlay 窗口的子控件（button），所以就无法正确给 button 传入消息了。

hang1014, 2020-08-28:
我想让上图红框中的部分在一个窗口中，这样便于操作，同时不要覆盖其他部分（可以点击下层窗口的按钮），请问有没有什么办法能实现类似的功能呢？

WNsACE, 2020-08-28:
可以试一下把三个红色框框分为三个 overlay 窗口，一次打开三个 overlay 窗口，而且设置好 overlay 窗口的 x 和 y 坐标，应该可以达到你想要的效果

xianjimli, 2020-08-28:
先等等，让我想想。

xianjimli, 2020-08-28:
> overlay 增加 click_throgh 属性，允许 overlay 窗口在没点击到子控件时，穿透到底层窗口。

你有空试试，可以参考demoui里的用法。

closed 2020-10-30 by hang1014