zlgopen/awtk #474: switch、check_button等具有EVT_VALUE_CHANGED事件的控件的使用问题

YinWenshuo, 2020-09-10:
我定义了一个switch控件，这个switch控件绑定了一个on_switchXXX()处理函数；当我点击这个switch控件时，会调用绑定的on_switchXXX()函数，我的问题是：
       我不想立马就改变这个switch控件的值，而是在on_switchXXX()里面发送一条消息给另一个模块，另一个模块来决定我这switch是开还是关，现实情况是，我点击这个switch，这个swich的显示会自动切换到开或者关，**所以，我在on_switchXXX()里面调用widget_set_value(switchBtn,我当前的值)来保持当前的开关显示状态，**当我收到另一个模块的消息后，我再调用widget_set_value(switchBtn,模块给我的值)；这就有个问题：每次当我调用widget_set_value(switchBtn,值)时，实际上是触发了一次EVT_VALUE_CHANGED事件，从而会调一次此前绑定的on_switchXXX()处理函数。

    所有有EVT_VALUE_CHANGED事件的控件都有这个问题。

当然，我可以使用button来替换switch，对button调用widget_use_style来实现switch的开关显示效果，但还是希望awtk能看下怎么处理这种情况。

YinWenshuo, 2020-09-10:
对了，使用我上面提到的方法，button替换switch，切换开和关状态，有个瑕疵，就是没有switch开和关状态切换的动态效果。

xianjimli, 2020-09-10:
用emitter_disable/emitter_enable可以禁止/启用事件触发。

YinWenshuo, 2020-09-10:
> 用emitter_disable/emitter_enable可以禁止/启用事件触发。

调用这个两个是可以禁止、启用事件触发。但是，点击还是会改变switch的开、关显示状态。on_switchXXX()的代码：
...
 emitter_disable(target->emitter);
    switch_set_value(target,1);
    emitter_enable(target->emitter);

  send_msg_to_aaaa()//发消息给aaaa模块。
...

我传的是1，但是点击后switch的显示状态变成了off。然道传的1没起作用？

closed 2020-09-10 by YinWenshuo

xianjimli, 2020-09-18:
新增功能：EVT_VALUE_WILL_CHANGE 事件返回RET_STOP可以阻止修改值。

你看看是不是你需要的。

WeSrc-Yin, 2020-10-08:
> 新增功能：EVT_VALUE_WILL_CHANGE 事件返回RET_STOP可以阻止修改值。
> 
> 你看看是不是你需要的。

不好意思，今天才看到你的这条回复。我试了下EVT_VALUE_WILL_CHANGE，如下：
widget_on(widget, EVT_VALUE_WILL_CHANGE, on_SwitchXXX, win);

在on_SwitchXXX里面最后return RET_STOP.

我上面描述的问题依然存在。

我的解法是这样的：
1、时间绑定仍然是这样：widget_on(widget, EVT_VALUE_CHANGED, on_SwitchXXX, win);

2、on_SwitchXXX
{
     emitter_disable(target->emitter);
     //这里是发送一条消息给另一个模块
     emitter_enable(target->emitter);
}
3、recive_msg_from_modualA()
{
   这里接收另一个模块的消息。来决定这个开关的显示状态
}

这样就就ok。所以，你加的EVT_VALUE_WILL_CHANGE不起作用，用
     emitter_disable(target->emitter);
     //这里是发送一条消息给另一个模块
     emitter_enable(target->emitter);

就好。

谢谢！！！

xianjimli, 2020-10-08:
你可以看看这个例子：https://github.com/zlgopen/awtk-c-demos/blob/master/demos/switch_on_changed.c