zlgopen/awtk #476: 关于输入法键盘！

ytchw101, 2020-09-15:
1.之前提过的问题 想重申下 望解答：个人觉得 输入法键盘 应该 加个属性来控制 键盘是否固定 并最好在键盘获得焦点时自动适配界面的某个编辑框控件，目前我们基于AWTK的嵌入式产品 还没出去多少 就已经有几个客户提出键盘能否固定 因为在LCD触摸屏上 很容易碰到除编辑框外的其他地方 导致编辑框失去焦点从而导致键盘关闭问题 - 这样以来不进界面空了一块 客户还得再次点下编辑框 键盘才能出来。

2。输入法中文输入时，中间的Space 空格键盘，为什么不起作用？ 英文输入可以   （AWTK源码是2020-06-22更新的)

xianjimli, 2020-09-15:
编辑器加了 属性closei_im_when_blured，你有空试试。最新代码，可以输入空格，你更新代码试试。谢谢

ytchw101, 2020-09-30:
1.edit设置closei_im_when_blured  = FALSE 后可以固定 但是mledit 没实现？
2.更新代码后，中文输入空格后，编辑框里没显示空格 且再次输入字符（中文输入状态时）都是英文 ？

ytchw101, 2020-10-09:
再咨询个问题  3.在嵌入式linux平台下 ，  我屏幕左半部分是一些按钮等控件，右半部分打算用来显示键盘  但是  键盘从下弹出的时候  左半部分的控件也跟着上移了？？        我之前做过的一个  在嵌入式ARM平台使用 没有这样问题   是可以的

xianjimli, 2020-10-09:
一直是这的啊。

ytchw101, 2020-10-09:
哦  现在知道了 是键盘高度超过edit控件  就会上移动     我现在遇到问题是 使用的屏幕较大（800*480 ）所以想分成左右两半  左边显示编辑框等其他控件 右边显示键盘    所以出现上面的问题      最好有什么属性能控制是否 允许上移动。。。

还有之前你们修改edit  增加属性closei_im_when_blured  试了可以   但是mledit什么时候能支持呢？

xianjimli, 2020-10-09:
mledit也加close_im_when_blured了，你更新代码试试。

好的，我想想。

xianjimli, 2020-10-11:
我改了一下，如果keyboard和edit没有重叠，不再上推窗口。你更新代码试试。

ytchw101, 2020-10-11:
哦  好  辛苦你啦！

ytchw101, 2020-10-12:
如果 keyboard 与 edit 没有重叠，不再上推窗口  这个试了可以    但    mledit没改！？

ytchw101, 2020-10-12:
另外 比如我界面有大于2个edit控件 每次点击某个edit时，都会重新弹出下键盘, 请问这个能否改成 在我设置了close_im_when_blured = false（即固定键盘） 时只弹出一次呢？

xianjimli, 2020-10-17:
这个需求好像不具有普适性。