zlgopen/awtk #478: widget_set_text_utf8()或者widget_set_text()，能使显示更新的最小时间问题

hanlong-coclean, 2020-09-21:
场景如下：
    通过中断发送信号，使用idle_queue通知AWTK，来更改lable显示的内容（比如从1累加到100，如此循环），中断信号慢的话显示 
   正常，但是如果中断太快了（ 比如1秒10个中断），发现lable需要2~3秒才能把更新这10次。
所以像这些使用widget_set_text_utf8()或者widget_set_text()，来更新label显示内容最小刷新时间大概多少，（或者刷新一个lable的显示最少需要多久呢）

xianjimli, 2020-09-21:
这个与帧率有关。什么平台？帧率大概多少

hanlong-coclean, 2020-09-21:
用的GD32F407zgt6。
320*240的屏，芯片是ST7789V，驱动配置的帧率是60HZ

xianjimli, 2020-09-21:
那有点怪。你在PC上用timer模拟一下，如果能重现，你放上来我试试。

hanlong-coclean, 2020-09-21:
widget_t* win;
widget_t* text_change_label;
int64_t  count=10;
ret_t text_change_label_cb(const timer_info_t* info)
{
	char res_buf[20]={0};
	count++;
	if(count == 100)
		count = 0;
	tk_snprintf(res_buf,sizeof(res_buf),"%d",count);
	widget_set_text_utf8(text_change_label,res_buf);
	widget_invalidate(text_change_label,NULL);
	printf("%d\r\n",count);
	return RET_REPEAT;
}
ret_t application_init() { 
   win = window_create(NULL, 0, 0, 0, 0);
   widget_set_style_color(win,"normal:bg_color",0xff00ff00);
	text_change_label = label_create(win,100,100,50,50);
	widget_set_text_utf8(text_change_label,"1");
	widget_invalidate(text_change_label,NULL);
	timer_add(text_change_label_cb,NULL,10);
}
PC上这么跑的代码看着没问题

xianjimli, 2020-09-22:
那你打开fps显示，看看实际的fps有多少

hanlong-coclean, 2020-09-22:
实际的板子上打开fps么，在哪里打开呢

xianjimli, 2020-09-22:
你可以参考demoui里的代码：

```c
487 static ret_t on_show_fps(void* ctx, event_t* e) {
488   widget_t* button = WIDGET(ctx);
489   widget_t* widget = window_manager();
490   window_manager_t* wm = WINDOW_MANAGER(widget);
491 
492   widget_invalidate(widget, NULL);
493   window_manager_set_show_fps(widget, !wm->show_fps);
494   widget_set_text(button, wm->show_fps ? L"Hide FPS" : L"Show FPS");
495 
496   return RET_OK;
497 }
```

hanlong-coclean, 2020-09-22:
有中断信号的时候在8fps，没操作的时候在9fps，偶尔也会跳，但是在7~10之间。

这个指标有什么含义呢？

closed 2020-09-23 by hanlong-coclean