zlgopen/awtk #481: idle_queue的使用问题

YinWenshuo, 2020-09-25:
自定义的数据结构：
typedef struct
{
     int a;
     int b;
}T__MSG, 

在非GUI线程里面发送数据，按照如下方式调用idle_queue跟GUI线程串起来。
{
    /*这是在非GUI线程里面*/
     ....
     T__MSG *pMsgData；
      ....
     //在调idle_queue之前将pMsgData数据打印出来是正确的
     idle_queue(on_idle,pMsgData);
     ....
}

static ret_t on_idle(onst idle_info_t* idleMsg)
{
    T_MSG *msgFrom = (T_MSG*)(idleMsg->ctx);
   //在这里将msgFrom 打印出来，有时候数据不正确。
}

请问我上面的这种用法有什么问题吗，为什么有时候在on_idle里面的数据不正确，大部分时候是正确的。

xianjimli, 2020-09-25:
主要看看xxx是什么，不要用临时变量。

```c
T__MSG *pMsgData；
pMsgData = xxx；//不同的操作会有不同的值
```

WeSrc-Yin, 2020-09-26:
> 主要看看xxx是什么，不要用临时变量。
> 
> ```c
> T__MSG *pMsgData；
> pMsgData = xxx；//不同的操作会有不同的值
> ```

xxx是在本文件定义的一个static 变量，我稍作修改后还是有个问题，概率还挺高的，10次出现一次数据不对的情况。我修改后的代码如下：

自定义的数据结构：
typedef struct
{
     int a;
     int b;
}T_MSG, 

static uint8_t *pparams;//定义一个静态全局变量

static ret_t on_idle(onst idle_info_t* idleMsg)
{
    T_MSG *msgFrom = (T_MSG*)(idleMsg->ctx);
   //在这里将msgFrom 打印出来，有时候数据不正确。
   ...

return RET_OK;
}

在非GUI线程里面发送数据，按照如下方式调用idle_queue跟GUI线程串起来。
void get_data(void *msg)//不同的操作msg会有不同的值
{
    /*这是在非GUI线程里面*/
    pparams = &msg;
     ....
     static T_MSG *pMsgData；
     pMsgData = (T_MSG *)pparams；
     //在调idle_queue之前将pMsgData数据打印出来一直都是正确的
     idle_queue(on_idle,pMsgData);
     ....
}

xianjimli, 2020-09-26:
```c
static T_MSG *pMsgData
```
其实这行代码没有意义，指针其实就是一个整数，调用idle_queue时已经把整数传递过去了，之后你修改它也不会影响idle函数里的参数，所以加不加static没有什么关系，关键在于指针指向的内容。上面的代码根本看不出问题，你弄一个可以重现的最小demo上来吧，否则我也没法帮你。

WeSrc-Yin, 2020-09-27:
> ```c
> static T_MSG *pMsgData
> ```
> 
> 其实这行代码没有意义，指针其实就是一个整数，调用idle_queue时已经把整数传递过去了，之后你修改它也不会影响idle函数里的参数，所以加不加static没有什么关系，关键在于指针指向的内容。上面的代码根本看不出问题，你弄一个可以重现的最小demo上来吧，否则我也没法帮你。

多谢，是我的问题，指针指向的内容被改变了。

WeSrc-Yin, 2020-09-28:
李老师，数据内容被改变的问题是好了，但我还有个疑问：
在这函数void get_data(void *msg)里面，每次msg变化，我都会调idle_queue(on_idle,pMsgData);
这样子，会不会影响系统性能？每次调idle_queue(on_idle,pMsgData);on_idle执行完后要不要“销毁”？怎么“销毁”？
@xianjimli

closed 2020-09-30 by WeSrc-Yin