zlgopen/awtk #486: awtk-restful-httpd: idl_gen running failed

suzp1984, 2020-10-05:
# ENV
  Mac OS

# Error Stack
>  internal/modules/cjs/loader.js:968
>   throw err;
>   ^
> 
> Error: Cannot find module 'glob'
> Require stack:
> - /Users/jacobsu/hack/absurd/awtk/tools/idl_gen/index.js
>     at Function.Module._resolveFilename (internal/modules/cjs/loader.js:965:15)
>     at Function.Module._load (internal/modules/cjs/loader.js:841:27)
>     at Module.require (internal/modules/cjs/loader.js:1025:19)
>     at require (internal/modules/cjs/helpers.js:72:18)
>     at Object.<anonymous> (/Users/jacobsu/hack/absurd/awtk/tools/idl_gen/index.js:3:14)
>     at Module._compile (internal/modules/cjs/loader.js:1137:30)
>     at Object.Module._extensions..js (internal/modules/cjs/loader.js:1157:10)
>     at Module.load (internal/modules/cjs/loader.js:985:32)
>     at Function.Module._load (internal/modules/cjs/loader.js:878:14)
>     at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:71:12) {
>   code: 'MODULE_NOT_FOUND',
>   requireStack: [ '/Users/jacobsu/hack/absurd/awtk/tools/idl_gen/index.js' ]
> }
> exe cmd: node "/Users/jacobsu/hack/absurd/awtk/tools/idl_gen/index.js" idl/idl.json src failed.

# Reason

awtk/tools/idl_gen/index.js has a dependency on the `glob` package.

awtk/tools/idl_gen/index.js 里面有个依赖包 `glob`.

# Solution One

在文档中指导全局安装`glob`，但是如果我用`npm i -g glob`, 安装的目的地是`/usr/local/lib/node_modules`, 此时还是会报这个错误。
但是如果我用`npm i glob`，安装的目的地是`$HOME`目录，此时可以解决这个问题，不知道原因啥，可能这就是node的工作方式。

add a document to remind the user to install `glob` globally.

In my case, if I install the glob package globally by `npm i -g glob` which would install it at `/usr/local/lib/node_modules`, but it doesn't work. 

But if I install the `glob` package by `npm i glob` which would install glob at $HOME directory, It works well, I don't know why? Maybe that's Node's way.

# Solution Two

在`awtk/tools/idl_gen`里面执行`npm init` 初始化一个`package.json`, 然后安装`npm i glob`; 这种方式可以工作；但是需要对工程配置做出点修改：
   1. 在Scons 配置脚本的合适位置执行 `npm i`;
   2. .gitignore 配置忽略`node_modules`, 不知道npm工程是否要忽略`package-lock.json`;

Initialize a package at `awtk/tools/idl_gen` by `npm init`, then install the `glob` package, `npm i glob`; It works well;
But we need to:
  1. a place to run the `npm install` command, maybe a hook place in Scons Script;
  2. add .gitignore to ignore `node_modules`;

xianjimli, 2020-10-05:
这样试试：
```
sudo npm install -g glob
````

suzp1984, 2020-10-05:
不行；

`npm root -g`
可以得到global的node_modules路径：`/usr/local/lib/node_modules/` 是能够安装的。

suzp1984, 2020-10-05:
[idl_gen.patch.txt](https://github.com/zlgopen/awtk/files/5326154/idl_gen.patch.txt)

不知道这样行不行，

`scons tools/idl_gen` 会跑 `npm install`;
默认的`scons`不会跑 `npm install`;

在文档中说明下；
例如 `docs/custom_widget_rules.md` 中的idl_gen部分
和 `awtk-restful-httpd/README.md`

xianjimli, 2020-10-05:
我觉得还是在文档里说明一下好，AWTK 还依赖其它一些东西(如pillow)，Linux 平台就更多了，这些用户安装一次就行了。

我再想想吧。谢谢。

suzp1984, 2020-10-05:
ok

closed 2020-10-05 by suzp1984