zlgopen/awtk #495: 关于给window_manager注册点击事件发现的问题

OUving, 2020-10-24:
有这样三个注册事件，分别为点击屏幕的按下和抬起，以及button的点击，如下：
widget_on(window_manager(), EVT_POINTER_DOWN, global_clicked,win);
widget_on(window_manager(), EVT_POINTER_UP, global_clicked,win);
widget_on(button，EVT_CLICK, button_clicked, win);

前两个是程序刚启动就初始化过了，最后一个button事件是到了具体页面注册的。global_clicked( )函数执行的时候，会打印type和（x, y）坐标，当点击了Button后，弹出一个模态对话框。

操作顺序是，按下button—抬起button—弹出对话框—按下对话框的OK按钮—抬起对话框的OK按钮—模态对话框关闭。共两次按下和两次抬起。但是global_clicked( )函数打印情况是：button被按下—对话框的OK按钮被按下—对话框的OK按钮被抬起—button被抬起。事实上模态对话框弹出来的时候，button抬起的事件在global_clicked（）函数一直接收不到，直到模态对话框关闭后才能接收到（并且也是晚于对话框最后一次点击事件）。

这个应该先完整处理一次按下抬起，也就是先把EVT_POINTER_UP类型分发给global_clicked（），然后再弹出模态对话框吧？

OUving, 2020-10-24:
还有个问题是自定义了一个消息类型，比如EVT_USER_START+5。定义了一个设置全局响应的函数：
int set_user_msg_listener(event_func_t on_event, void* ctx)
{
    return widget_on(window_manager( ), EVT_USER_START+5, on_event, ctx);
}
某些界面需要收到此类型的消息（通过main_loop_queue_event( )分发的），所以进入页面都会执行这个函数设置监听，但是退出界面这个监听并不会被销毁，导致第一次进来，底层发消息，UI界面的监听函数执行一次，第二次进来会执行两次，依次类推。

这不难理解，注册window_manager( )发起的消息肯定要自己手动销毁，于是我在界面退出之前执行widget_off(window_manager( ), id)函数，id值就是上面注册监听函数的返回值。但是发现并没有效果，直到改成了下面的函数才得到解决：
int set_user_msg_listener(event_func_t on_event, void* ctx)
{
	static USER_MSG_LISTENER listener = {0};

	listener.wm = window_manager();
	if(listener.id != TK_INVALID_ID)	//保证只能存在一个user_listener 
	{
		widget_off(listener.wm, listener.id);
	}
	listener.id = widget_on(listener.wm, MSG_TYPE_UI_USR_MSG + EVT_USER_START, on_event, ctx);
	return listener.id;
}
这个不知道什么情况，还是说我用法不对，麻烦先静大佬指教。

xianjimli, 2020-10-24:
> 这个应该先完整处理一次按下抬起，也就是先把EVT_POINTER_UP类型分发给global_clicked（），然后再弹出模态对话框吧？

可以注册EVT_POINTER_DOWN_BEFORE_CHILDREN事件：

```
 44   /**
 45    * @const EVT_POINTER_DOWN_BEFORE_CHILDREN
 46    * 指针按下事件名，在子控件处理之前触发(pointer_event_t)。
 47    */
 48   EVT_POINTER_DOWN_BEFORE_CHILDREN,
```

xianjimli, 2020-10-24:
> 但是发现并没有效果，直到改成了下面的函数才得到解决

理论上是可以的。调试一下，看看是widget_off没有执行，还是执行失败。

OUving, 2020-10-24:
> > 但是发现并没有效果，直到改成了下面的函数才得到解决
> 
> 理论上是可以的。调试一下，看看是widget_off没有执行，还是执行失败。

好的谢谢，之前在窗口退出之前打印widget_off返回值是0，这个应该是正确的，但是为啥在自己UI退出前调用这个并没有起作用，不知道为啥，需要再排查。