zlgopen/awtk #497: 问题：点击页面上的编辑框，弹出软键盘屏幕被压缩后，点击压缩之后屏幕上的按键，会出现按键无响应或者响应错误的事件的现象。

ppinger, 2020-10-26:
前提条件：
1.有竖排按钮。
2.编辑框在屏幕底部，致使点击编辑框弹出软键盘压缩屏幕页面。
操作方法:
1.点击编辑框弹出软键盘。
2.点击页面上的按钮。
现象：
如果点击屏幕偏上的按钮，按键会无响应。
如果点击屏幕偏下的按钮，按钮响应的事件非当前按钮绑定的事件。

WNsACE, 2020-10-28:
你好，可以提供相关的 UI 文件吗？
我把在 demoui 的 edit.xml 文件修改如下，也没有出现你上面说的情况：
`
<window >
  <button name="close" x="c" y="10" w="25%" h="30" text="close" />
  <button name="close" x="c" y="50" w="25%" h="30" text="close" />
  <button name="close" x="c" y="90" w="25%" h="30" text="close" />
  <button name="close" x="c" y="130" w="25%" h="30" text="close" />
  <button name="close" x="c" y="170" w="25%" h="30" text="close" />
  <button name="close" x="c" y="210" w="25%" h="30" text="close" />
  <button name="close" x="c" y="250" w="25%" h="30" text="close" />
  <button name="close" x="c" y="290" w="25%" h="30" text="close" />
  <button name="close" x="c" y="330" w="25%" h="30" text="close" />
  <button name="close" x="c" y="370" w="25%" h="30" text="close" />
  
  <edit x="c" y="b" w="80%" h=36 />
</window>
`
我按照上面的 ui 文件测试出来的结果是点击编辑框后，弹出软键盘同时编辑框和按钮都移动到软件盘上面，然后点击按钮后，软键盘会关闭，然后再次点击按钮才触发按钮的事件，没有出现你上面描述的问题。

ppinger, 2020-10-28:
你需要在c代码里注册这一竖排按钮的点击事件，每个事件绑定输出标记按钮的log，这样你就知道点击的按钮与响应的事件是否对应。
---------------------------------------------

WNsACE, 2020-10-28:
上面的测试程序为 demoui，当 demoui 打开窗口的时候，会为窗口中的名为 close 的按钮注册退出窗口的事件的，我使用的 awtk 版本是 10-27 号的，所以点击按钮会会触发退出当前窗口的事件，但是是在第二次点击的时候才发生的，上面也提到触发按钮事件的情况，要不你提供一下测试项目出来看看？

ppinger, 2021-02-23:
明天我给你提供个测试的demo吧。