zlgopen/awtk #498: 如何将第三方模块的触摸信息嵌入到我们的awtk UI程序里面？

WeSrc-Yin, 2020-11-02:
问题详细描述：linux系统ARM平台。我们有A、B两个程序，A、B两个程序跑在不同的芯片上。A程序是我们使用AWTK做的上层应用UI程序；B程序跟触摸屏相连，B程序负责将触摸屏信息（up、down、hold，x，y）发送给A程序;

我现在的做法是：在A程序写个函数，去处理B程序发过的触摸信息，函数里面去判断按下、抬起，坐标。举例如下：
    if((x>=1184)&&(x<=1286)&&(y>=100)&&(y<=170))
    {//处理 “确定” 按钮
       if(pressDown)
           do something;
       if(pressUp)
          do something;
    }
    else if((x>=1184)&&(x<=1286)&&(y>=530)&&(y<=600))
    {//处理 “取消” 按钮
       if(pressDown)
           do something;
       if(pressUp)
          do something;
    }

这样的做法显得很笨很low；
要怎么样才能让程序自动去判断pressDown pressUp以及坐标？因为触摸屏不跟A程序跑的芯片相连，我无法tslib_thread_run(TS_DEVICE_FILENAME, input_dispatch_to_main_loop, loop, lcd->w, lcd->h);这样从驱动层获得触摸信息。

不知道我把问题描述清楚没？

xianjimli, 2020-11-05:
模拟成一个输入设备。

WeSrc-Yin, 2020-11-05:
> 模拟成一个输入设备。
能不能稍微详细点?
我目前的想法是：在收触摸数据的地方使用widget_dispatch分发消息，然后再widget_on
不知道是否可行。

WeSrc-Yin, 2020-11-26:
建议awtk开发个模拟器，将第三方信息模拟后发送到main_loop,类似于我以前用过的一个GUI，如下：
                    HWND hForgraund = GDIOSSYS::GetForegroundWindow();
                    GDIOSSYS::PostMessage(hForgraund, WM_BUTTONDOWN, NULL, lParam);//lParam是触摸坐标

closed 2020-12-01 by WeSrc-Yin