zlgopen/awtk #500: 应用长时间运行（12小时以上），出现触摸失效

zhaoyangxia1990, 2020-11-13:
应用长时间运行（12小时以上），出现触摸失效，linux底层驱动还是ok的，cat /dev/input/event1有数据出来，i2cdetect也是没有问题，awtk不打印tslib_dispatch_one_event事件， 独立的测试程序读取出来TP坐标也是跟正常的时候是一样的

谢谢！

xianjimli, 2020-11-14:
跑的demoui吗，前段时间我在stm32和lpc上跑了4天，没有出现问题。

用gdbserver调试一下。

zhaoyangxia1990, 2020-11-16:
跑的自己的应用，我先用gdbserver调试一下，谢谢！

zhaoyangxia1990, 2020-11-16:
李工，请教gdbserver调试连接.so库的问题，现在老是提示GDB will be unable to debug shared library initializers
我用设置库的路径也没有用，set solib-search-path /opt/gcc-linaro-6.4.1-2017.08-x86_64_arm-linux-gnueabihf/arm-linux-gnueabihf/libc/lib/:/work/s3/buildroot-2017.11/output/target/usr/lib/

xianjimli, 2020-11-18:
linux系统，最好在虚拟机里，用valgrind跑一下，确认没有内存问题。

xianjimli, 2020-11-18:
> 李工，请教gdbserver调试连接.so库的问题，现在老是提示GDB will be unable to debug shared library initializers
> 我用设置库的路径也没有用，set solib-search-path /opt/gcc-linaro-6.4.1-2017.08-x86_64_arm-linux-gnueabihf/arm-linux-gnueabihf/libc/lib/:/work/s3/buildroot-2017.11/output/target/usr/lib/

记得有个线程库，一定要拷贝过去。

zhaoyangxia1990, 2020-11-19:
李工，帮忙看一下，这是有内存泄露码？为啥最后段错误？感谢！
==199108== LEAK SUMMARY:
==199108==    definitely lost: 0 bytes in 0 blocks
==199108==    indirectly lost: 0 bytes in 0 blocks
==199108==      possibly lost: 84,810 bytes in 2,225 blocks
==199108==    still reachable: 1,320,441 bytes in 4,270 blocks
==199108==                       of which reachable via heuristic:
==199108==                         newarray           : 1,536 bytes in 16 blocks
==199108==         suppressed: 0 bytes in 0 blocks
==199108== 
==199108== For lists of detected and suppressed errors, rerun with: -s
==199108== ERROR SUMMARY: 499 errors from 498 contexts (suppressed: 2 from 2)
段错误 (核心已转储)

zhaoyangxia1990, 2020-11-19:
李工，现在多台设备测试，有的三四天都不出现，有的一两天就出现了，TP驱动啥的确认没有问题，把应用kill，再重启又是OK，出问题之后tslib_thread线程没有任何调试信息输出，同时input_thread线程也是没有任何输出。

xianjimli, 2020-11-20:
valgrind 不能跑 SDL 版本，要跑awtk-linux-fb版本。

tomyqg, 2021-01-03:
我是出现应用长时间运行(大概5到6天)，系统运行会变慢