zlgopen/awtk #501: 关于定时器timer的使用

WeSrc-Yin, 2020-11-17:
我的需求：
有个事件A，最快时20ms发生一次，循环发生，周期20ms；也有可能A事件就发生一次，然后隔几秒钟或者几分钟再发生一次。当A事件发生时，我要在UI上显示一个图标。这个图标显示停留时间3秒钟（如果3S钟内，发生了多次A事件，那么以最后一次为计时起点，显示3S钟）。（根据前面描述，所以在3秒内有可能A事件发生了好多次，也有可能3s内只发生一次）

我的做法A：
1、定义一个隐藏A事件图标的回调函数：
static ret_t hideAicon(const timer_info_t* timer)
{
    if(A事件图标正在显示)
    {
         隐藏A事件图标
    }

return  RET_REPEAT;  //注意这里是repeat，不会退出这个timer
}
2、在系统一起来时，AtimerID=timer_add(hideAicon,3000)//添加一个timer，每3秒钟去执行一次隐藏A事件图标的函数
3、在A事件发生时，timer_reset(AtimerID);//每次A事件发生就重新计时

做法A 有个不算问题的问题，就是，不管A事件有没有发生，系统一直会每3秒钟执行一下hideAicon，感觉不太合理。

我的做法B：
1、定义一个隐藏A事件图标的回调函数：
static ret_t hideAicon(const timer_info_t* timer)
{
    if(A事件图标正在显示)
    {
         隐藏A事件图标
    }

return  RET_REMOVE;  //注意这里是remove，执行一次就退出这个timer
}
2、在A事件发生时，
      if（0！=AtimerID）
      {
           timer_remove(AtimerID);//因为A事件最快20ms发生一次，那么，有可能下面添加的timer还没有执行一次，就又跑这里 
          //了，所以如果在3s内发生多次A事件，那么要remove上一次的timer，防止多次进入 hideAicon函数
      }
     AtimerID=timer_add(hideAicon,3000)//添加一个timer，3秒钟后去执行隐藏A事件图标的函数
     
做法B的问题：A事件每发生一次，AtimerID值就加1，到后面，这个值会很大很大.感觉让人也不太舒服。

我更倾向于做法B，
那么，请问，有没有更好的更合理的使用timer的方法，解决我这个需求？

xianjimli, 2020-11-18:
试试timer_modify：
```
/**
 * @method timer_modify
 * 修改指定的timer的duration，修改之后定时器重新开始计时。
 * @annotation ["scriptable", "static"]
 * @param {uint32_t} timer_id timerID。
 * @param {uint32_t} duration 新的时间。
 *
 * @return {ret_t} 返回RET_OK表示成功，否则表示失败。
 */
ret_t timer_modify(uint32_t timer_id, uint32_t duration);
```

WeSrc-Yin, 2020-11-18:
李老师，你好：
1、我的理解timer_modify和timer_reset功能基本一样，timer_reset(TimerID)是使TimerID的timer重新计时，而timer_modify（TimerID，duration）也是使TimerID的timer重新计时，只是可以修改duration而已，我不需要修改duration，当timer_modify（TimerID，duration）的第二个参数设置为跟刚开始timer_add设定的duration一样时，那么，timer_modify和timer_reset功能完全一样；

2、您的回复仍然没有解决我上面两种做法的问题。
@xianjimli

WeSrc-Yin, 2020-11-24:
李老师，我看最新的代码有加timer_suspend 和timer_resume.
我将做法A改成如下：
1、在系统一起来时，AtimerID=timer_add(hideAicon,3000)//添加一个timer，每3秒钟去执行一次隐藏A事件图标的函数
                                   timer_suspend(AtimerID);//将刚创建的timer挂起，不要让这个timer一直跑。

2、定义一个隐藏A事件图标的回调函数：
static ret_t hideAicon(const timer_info_t* timer)
{
if(A事件图标正在显示)
{
隐藏A事件图标
}
 timer_suspend(AtimerID);//执行完隐藏A事件图标后就将该timer挂起，直到下一次timer_resume重新计时时间到达触发隐藏A事件图标。
return RET_REPEAT; //注意这里是repeat，不会退出这个timer
}
3、在A事件发生时，timer_resume(AtimerID);//每次A事件唤起timer并重新计时

目前测试暂无问题，我的用法应该没用错吧。

closed 2020-11-26 by WeSrc-Yin