zlgopen/awtk #506: 输入法某些字打不出来？

ytchw101, 2020-12-21:
系统默认使用某个方正字体， 用输入法打不出来，（但实际上这个字在字库里是存在的）  比如 荠菜的荠， 焗油的焗等  我的程序实现了U盘导入TXT然后解析文字， 实际使用荠  和焗  都能用这个默认字库解析出来（所以字是存在在字库里的）。
请问这该怎么办？

xianjimli, 2020-12-21:
我看了一下，应该是google pinyin问题，有点麻烦，最近有点忙。急的话，你先自己查一下。

ytchw101, 2020-12-21:
哦  我这边再看看 你那边忙好 空闲帮看看   如果弄好     @我下

xianjimli, 2020-12-26:
是buffer设置得小了一点，你更新代码试试。谢谢

ytchw101, 2020-12-28:
好的 谢谢 ！幸苦幸苦！

ytchw101, 2020-12-28:
你好我更新到最新源码  （连同资源也更新 ）后 ，输入法中文 出不来汉字，一个字都打不出来  咋回事？  平台：嵌入式ARM +LCD 480*242

ytchw101, 2020-12-28:
补充：我在老程序里 直接修改 TK_IM_MAX_CANDIDATE_CHARS  1023   试了下是可以的之前打不出来的 字 比如 荠菜的荠   都能找到了     更新新源码  中文输入不行  应该是之前你们改了啥  导致的吧？  老程序awtk源码是 2020-06-22 更新的

xianjimli, 2020-12-28:
在PC上试试。我这里也有些字也显示不出来，是因为TK_IM_MAX_CANDIDATE_CHARS改大了之后，有些生僻字也可以输入了，但是字体里面没有对应的字模。

“一个字都打不出来 ” 我这里没有出现过啊

xianjimli, 2020-12-28:
如果是生僻字的问题，要么用大字体，要么去掉词典里的生僻字。

ytchw101, 2020-12-29:
“一个字打不出来”  意思是  中文输入时，候选字那一栏 什么字都出不来

xianjimli, 2020-12-29:
重新生成资源了吗

ytchw101, 2020-12-29:
字体 、图片、style、string ui 都重新 生成了   除了 data (不知道咋生成，所以我用的是最新源码res\assets\default\inc\data  里面的data)目前程序其他都正常 除了输入法 中文输入时 候选字一个出不来

xianjimli, 2020-12-29:
有文件系统吗

ytchw101, 2020-12-29:
有

xianjimli, 2020-12-29:
这个有点麻烦，我在几个平台上测试，都是正常的。

ytchw101, 2020-12-30:
不会吧》？！！ 嵌入式平台也试过可以吗？？ （我的是嵌入式ARM+文件系统+RTT +LCD(480*272)）   其他代码都不动只是替换了src 和3rd文件夹    老的输入法没问题  新的中文还是不行        那个data资源我直接用的最新源码res...inc文件夹下的 应该可以吧？

xianjimli, 2020-12-30:
有文件系统时，资源用的res/assets/default/raw，不是res/assets/default/inc。

ytchw101, 2020-12-30:
放在 inc中 我编译时 他会直接编译到代码中  以前也我都是这样操作呢  都没问题     现在新awtk源码变了？  那个拼音相关资源都要必须放到res/assets/default/raw 夹下吗？

ytchw101, 2020-12-30:
[image]

xianjimli, 2020-12-30:
是的：

res/assets/default/raw/data/gpinyin.dat
res/assets/default/raw/data/suggest_words_zh_cn.dat

ytchw101, 2020-12-30:
gpinyin.dat 和 suggest_words_zh_cn.dat 不能向以前那样编译到程序里吗？  或者定义宏 让客户选择 放在res目录 或者编译进程序（和字体、图片等数据一样）？？             因为 我们用的SD卡芯片直接贴片到板子上  那么我们还得程序里加个功能就是把在SD卡目录下创建 res/assets/default/raw/data/ 目录 并把 gpinyin.dat。。导入到SD卡目录下

xianjimli, 2020-12-30:
应该可以。你试一下：

1. 先include 资源文件：

```
#include "default/inc/data/gpinyin_dat.data"
```

2.加入到资源管理器

```
#include "base/assets_manager.h"
...
assets_manager_t* am = assets_manager();
assets_manager_add(am, data_gpinyin_dat);
```

ytchw101, 2020-12-31:
我现在的程序 一直都是这样做的 没动过  更新过awtk源码后  就中文不能打了

 
应该可以。你试一下：

ytchw101, 2021-01-06:
采用 #include "default/inc/data/gpinyin_dat.data"   方式 还是不行！！