zlgopen/awtk #514: 编译awtk禁止生成主题文件的data文件时输出信息到stdout. NDEBUG不起作用

bung87, 2021-01-16:
我需要在node module 里调用生成资源文件的data,res类型文件
项目地址 https://github.com/bung87/awtk-xml2ui
自建了个`awtk_config.py` 增加  
`COMMON_CCFLAGS=COMMON_CCFLAGS+' -DNDEBUG=1 '`

生成主题文件default.xml依然会显示`...size=7 widget_type=edit name=number state=error...`等信息。

xianjimli, 2021-01-17:
以前用的printf，NDEBUG对此无效。我改成了log_debug，你可以用log_set_log_level关闭log。

bung87, 2021-01-17:
close in fc92378

bung87, 2021-01-22:
改成log_debug之后，   printf fflush  identifier not found
https://github.com/bung87/awtk-xml2ui/runs/1746652930?check_suite_focus=true

```
Error: D:\a\awtk-xml2ui\awtk-xml2ui\src\gen_theme.cc(10,7): error C3861: 'printf': identifier not found [D:\a\awtk-xml2ui\awtk-xml2ui\build\addon.vcxproj]
74
Error: D:\a\awtk-xml2ui\awtk-xml2ui\src\gen_theme.cc(10,7): error C3861: 'fflush': identifier not found [D:\a\awtk-xml2ui\awtk-xml2ui\build\addon.vcxproj]
```

xianjimli, 2021-01-23:
你定义一下宏：HAS_STDIO

bung87, 2021-01-23:
没事了 代码里用了`GEN_ERROR` ，作为node module也不需要这个， 去掉了就好了。

closed 2021-01-23 by bung87