zlgopen/awtk #520: 单选按钮radio_button显示icon问题

LYJ69, 2021-02-22:
一、把所有icon去掉了，预览的时候over_of_checked状态还是会显示icon。把窗口xml文件关掉重新打开，发现over_of_checked状态下的icon恢复原来的值，把icon选为无之后，重复上面的操作，发现icon还是存在。

二、从另外一个xml复制的单选按钮radio_button预览不显示，在原来的xml上是能正常显示的

（可用下面xml代码进行预览）

<popup x="c" y="m" w="100%" h="100%" name="test" style:normal:border_color="#00000000" style:normal:bg_color="#000000">
  <radio_button name="radio_button" x="363" y="48" w="427" h="116" style:normal:font_size="64" style:normal:text_color="#FFFFFF" style:normal:icon="" style:normal_of_checked:icon="" style:normal_of_checked:text_color="#FFFFFF" style:normal_of_checked:font_size="64" style:pressed:border_color="#1971DD" style:pressed:text_color="#FFFFFF" style:pressed:font_size="64" style:pressed:icon="" style:over:border_color="#1971DD" style:over:font_size="64" style:over:text_color="#FFFFFF" style:over:icon="" style:disable:font_size="64" style:disable:icon="" style:focused:text_color="#FFFFFF" style:focused:font_size="64" style:focused:icon="">
    <property name="style:disable:text_color">
      <![CDATA[#FFFFFF]]>
    </property>
    <property name="style:normal:text_align_h">
      <![CDATA[center]]>
    </property>
    <property name="style:pressed:text_align_h">
      <![CDATA[center]]>
    </property>
    <property name="style:over:text_align_h">
      <![CDATA[center]]>
    </property>
    <property name="style:normal:margin_top">
      <![CDATA[10]]>
    </property>
    <property name="style:pressed:margin_top">
      <![CDATA[10]]>
    </property>
    <property name="style:over:margin_top">
      <![CDATA[10]]>
    </property>
    <property name="style:disable:text_align_h">
      <![CDATA[center]]>
    </property>
    <property name="style:disable:margin_top">
      <![CDATA[10]]>
    </property>
    <property name="style:focused:text_align_h">
      <![CDATA[center]]>
    </property>
    <property name="style:focused:margin_top">
      <![CDATA[10]]>
    </property>
    <property name="style:normal_of_checked:border_color">
      <![CDATA[#1971DD]]>
    </property>
    <property name="style:normal_of_checked:border_width">
      <![CDATA[2]]>
    </property>
    <property name="style:normal_of_checked:text_align_h">
      <![CDATA[center]]>
    </property>
    <property name="style:pressed_of_checked:border_color">
      <![CDATA[#1971DD]]>
    </property>
    <property name="style:pressed_of_checked:border_width">
      <![CDATA[2]]>
    </property>
    <property name="style:pressed_of_checked:text_color">
      <![CDATA[#FFFFFF]]>
    </property>
    <property name="style:pressed_of_checked:font_size">
      <![CDATA[64]]>
    </property>
    <property name="style:pressed_of_checked:text_align_h">
      <![CDATA[center]]>
    </property>
    <property name="style:pressed_of_checked:margin_top">
      <![CDATA[10]]>
    </property>
    <property name="style:normal_of_checked:margin_top">
      <![CDATA[10]]>
    </property>
    <property name="style:pressed:border_width">
      <![CDATA[2]]>
    </property>
    <property name="style:over:border_width">
      <![CDATA[2]]>
    </property>
    <property name="style:over_of_checked:border_color">
      <![CDATA[#1971DD]]>
    </property>
    <property name="style:over_of_checked:border_width">
      <![CDATA[2]]>
    </property>
    <property name="style:over_of_checked:text_color">
      <![CDATA[#FFFFFF]]>
    </property>
    <property name="style:over_of_checked:font_size">
      <![CDATA[64]]>
    </property>
    <property name="style:over_of_checked:text_align_h">
      <![CDATA[center]]>
    </property>
    <property name="style:over_of_checked:margin_top">
      <![CDATA[10]]>
    </property>
    <property name="style:disable_of_checked:text_color">
      <![CDATA[#FFFFFF]]>
    </property>
    <property name="style:disable_of_checked:font_size">
      <![CDATA[64]]>
    </property>
    <property name="style:disable_of_checked:text_align_h">
      <![CDATA[center]]>
    </property>
    <property name="style:disable_of_checked:margin_top">
      <![CDATA[10]]>
    </property>
    <property name="style:disable_of_checked:border_color">
      <![CDATA[#1971DD]]>
    </property>
    <property name="style:disable_of_checked:border_width">
      <![CDATA[2]]>
    </property>
    <property name="style:focused_of_checked:border_color">
      <![CDATA[#1971DD]]>
    </property>
    <property name="style:focused_of_checked:border_width">
      <![CDATA[2]]>
    </property>
    <property name="style:focused_of_checked:text_color">
      <![CDATA[#FFFFFF]]>
    </property>
    <property name="style:focused_of_checked:font_size">
      <![CDATA[64]]>
    </property>
    <property name="style:focused_of_checked:text_align_h">
      <![CDATA[center]]>
    </property>
    <property name="text">
      <![CDATA[radio_button]]>
    </property>
  </radio_button>
  <radio_button name="" style:normal:text_color="#FFFFFF" value="false" style:normal:font_size="64" style:pressed:text_color="#1971DD" style:pressed:font_size="64" style:over:text_color="#1971DD" style:over:font_size="64" style:disable:text_color="#FFFFFF" style:disable:font_size="64" style:focused:text_color="#FFFFFF" style:focused:font_size="64" style:normal_of_checked:text_color="#FFFFFF" style:normal_of_checked:font_size="64" style:pressed_of_checked:text_color="#1971DD" style:pressed_of_checked:font_size="64" style:over_of_checked:font_size="64" style:over_of_checked:text_color="#1971DD" style:disable_of_checked:text_color="#FFFFFF" style:disable_of_checked:font_size="64" style:focused_of_checked:text_color="#FFFFFF" style:focused_of_checked:font_size="64" feedback="true" x="389" y="205" w="428" h="81">
    <property name="style:normal:text_align_h">
      <![CDATA[left]]>
    </property>
    <property name="text">
      <![CDATA[Do not show]]>
    </property>
  </radio_button>
</popup>

WNsACE, 2021-02-22:
1，在 xml 上面没有看到 over_of_checked:icon 属性的相关设置啊，应该没有去除 over_of_checked:icon 属性。
2，要看是不是其他项目的 xml 的风格中写了 radio_button 相关属性，而在你原来的项目中的风格没有配置任何的 radio_button 相关属性。
第二个你可否提供相关测试项目来测试呢？

LYJ69, 2021-02-22:
xml都是通过AWTK Designer生成的，并且我只是拖一个进去然后修改 radio_button 所有状态的风格，修改完后直接通过AWTK Designer预览，就发现 over_of_checked状态icon显示不对。之后我还测试过，如果只修改 radio_button 少量的风格属性，icon全部设为无，预览时是正常的。当接着修改比较多的风格属性，再预览就出现这种现象了。而且有时候不一定是over_of_checked状态异常，有时候时normal_of_checked

WNsACE, 2021-02-22:
但是从 xml 上面看，的确是没有把 over_of_checked:icon 设置为无，如果出现异常了后，你查看一下 xml 中没有设置？

LYJ69, 2021-02-23:
今早我发现xml里面只要icon是这样写的就没问题
[image]

如果是下面这样就不行
[image]