zlgopen/awtk #521: scrollview里加入两种不同高度的list_item，导致滑动不会自动保持出现在屏幕中

Sokyx, 2021-02-23:
正常情况：scrollview中加入list_item1.xml，scrollview作为父控件，在代码里添加子控件list_item1,。
1、scrollview如下：
<list_view name="home_addrbook_list" x="0" y="0"  w="100%" h="-109" item_height="80" default_item_height="80" >
      <scroll_view name="addrbook_view" x="0" y="0" w="100%" h="100%">
	      
      </scroll_view>
      <scroll_bar_m name="bar" x="right" y="0" w="6" h="100%" value="0"/>
</list_view>

2、list_item1.xml如下：
<list_item w="100%" text="">
	<button style="home_addrbook" name="addrbook_button" x="0" y="0" w="100%" h="100%" text=""/>
</list_item>

代码中通过widget_add_child(scroll_view, list_item);的方式添加item，假设一页可以显示10个，如果我只添加5个，那往上滑动松开手指，5个item会自动掉下来，又显示在屏幕之中，这个是正常情况。

不正常情况：多了一个list_item2.xml
3、list_item2.xml如下：
<list_item w="100%" h="40" text="">
	<label style="home_floor_label" name="home_floor_label" x="0" y="0" w="100%" h="100%" text=""/>
</list_item>
指定了高度为40，小了一半。目的是为了和list_item1中区分开。这时就会出现滑动上去，部分item不会自动掉下来，如果item很多，这时候滑动上去，松开手指，整个屏幕就会一个item都没有

请问这个是什么情况？

WNsACE, 2021-02-23:
是一定需要通过 widget_add_child 添加才会出现这种情况的吗？ xml 上面直接分别带有不同高度的 list_item 的话，会出现上面表述的问题？

Sokyx, 2021-02-23:
1、目前只用widget_add_child试过，两个list_item.xml高度不同，加进去就有这个问题。
2、h=40的item，显示高度确实只有40。并且滑动会出现上述问题
3、如果我把list_view设置的item_height和default_item_height都去掉，而是在list_item1.xml设置h=80，list_item2.xml指定h=40，就不会有问题。

WNsACE, 2021-02-24:
我今天使用最新的 awtk 测试并没有出现你说的情况，不管是 list_item1.xml 还是 list_item2.xml 的高都是为 item_height 的值，你测试的 awtk 版本是最新的吗？你试一下用最新的 awtk 测试一下。

Sokyx, 2021-02-24:
不是最新的，是几个月之前的。

WNsACE, 2021-02-24:
今年1月25号的时候，重构了这一部分的代码并且修复了一些当时已知的代码，可能在那个时候已经修复了这个问题了

Sokyx, 2021-02-24:
好的，谢谢。