zlgopen/awtk #523: 切换主题后，当前窗口不会刷新主题，新打开的窗口会以新主题显示

LYJ69, 2021-02-25:
调用widget_set_theme(window_manager(), "dark")后，当前打开的窗口不会切换主题，新打开的窗口会以新主题显示。

WNsACE, 2021-02-25:
1，调用 widget_set_theme 切换主题需要文件系统
2，当前窗口的主题资源是否在新主题的目录下（assets/新主题/style/当前窗口风格文件），如果找不到对应主题的话风格会使用该新主题 default 风格，如果连新主题 default 风格都没有的话就会使用默认主题的 default 风格
今天我用新的 awtk 测试了是可以的，所以你可以更新一下 awtk 看看

LYJ69, 2021-02-25:
调用 widget_set_theme之后当前窗口没有重新载入，有没有例程让我参考。
我用的是MVVM框架，页面都是数据绑定+命令绑定，如果当前窗口重新载入，是不是模型数据也重新加载了。

WNsACE, 2021-02-25:
你用最新的 awtk 中的 demoui ，有切换主题的功能，但是需要提前通过 python ./scripts/release.py all 来把 awtk 的资源生成出来就可以测试了

LYJ69, 2021-02-26:
还有个问题：
在 demoui 中，修改 system_bar 的 default 主题下 digit_clock 控件的 font_name 为 trado，然后再修改dark\styles\system_bar.xml添加 font_name="ap"。
运行程序切换主题，digit_clock 控件字体样式并没有变化。按照上面操作步骤，修改字体大小的结果也是一样。

是不是修改默认主题下风格属性值，在另外一个主题也修改同样的风格，切换主题后不会变化。

<digit_clock>
  <style name="time">
    <normal text_color="green" font_name="ap" />
  </style>
</digit_clock>

WNsACE, 2021-02-26:
你修改后资源需要重新生成的，你去查看资源是否有生成出来？

LYJ69, 2021-02-27:
有重新生成

WNsACE, 2021-03-01:
你能不能把你觉得有问题的项目发出来看看？尽量简单一点的项目

LYJ69, 2021-03-01:
就拿awtk例程，demoui修改就行了。
1、修改 system_bar.xml 的 digit_clock 控件样式-->normal下的text_color为#FF0000，然后重新生成资源文件。
2、运行demoui，点击切换主题，然后就可以看到digit_clock 控件颜色没有变化。正常应该是变成绿色

WNsACE, 2021-03-03:
1，text_color为#FF0000 是红色
2，重新生成资源的命令是（python ./scripts/update_res.py all）,生成资源后，在 awtk\res\assets\dark\raw\styles\system_bar.bin，你最好先把 res 文件夹删除，再生成资源，看一下是否正常生成出来。
3，我用最新的 awtk 的话，是可以正常切换的，你用的 awtk 版本是什么时候的？

LYJ69, 2021-03-04:
我指的是修改design/default/ui/system_bar.xml的digit_clock ，修改text_color为#FF0000（原本默认白色）。然后重新生成资源文件。
运行demoui，此时digit_clock显示红色，点击切换主题，然后就可以看到digit_clock 控件颜色没有变化。正常应该是变成绿色

最新的awtk源码也是一样有这个问题。

WNsACE, 2021-03-04:
这个是 gif 是我刚刚用最新的 awtk 录出来的，
[image]

LYJ69, 2021-03-04:
我知道例程是可以的。但问题是如果我修改了digit_clock 的颜色之后，再切换主题，digit_clock 的颜色没有变化。

你这个并没有修改design/default/ui/system_bar.xml的digit_clock 的颜色。
[image]

WNsACE, 2021-03-04:
这是我刚刚只修改了 default 主题的 system_bar.xml 的 digit_clock 的颜色，所录制的视频，
[image]

LYJ69, 2021-03-05:
能把改过的xml给我看看吗，我看一下和我改的有什么不一样

WNsACE, 2021-03-05:
整个 awtk 项目只修改了截图中的地方，我在想是不是你也把其他的 system_bar.xml 的 digit_clock 的颜色也修改了才出现这个问题的。
[image]

LYJ69, 2021-03-05:
原来你修改的是风格文件default/**style**/system_bar.xml

我修改的UI文件default/**ui**/system_bar.xml

WNsACE, 2021-03-05:
我的版本就是这两天的版本，应该不是版本的问题

WNsACE, 2021-03-05:
在 ui 上面修改风格是最优先的，这个和用什么主题没有关系，切换主题就是切换主题用的资源，所以只能修改 style 文件夹下的风格