zlgopen/awtk #524: 图片缓存后，将图片资源目录中的图片更换，不重启就无法显示的问题

chenchangchang111, 2021-02-27:
现在我想做一个来回切换图片的界面，将资源中./image/ ,切换的某一个图片换成一个新的图片,来回切换还是旧的图片，重启就可以用刚刚换的新的了，有什么方法更换了图片目录的图片，界面上不重启也能更换呢，换图片用的image_set_image

chenchangchang111, 2021-03-01:
已解决

Sokyx, 2021-03-04:
这个是要手动widget_load_image，然后再widget_unload_image卸载了，就可以吧？

WeSrc-Yin, 2021-05-29:
> 已解决

怎么解决的？我遇到同样的问题。widget_unload_image 需要传一个bitmap的指针过去，怎么转换？
你能贴下你的代码看看吗?

Sokyx, 2021-05-29:
> > 已解决
> 
> 怎么解决的？我遇到同样的问题。widget_unload_image 需要传一个bitmap的指针过去，怎么转换？
> 你能贴下你的代码看看吗?

我之前是widget_load_image先加载出来，再widget_unload_image卸载就可以了。

WeSrc-Yin, 2021-05-29:
> widget_load_image

我是用image_set_image显示文件系统某个目录下的图片文件，然后，图片刷新了（图片名字没变），但我的显示没刷新。就跟楼主chenchangchang111的问题应该是一样的。

Sokyx, 2021-05-29:
> > widget_load_image
> 
> 我是用image_set_image显示文件系统某个目录下的图片文件，然后，图片刷新了（图片名字没变），但我的显示没刷新。就跟楼主chenchangchang111的问题应该是一样的。

嗯，我知道，先widget_load_image(win, bitmap, "xx"); 再widget_unload_image(win, bitmap); 再image_set_image

WeSrc-Yin, 2021-05-29:
@xianjimli 李老师，怎么解决这个问题？