zlgopen/awtk #526: 使用STM32F4的板子，移植完AWTK之后软键盘无法呼出

MrDurian, 2021-02-27:
1.硬件:stm32F4  系统:rt-thread   
使用HelloDesigner例程，移植到板子后发现软键盘无法呼出错误信息为
*** assertion failed: !"supported", dile packages \awtk\src\ui_loader\ui_builder_default.c, line 43

SIGABRT: Abnormal termination
仿真查看错误信息后 发现程序在这里死的。好像是keyboard 控件没有加载成功
widget = widget_factory_create_widget(widget_factory(), type, parent, x, y, w, h);
  if (widget == NULL) {
    log_debug("%s: not supported type %s\n", __FUNCTION__, type);
    assert(!"not supported");
  }

  b->widget = widget;
  b->widget->loading = TRUE;
  if (b->root == NULL) {
    b->root = widget;
  }

xianjimli, 2021-02-27:
可能是扩展控件没有注册，看看是不是定义了WITHOUT_EXT_WIDGETS：
```c
177 #ifndef WITHOUT_EXT_WIDGETS
178   tk_ext_widgets_init();
179 #endif /*WITHOUT_EXT_WIDGETS*/
```

MrDurian, 2021-02-27:
您好，我把这个宏打开之后，编译完成之后烧到板子中后，发现生成的文件太大了，F4的flash是1M的生成的文件就900多K，请问能不能裁剪一下？您那便能不能给个建议呢？

xianjimli, 2021-02-28:
如果不需要完整字体和输入法，可以在tk_ext_widgets_init里，把不需要的控件注释掉。否则需要外接flash，启用文件系统，把资源放到文件系统中。

xxs5387, 2021-03-01:
谢谢，目前我们也是这样解决的。今天我试了一下，把HelloDesigner-Demo  移植到了板子上，点击edit控件键盘还是无法出现。还是报出以下错误信息，WITHOUT_EXT_WIDGETS： 这个宏我也 注释掉了。现在板子上移植的版本和designer的发行版本差别有点大，不知道是不是这个问题，造成的键盘无法呼出

*** assertion failed: !"supported", dile packages \awtk\src\ui_loader\ui_builder_default.c, line 43

SIGABRT: Abnormal termination