zlgopen/awtk #529: 嵌入式linux，更新最新的AWTK后旋转270度死机必现

Sokyx, 2021-03-04:
1、昨晚（2021-03-03）更新最新的awtk和awtk-linux-fb后，编译生成新的库，应用层初始化旋转270死机必现。
经调试，死机打印在src/blend/rotate_image_bgra8888.c的rotate_bgra8888_image函数，调用rotate_image里，发现是：
case LCD_ORIENTATION_270: {
      for (i = 0; i < h; i++) {
        pixel_dst_t* s = src_p;
        pixel_dst_t* d = dst_p;

        for (k = 0; k < w ; k++) {
          *d = *s++; //死机的地方
          d = (pixel_dst_t*)(((char*)d) + dst_line_length);
        }

        dst_p--;
        src_p = (pixel_dst_t*)(((char*)src_p) + src_line_length);
      }
      break;
    }
估计是d的地址越界了。将for（k = 0； k < w; k++）改成for(k=0; k < w-1; k++)就不会死机，但是旋转完屏幕最右侧竖向的一个像素等于是没渲染的（白色）。但具体死机原因不清楚。

2、更新完之后，发现scrollview加入的list_item没填满的时候（比如按高度来说，scrollview一页能装6个item，但是我整个就只加一个item），滑动刷新会有如下打印：
scroll_bar_mobile_get_dragger_size:78 condition(virtual_size >= widget_h) failed!
scroll_bar_mobile_on_paint_self:100 condition(scroll_bar_mobile_get_dragger_size(widget, &r) == RET_OK) failed!
看代码这个好像是正常的，但是我记得更新之前没见过有这个打印啊？

WNsACE, 2021-03-04:
1，可否提供一个简单可以复现的项目给我们测试，我用 awtk 的 demoui 项目测试不出来上面的说的问题。
2，如果是第一次调用 rotate_image 函数也有这个问题的话，检查一下是否是传入的 fb 地址（dst_p 本质是 fb 的地址来的）或者 rotate_image 函数的参数。
3，第二个问题的话，是因为 1 月 25 号重构了相关的代码，虚拟高最小不再是 widge 的高而导致的出现的打印日志信息的。
主要是第一个问题，最好可以提供一个可以复现的简单项目给到我们测试，这样子容易定位到问题

Sokyx, 2021-03-04:
我看了下，是第二次进去的时候死机的，写了个测试demo，怎么发给你，打印如下：
=====dvp_gui_init,9=====
try /data/awtk_demo
app_root=/data/awtk_demo
fb_info_t: /dev/fb0
xres=1024 yres=600
xres_virtual=1024 yres_virtual=1200
bits_per_pixel=32 line_length=4096
fb_info_t: red(16 8) green(8 8) blue(0 8)
xpanstep=1 ywrapstep=0
fb_size=2457600 fb_total_size=4915200 fb_nr=2 smem_len=4915200
fb_open clear
fb_open ok
sh: dmidecode: not found
display_thread start
ratio=1.000000 1024 600
tslib_run:114: open tslib successful, filename=/dev/input/event1
input_run:310: open keyboard failed, fd=-1, filename=/dev/input/event10
input_run:254: open mouse failed, fd=-1, filename=/dev/input/mouse0
vgcanvas_clip_rect:92 condition(vg != NULL && vg->vt->clip_rect != NULL) failed!
window system_bar open
window stair_main open
#######11111111,rotate_image,o=270, fb_bpp=4, img_bpp=4, dst_line_length=4096, src_line_length=2400
######222222,rotate_image,dx=1024, dy=0, src=0xbe952a3c, dst=0xbe9529f8, src_p=0xb6570008,dst_p=0xb67ca000
#######11111111,rotate_image,o=270, fb_bpp=4, img_bpp=4, dst_line_length=4096, src_line_length=2400
######222222,rotate_image,dx=1024, dy=0, src=0xbe952a3c, dst=0xbe9529f8, src_p=0xb6570008,dst_p=0xb6a22000
Segmentation fault

Sokyx, 2021-03-04:
我刚测试了一下，发现，就一个点的像素有问题，去除k==w-1并且i==0的点，不执行*d=*s++，就不会死机，该点是旋转270度后，屏幕右上角的顶点。
case LCD_ORIENTATION_270: {
      for (i = 0; i < h; i++) {
        pixel_dst_t* s = src_p;
        pixel_dst_t* d = dst_p;

        for (k = 0; k < w; k++) {
			if((k<w-1) || i > 0)
          *d = *s++;
          d = (pixel_dst_t*)(((char*)d) + dst_line_length);
        }

        dst_p--;
        src_p = (pixel_dst_t*)(((char*)src_p) + src_line_length);
      }
      break;
    }

WNsACE, 2021-03-04:
感觉是 d 的地址的问题吧，你试一下用 demoui 会不会出现这个问题？

Sokyx, 2021-03-04:
好的，明天去测试一下看看

Sokyx, 2021-03-05:
在demo_ui_app.c的application_init函数里，加上tk_set_lcd_orientation(LCD_ORIENTATION_270);不会死机，有点懵逼了。。。我暂时先在rotate_image里加上if((k<w-1) || i > 0)，去除这个顶点吧。附件是我弄的一个简单的项目，如果你方便测试的话，里面需要替换Makefile的编译链、-I引用源码的路径、awtklib里的库需要重新编、libts.so需要替换，然后应该可以直接make成功。

WNsACE, 2021-03-06:
我在家用 linux 的虚拟机跑你的 demo 是正常的，没有出现崩毁的问题，
[image]

Sokyx, 2021-03-06:
好吧，神奇。还有个问题就是，我发现更新了AWTK后，有个dialog模态对话框，有个定时器3s后会执行dialog_quit函数来执行这个对话框，但是在这个对话框弹出来还没到3s，就马上进入另一个界面，3s时间到，执行那个定时器里的dialog_quit函数，里面会判断dialog是否已打开，这时候会判断出错，导致并没有去关闭dialog，然后窗口退出的时候，就会发现那个dialog一直都在，不会退出的bug。没更新之前，是没有这个问题的。

xianjimli, 2021-03-07:
用valgrind跑一下，看看有没有内存问题。

WNsACE, 2021-03-08:
在 is_dialog_opened 函数中，执行到哪里返回 FLASE 呢？其中 widget 的参数有没有问题？或者检查一下 dialog_queit 函数传入的 widget 是否是该对话框的对象。

Sokyx, 2021-03-08:
我刚刚测试了一下，发现是那个dialog弹窗刚弹出来，马上就进入了另一个window界面，太快了，导致dialog都没有执行EVT_WINDOW_WILL_OPEN、EVT_WINDOW_OPEN和EVT_WINDOW_TO_BACKGROUND这三个步骤，然后is_dialog_open函数里判断的stage值是WINDOW_STAGE_NONE，当window退出来的时候，才打印了一句EVT_WINDOW_TO_FOREGROUND，然后dialog就一直没有被关闭了，除非下一个dialog弹出来。打印如下：
555555555555,widget->name=stair_main,EVT_WINDOW_TO_BACKGROUND,window_base_on_event
222222222222222,widget->name=stair_main_calling,EVT_WINDOW_OPEN,window_base_on_event
........
333333333333333,widget->name=stair_main_calling,EVT_WINDOW_CLOSE,window_base_on_event
(stair_main_calling关闭的时候，先打开的是comm_msg_box-dialog弹窗，然后瞬间又进入stair_record_audio，结果dialog都没有打印)
44444444444444,widget->name=stair_main,EVT_WINDOW_TO_FOREGROUND,window_base_on_event
44444444444444,widget->name=stair_record_audio,EVT_WINDOW_TO_FOREGROUND,window_base_on_event
1111111111111,widget->name=stair_record_audio,EVT_WINDOW_WILL_OPEN,window_base_on_event
555555555555,widget->name=stair_main,EVT_WINDOW_TO_BACKGROUND,window_base_on_event
222222222222222,widget->name=stair_record_audio,EVT_WINDOW_OPEN,window_base_on_event
....
##### 2222222222stage=0 || widget->parent == NULL=0( 这里返回的stage状态是WINDOW_STAGE_NONE)

333333333333333,widget->name=stair_record_audio,EVT_WINDOW_CLOSE,window_base_on_event
44444444444444,widget->name=stair_main,EVT_WINDOW_TO_FOREGROUND,window_base_on_event
44444444444444,widget->name=comm_msg_box,EVT_WINDOW_TO_FOREGROUND,window_base_on_event
（只打印了dialog这一条信息，stair_record_audio界面退出的时候，发现这个dialog就一直在。）

WNsACE, 2021-03-08:
正常流程不会这样子的，
在dialog_create的时候就会注册一个 idle (widnwo_manager_default_do_open_window 函数)，在这个 idle 的回调函数中就会修改窗口的 stage 值。你可以做一个简单的 demo 发出来吗？

Sokyx, 2021-03-08:
OK，测试demo用的是前两天发的那个，在那个基础上多加了两个界面，一个dialog，一个window。测试发现dialog进入window太快就会出现问题，起初在167行：evawtk_message_box("dialog_info_tile", text, 0, 2);这个函数，timeout设置10ms，也就是10ms进入window，都是没问题的。我直接改成2ms马上进入window，就会出现我说的那个bug。
[awtk_dialog_quit_bug.zip](https://github.com/zlgopen/awtk/files/6099531/awtk_dialog_quit_bug.zip)

WNsACE, 2021-03-09:
我在虚拟机上面测试，没有发现你说的问题，下面 gif 是虚拟机的录屏，只是修改了 tk_init 的宽高而已，没有修改其他的地方，但是我发现你的代码有些问题，你设置计时器为 2ms，运行 1000 次后才关闭 dialog，但是 UI 计时器的最小时间间隔是当前帧消耗的时间（ awtk 会锁帧的），所以根本不可能是 2ms 触发一次计时器，导致出现每一帧都调用这个计时器，假设每帧是16ms的话，那么就是等于是16s才会触发关闭 dialog，而你在嵌入式上面旋转 lcd，估计会更慢，所以产生了对话框不会关闭的情况？？
[image]

closed 2021-03-24 by Sokyx