zlgopen/awtk #537: 关于控件image_animation的一些优化建议

hantianheng, 2021-04-02:
你好：
我在使用image_animation做一些简单的动画效果的时候发现该控件还有一些不足的地方，这里和你们同步下：

1.在使用ret_t image_animation_set_range_sequence(widget_t* widget, uint32_t start_index,uint32_t end_index) 函数设置播放序列范围时，start_index只能小于end_index，其实这样是不对的，因为如果start_index<end_index是动画顺序播放的话，那么当我需要将动画倒放时，该函数就不能满足了。

2.文档中对于image_animation_set_range_sequence()函数的说明有一些误导，“设置播放序列比如 image 为"fire"，start_index 为 0, end_index 为 99, 将依次播放"fire0", ...,”这里没有强调这个序列号应该是十六进制的，正确的顺序为0123456789abcdef。

3.动画执行结束之后，图片就自动消失了，这样也是不正确的，我看到了是因为动画结束之后当前的index=-1，这样的话重绘的时候由于index <0 所以直接就返回了，从而产生图片自动消失的现象。 但其实如果我希望动画结束之后，动画区域不可见的话，就直接接收动画结束事件，将其visible =false就行了。

4. image_animation_is_playing()这个函数不知道是不是我的理解有问题，我认为它应该是如果当前控件不在执行动画中，那么就应该给我返回false，但是目前来看，只有在image_animation_pause()的情况下，调用image_animation_is_playing()才会返回false，这样其实该函数叫做更像是在指示，当前控件是否在暂停状态了。

我在补丁文件中对第2,4个问题进行了修改，但是第3个问题，不知道是否有其它的考量，所以就没有改。另外我们在使用lua进行awtk的开发，然后我们发现awtk-lua每次在更新之后代码的顺序都会被打乱，这样的话每次将自己的补丁和awtk-lua合并起来时，就会很麻烦，所以这里有个自私的请求，希望你们能够关注下，在自动生成awtk-lua这块的代码时，有没有办法，能够对其进行输出的代码顺序进行固定的排序。

WNsACE, 2021-04-03:
image_animation_set_range_sequence 函数是根据 format 来觉得字符串合并的格式的，并不是固定的，默认是十进制的，并不是十六进制的，其他的问题已经在新版本中修复了，具体可以看 demoui 的 image_animation 界面的 ui

hantianheng, 2021-04-06:
> image_animation_set_range_sequence 函数是根据 format 来觉得字符串合并的格式的，并不是固定的，默认是十进制的，并不是十六进制的，其他的问题已经在新版本中修复了，具体可以看 demoui 的 image_animation 界面的 ui

好的，非常感谢你的回复,