zlgopen/awtk #544: undefined reference to

308066400, 2021-05-03:
# github.com/zlgopen/awtk-go/awtk
D:\temp\go-build212845066\b038\_x001.o:_cgo_export.c:(.text+0x8e): undefined reference to `widget_on'
D:\temp\go-build212845066\b038\_x001.o:_cgo_export.c:(.text+0xc8): undefined reference to `emitter_set_on_destroy'
D:\temp\go-build212845066\b038\_x001.o:_cgo_export.c:(.text+0x100): undefined reference to `emitter_on'
D:\temp\go-build212845066\b038\_x001.o:_cgo_export.c:(.text+0x132): undefined reference to `emitter_set_on_destroy'
D:\temp\go-build212845066\b038\_x001.o:_cgo_export.c:(.text+0x1b1): undefined reference to `timer_add'
D:\temp\go-build212845066\b038\_x001.o:_cgo_export.c:(.text+0x1dc): undefined reference to `timer_set_on_destroy'
D:\temp\go-build212845066\b038\_x001.o:_cgo_export.c:(.text+0x21a): undefined reference to `timer_queue_ex'
D:\temp\go-build212845066\b038\_x001.o:_cgo_export.c:(.text+0x24a): undefined reference to `widget_add_timer'
D:\temp\go-build212845066\b038\_x001.o:_cgo_export.c:(.text+0x26b): undefined reference to `timer_find'
D:\temp\go-build212845066\b038\_x001.o:_cgo_export.c:(.text+0x299): undefined reference to `timer_set_on_destroy'
D:\temp\go-build212845066\b038\_x001.o:_cgo_export.c:(.text+0x30f): undefined reference to `idle_add'
D:\temp\go-build212845066\b038\_x001.o:_cgo_export.c:(.text+0x33a): undefined reference to `idle_set_on_destroy'
D:\temp\go-build212845066\b038\_x001.o:_cgo_export.c:(.text+0x36c): undefined reference to `idle_queue_ex'
D:\temp\go-build212845066\b038\_x001.o:_cgo_export.c:(.text+0x392): undefined reference to `widget_add_idle'
D:\temp\go-build212845066\b038\_x001.o:_cgo_export.c:(.text+0x3b3): undefined reference to `idle_find'
D:\temp\go-build212845066\b038\_x001.o:_cgo_export.c:(.text+0x3e1): undefined reference to `idle_set_on_destroy'
D:\temp\go-build212845066\b038\_x002.o:awtk.cgo2.c:(.text+0x6e): undefined reference to `assets_manager'
D:\temp\go-build212845066\b038\_x002.o:awtk.cgo2.c:(.text+0x85): undefined reference to `assets_manager_set_theme'
D:\temp\go-build212845066\b038\_x002.o:awtk.cgo2.c:(.text+0x9d): undefined reference to `assets_manager_preload'
D:\temp\go-build212845066\b038\_x002.o:awtk.cgo2.c:(.text+0xb5): undefined reference to `assets_manager_preload'
D:\temp\go-build212845066\b038\_x002.o:awtk.cgo2.c:(.text+0xba): undefined reference to `tk_init_assets'
D:\temp\go-build212845066\b038\_x002.o:awtk.cgo2.c:(.text+0xdd): undefined reference to `log_get_log_level'
D:\temp\go-build212845066\b038\_x002.o:awtk.cgo2.c:(.text+0x150): undefined reference to `assets_manager'
D:\temp\go-build212845066\b038\_x002.o:awtk.cgo2.c:(.text+0x167): undefined reference to `log_get_log_level'
D:\temp\go-build212845066\b038\_x002.o:awtk.cgo2.c:(.text+0x19c): undefined reference to `assets_manager_set_theme'
D:\temp\go-build212845066\b038\_x002.o:awtk.cgo2.c:(.text+0x1d3): undefined reference to `log_get_log_level'
D:\temp\go-build212845066\b038\_x002.o:awtk.cgo2.c:(.text+0x21d): undefined reference to `window_manager'
D:\temp\go-build212845066\b038\_x002.o:awtk.cgo2.c:(.text+0x229): undefined reference to `widget_set_theme'
D:\temp\go-build212845066\b038\_x003.o:awtk_wrap.cgo2.c:(.text+0xe4): undefined reference to `widget_on'
D:\temp\go-build212845066\b038\_x003.o:awtk_wrap.cgo2.c:(.text+0x11e): undefined reference to `emitter_set_on_destroy'
D:\temp\go-build212845066\b038\_x003.o:awtk_wrap.cgo2.c:(.text+0x156): undefined reference to `emitter_on'
D:\temp\go-build212845066\b038\_x003.o:awtk_wrap.cgo2.c:(.text+0x188): undefined reference to `emitter_set_on_destroy'
D:\temp\go-build212845066\b038\_x003.o:awtk_wrap.cgo2.c:(.text+0x207): undefined reference to `timer_add'
D:\temp\go-build212845066\b038\_x003.o:awtk_wrap.cgo2.c:(.text+0x232): undefined reference to `timer_set_on_destroy'
D:\temp\go-build212845066\b038\_x003.o:awtk_wrap.cgo2.c:(.text+0x270): undefined reference to `timer_queue_ex'
D:\temp\go-build212845066\b038\_x003.o:awtk_wrap.cgo2.c:(.text+0x2a0): undefined reference to `widget_add_timer'
D:\temp\go-build212845066\b038\_x003.o:awtk_wrap.cgo2.c:(.text+0x2c1): undefined reference to `timer_find'
D:\temp\go-build212845066\b038\_x003.o:awtk_wrap.cgo2.c:(.text+0x2ef): undefined reference to `timer_set_on_destroy'
D:\temp\go-build212845066\b038\_x003.o:awtk_wrap.cgo2.c:(.text+0x365): undefined reference to `idle_add'
D:\temp\go-build212845066\b038\_x003.o:awtk_wrap.cgo2.c:(.text+0x390): undefined reference to `idle_set_on_destroy'
D:\temp\go-build212845066\b038\_x003.o:awtk_wrap.cgo2.c:(.text+0x3c2): undefined reference to `idle_queue_ex'
D:\temp\go-build212845066\b038\_x003.o:awtk_wrap.cgo2.c:(.text+0x3e8): undefined reference to `widget_add_idle'
D:\temp\go-build212845066\b038\_x003.o:awtk_wrap.cgo2.c:(.text+0x409): undefined reference to `idle_find'
D:\temp\go-build212845066\b038\_x003.o:awtk_wrap.cgo2.c:(.text+0x437): undefined reference to `idle_set_on_destroy'
collect2.exe: error: ld returned 1 exit status
错误: 进程退出代码 2.

xianjimli, 2021-05-04:
用mingw编译的吗

308066400, 2021-05-04:
是的，CGO_CFLAGS参数就带了-DMINGW=1，在liteide环境变量配置的，LITEIDE_MAKE=mingw64-make。
直接用cmd命令行go build button.go提示：..\awtk\awtk.go:4:18: fatal error: awtk.h: No such file or directory。
都错误。

xianjimli, 2021-05-04:
cmd下我没试过。你可以在bash下试一下，设置环境变量再编译：

```
source env.sh
```

308066400, 2021-05-04:
@xianjimli 
一样，环境变量问题，但是不知错在那里。已经source env.sh，也是没找到头文件或可能lib。
bash下也是mingw64
执行效果：
$ go build demos/button.go
# runtime/cgo
C:/TDM-GCC-64/bin/../lib/gcc/x86_64-w64-mingw32/5.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: cannot find -lawtk
collect2.exe: error: ld returned 1 exit status

这些问题反而比主题说的问题更靠前。。。主题的问题已经算是找到头文件顺利进入编译阶段，只是编译出错。

xianjimli, 2021-05-04:
awtk也是用mingw编译的吗

308066400, 2021-05-04:
@xianjimli 是的mingw64
scons: done reading SConscript files.
scons: Building targets ...
scons: `.' is up to date.
scons: done building targets.

closed 2021-05-04 by 308066400