zlgopen/awtk #548: 关于切换主题后，字符串翻译不能随之进行切换的问题

hantianheng, 2021-05-08:
你好，我在使用AWTK的时候碰到这么一个问题：
        我有三个主题，default/dark/other 每个主题下所对应的资源都不一样，它们都有属于自己的fonts/image/strings/style/ui。当我切换主题和语言之后，会对其进行掉电保存，并在系统再次上电之后，将保存的主题和语言设置下去。但是我碰到了一个问题就是**当我把主题设置为非default如dark/other，然后语言设置为en_US之后，系统再次上电字符串翻译显示的内容为default主题下的，而不是当前主题下的字符串翻译信息。其它语言不会出现这种情况**。

问题的原因：
        我看了下源码发现是当通过locale_info_change()去设置当前语言时，它会检测如果当前设置的语言和上一次locale_info中的语言信息是否相同，如果不相同的话才会去调用locale_info_reload()进行重载。
而系统上电后，将default设置为默认主题，将en_US设置为默认语言。这样的话，我在系统起来之后设置主题为dark，并通过locale_info_change()去设置当前语言为en_US的话，由于检测到当前设置语言和默认语言一样，所以将不会去调用locale_info_reload()进行重载，因此翻译信息仍然为default主题下的。 如果我在切换主题之后，去手动调用一下locale_info_reload()进行强制重载，那么该问题将会被解决。

xianjimli, 2021-05-10:
目前没什么好办法，你先这样 用吧。我再想想。

hantianheng, 2021-05-10:
> 目前没什么好办法，你先这样 用吧。我再想想。

好的