zlgopen/awtk #555: 内存泄漏问题

zxc3489, 2021-05-20:
在调用emitter_dispatch_simple_event(emitter, EVT_PROPS_CHANGED)时，发现有内存泄漏，以下为简易demo

`/**
 * Created by veng on 2020/10/26.
 */

#include "home.h"
#include "tkc/mem.h"
#include "tkc/str.h"
#include "helper/translater.h"
#include "base/timer.h"

static ret_t _test_timer(const timer_info_t* timer)
{
    home_t* home = (home_t*)(timer->ctx);
    emitter_dispatch_simple_event(EMITTER(home), EVT_PROPS_CHANGED);
    return RET_REPEAT;
}
home_t *home_create(void) {
    home_t *home = TKMEM_ZALLOC(home_t);
    emitter_init(EMITTER(home));
    timer_add(_test_timer,home,1);
    return home;
}

ret_t home_destroy(home_t *home) {
    return_value_if_fail(home != NULL, RET_BAD_PARAMS);
    emitter_deinit(EMITTER(home));
    TKMEM_FREE(home);
    return RET_OK;
}`

zxc3489, 2021-05-20:
分发其它EVT事件时，并未发现问题

xianjimli, 2021-05-21:
是mvvm吗

zxc3489, 2021-05-21:
嗯，根据进一步定位，发现是xml的数据绑定异常，类似如下的字符串拼接方式都会导致内存泄漏。代码是去年十月份的。
<label w="100%" v-data:text='{"拷机时长:"+$time_h+"时"+$time_m+"分"+$time_s+"秒"}'/>

zxc3489, 2021-05-21:
`<label w="100%" v-data:text='{"拷机时长:"+$time_h+"时"+$time_m+"分"+$time_s+"秒"}'/>`

xianjimli, 2021-05-21:
好的，谢谢。你用最新的代码试试。

zxc3489, 2021-05-24:
mvvm编译报错呢;
src/mvvm/awtk/binding_context_awtk.c:474:9: error: 'widget_t {aka struct _widget_t}' has no member named 'need_relayout_children'

xianjimli, 2021-05-25:
awtk/awtk-mvvm都要更新。

closed 2021-10-19 by zxc3489